OpenCPN/OpenCPN #1072: Messaging schemas

jongough, 2018-05-03:
OCPN messaging uses JSON which is easy to use, but for more than simple text messages the data content of a JSON message can be complex. However, currently there is no method in the JSON that OCPN is using of specifying a message content and having that validated. Each plugin and user of the messaging service has to know intimately the content of each message and parse the message to ensure it has all the information needed to perform the requested function. If the JSON message does not have all the information needed there is no standard way of telling someone about it.

This issue came up when Watchdog and OCPN_Draw started talking to each other through the messaging system. As it was one person constructing both ends of the conversation it was not too bad, but there had to be duplicate code in each plugin to handle the message correctly. This is overhead and is reinventing the wheel.

Either a psuedo schema is needed with a library that supports standard validation, or a move to a different JSON message processing library is needed. http://json-schema.org/ has some information on this.

mgrouch, 2022-07-16:
Might be swagger can help. We use swagger for that mainly in java but it supports many languages. However it's for REST APIs.

For messaging there are options Avro, Protobuf, JSON. You can convert schema from one to another

nohal, 2024-02-15:
You can easily achieve what you want in many ways, for example by using JSON encoded protobuf for your messages (with both plugins communicating simply sharing the protofile and generating the needed code from it)

Re JSON Schema, nothing prevents originator of the message to publish JSON Schema for it. Nothing makes it obligatory for the receiver of the message to use the schema (And nothing prevent him from doing so). They do not need any help from OpenCPN core with it.
The JSON Schema itself though is not exactly a mature and easy to adopt technology, incompatible with itself between the various releases, with poor tooling and is an overkill for most of the usecases I have seen in OpenCPN universe so far.

closed 2024-02-15 by nohal