OpenCPN/OpenCPN #1073: NMEA checksum '00' problems

nohal, 2018-05-04:
From the forum:

> NEMA messages with 00 checksum do not work (possibly bug)
> 
> CPN 4.8.0
> For some reason messages HDT, HDM with 00 checksum does not recognized by cpn, even no output in debug for these
> 
> $IIHDM,48,M*00
> $IIHDT,48,T*00
> 
> however messages with another checksum work perfect!
> I try another value and try to change prefix to verify messages with another checksum
> 
> $IIHDM,49,M*01
> $IIHDT,49,T*01
> and 
> $IKHDM,48,M*02
> $IKHDT,48,T*02
> These work well
>

transmitterdan, 2018-05-23:
I looked into this a little bit. I did not find anything special about having 00 as the checksum. I did find a something that can cause a problem. The code that populates the connections dialog box defaults the "Control checksum" box to true in certain cases. This then overwrites the user selection of unchecked. After that the NMEA mux ignores all messages with invalid checksums whether 00 or any other invalid one. To see if this is the problem close OpenCPN and look at opencpn.ini for the NMEA connection settings:

```
[Settings/NMEADataSource]
DataConnections=1;1;0.0.0.0;10110;0;;4800;0;0;0;;0;;1;0;1;0;1
```

If the number after the 7th ; (the number after the baud rate, 4800 in this case) is 1 then checksums are enabled for this connection. Try editing this number to 0 and then restart OpenCPN. See if the HDM data is then accepted. For me any HDM sentence with wrong checksum (even 00) is accepted if this flag is 0 and not 1. Under somewhat confusing circumstances I found that this value gets set to 1 even if I unchecked "Control checksum" box.

rose-la, 2020-09-18:
Unfortunately, this bug is back, at least in 5.2 for ubuntu.  Same work-around does cure it, in lieu of a real fix.  When I get 'round to wading into the source, I'll try to find the actual path that burned me.