OpenCPN/OpenCPN #1079: Dashboard history do not scale up well

tchiwam, 2018-05-23:
Hello I noticed that the history does not scale well in the dashboard. 

[image]
[image]

I also get this when I try to move things around in 2 dash boards to try and see how to arrange this:
opencpn: /var/tmp/portage/x11-libs/cairo-1.14.12/work/cairo-1.14.12/src/cairo-arc.c:189: _cairo_arc_in_direction: Assertion `angle_max >= angle_min' failed.
Aborted
tchiwam@x

rgleason, 2018-05-23:
Yes, you have a point, but have you tried Plots_pi by Sean D'Epagnier?

tchiwam, 2018-05-24:
Thank you for the heads up. This is a nice plugin, I just did an ebuild for Gentoo. I'll see if I can add it to the other plugins

rgleason, 2021-04-23:
The wind history and other histories are best displayed as separate dashboards where they can be adjusted freely and lengthened in the horizontal direction.   Closing now.  This is very old.

closed 2021-04-23 by rgleason