OpenCPN/OpenCPN #1095: Under linux, memory run away

tchiwam, 2018-06-22:
Hello all, I am starting to dig around a weird bug I have with s63 maps.

Maybe someone can give a me a cue on where to look next. Affected versions are 4.99 and 4.8.4

Affected s63 plugin 1.6.0  and 1.8.0

I can launch OpenCPN in a s63 map area, I can then move, zoom out and zoom in, without an issue.

Once I hit the right button, to drop a mark or select a chart group, the dialog will never show up, then the memory will increase non stop until it will stall the machine and hit the swap, then eventually die OOM.

The work around seems to be: opencpn -parse_all_enc
-Then move to an area without s63 chart
-Find an area where opencpn does not Segfault on exit
-Start opencpn once more and go directly to settings and scan Charts and Update Database , Prepare All Enc Charts, OK
-Then move to the area of interest with s63 charts, and start dropping marks and use the right click as normal. Then OpenCPN will behave until you close it again.

did-g, 2018-06-23:
Hi,
Could you run it under gdb and then Ctrl+C , bt when it eats memory? 
One reason I've seen could be memory corruption and STL  (vector, map) code.

tchiwam, 2018-06-24:
(gdb) 
  Id   Target Id         Frame 
* 1    Thread 0x7ffff7fa1940 (LWP 14867) "opencpn" 0x00007ffff21cfe24 in g_main_context_prepare () from /usr/lib64/libglib-2.0.so.0
  2    Thread 0x7fffd8f20700 (LWP 14881) "threaded-ml" 0x00007ffff3055964 in poll ()
   from /lib64/libc.so.6
(gdb) thread 1
[Switching to thread 1 (Thread 0x7ffff7fa1940 (LWP 14867))]
#0  0x00007ffff21cfe24 in g_main_context_prepare () from /usr/lib64/libglib-2.0.so.0
(gdb) l +0
915     in /var/tmp/portage/sci-geosciences/opencpn-9999/work/opencpn-9999/src/chart1.cpp
(gdb) thread 2
[Switching to thread 2 (Thread 0x7fffd8f20700 (LWP 14881))]
#0  0x00007ffff3055964 in poll () from /lib64/libc.so.6
(gdb) l +0
915     in /var/tmp/portage/sci-geosciences/opencpn-9999/work/opencpn-9999/src/chart1.cpp
(gdb) 

This is strange as it is under a __WXMSW__ ifdef section ? 
I've added a dumb print out just to make sure the debugger is not just taking the next line.

tchiwam, 2018-06-24:
Correct this, it continues to the end of the function 

wxString newPrivateFileName(wxString home_locn, const char *name, const char *windowsName)

It does not go in the WXMSW ifdef.

tchiwam, 2018-06-24:
US ENC maps, do not cause the issue, and I can use this map set to start and get to the Chart menu and do the work around. Seems to get confused when it starts using this chartlist.dat file, could I delete this file ?

nohal, 2018-06-24:
Yes, it is the chart database and will have to be rebuilt next time you start the program (immediately when it starts, not even the GUI gets initialised until it finishes), but there should be no problem if you delete it.

tchiwam, 2018-06-25:
I know I have quite many s63 charts in that database. I don't have much time to continue on this just now. But every time it is loading this file it is causing an issue.

The memory gobbling is starting as soon as you have the interface on, so even before you try to right click. For some reason moving with the arrows and page up and down works, until you try to hit any icon or right click then everything stops and the memory allocation is getting out of control.

If you press the X in the corner of the window, opencpn exits with a segfault. 

Every time gdb stops in that function. I may add some statements in there, my guess is it will not find the last record "marker" . How is that file created ? So maybe adding the s63 charts before the others might solve this issue ?

nohal, 2018-06-25:
I would be very surprised if the chart database itself had anything to do with your problem. How many is "quite many"? Thousands?

tchiwam, 2018-07-05:
The database file is about 5.4MB. I am back in the not living in a boat very limited internet. I'll take a look at this again today.

nohal, 2019-03-12:
@tchiwam Any news? I will keep it around for a while but as nobody seems to be able to reproduce this issue, it will eventually get closed.

tchiwam, 2019-03-15:
Seems this problem went away with 4.8.8 and was still there as 4.8.6.   I think this can be closed

nohal, 2019-03-15:
Thanks for the info

closed 2019-03-15 by nohal