OpenCPN/OpenCPN #1097: proper plugin header is not installed

leamas, 2018-07-05:
Devs should be able to create new plugins using an installed version of OCPN. The API is simple and described in the file ocpn_plugin.h. However, it's not installed with the package.  This makes devs bundle the file in whatever version they get hold of. Of course, this is not ideal, if/when a new version of OCPN arrives. 

To add to this, the four plugins delivered with the main package uses a few other headers.
I have patches under way for this is the flatpak packaging effort, but feels that this is worth filing a separate issue.

nohal, 2018-07-05:
This is by design - Plugins must include the `ocpn_plugin.h` corresponding to the oldest OpenCPN API version they intend to support. The API is backward compatible, so the plugins can be used with all newer OpenCPN releases and they can be built for older versions even if the developer normally uses much newer OpenCPN version than the target (as all the devs I know do all the time). On all the platforms OpenCPN supports.
Installing the header with the normal end user release IMO makes absolutely no sense. Creating a `-dev` type packages would make at least some sense on Linux (although maintaining it for a single header file seems not exactly something worth the effort), but that paradigm does not exist on Windows and on macOS or at least requires using a 3rd party system like Homebrew or vcpkg.

leamas, 2018-07-06:
OK... so we should not just install one, but all reasonable versions of the API? something like this?
[opencpn.tar.gz](https://github.com/OpenCPN/OpenCPN/files/2168759/opencpn.tar.gz)

bdbcat, 2018-07-06:
leamas

This usecase is reasonable for devs, not end users.  A dev must decide which API version to support in her plugin, and communicate that datum to the cummunity when the plugin is released.
I see no practical value in distributing the plugin API definition in a release package.

Indeed, where would we put it in *ix?  /usr/local/include?  Serves no purpose for end users, IMO.

Dave

leamas, 2018-07-06:
> This usecase is reasonable for devs, not end users. 

Indeed. That said, the common flow is that upstream installs files for all usecases while downstream packagers splits the upstream into parts like -devel, -doc, etc.

Could we agree on having the file(s) included if we define a configuration flag like -DINCLUDE_HEADERS, by default false? So that devs and packagers who needs it can get it, without affecting "normal users"?

> indeed, where would we put it in *ix? /usr/local/include?

In $CMAKE_INSTALL_PREFIX/include

nohal, 2018-07-06:
This will IMO have only one effect, confusion everywhere, which I suppose is not what you are trying to achieve. And there is no reasonable way to implement it to be reasonably be cross-platform.

No, there is no way to check anything at runtime, you simply are not able to load the dynamic library at all if there are missing symbols, which is the case as soon as it was built against a newer OpenCPN core than the user has (or there is a missing dependency, or there is a dependency conflict or...). At least not on Windows where approximately 80% of the userbase is. And our goal is not to fail, our goal is to work if possible... That goal is easily achievable the way it is done now and will be less achievable the way you suppose as at least on the most widely used platform people will have to maintain the headers manually due to the lack of packaging system or standard include locations.

It will also not be easier for anybody trying to work on someone else's code, even on Linux - while now he only needs to git clone the plug-in and is all set, if we implement your solution, he will first have to start looking for that development package with the one header file, or git clone OpenCPN and copy the headers somewhere to /usr/local/include, otherwise he will not even be able to build it.
I simply see no use case at all where someone would really profit from this.

leamas, 2018-07-06:
Oops... too much python lately. Sorry for that noise. You are of course right, the plugin won't load.
This is a sidenote which I shouldn't have done for sure. Back to the headers: are you comfortable with something like *-DINCLUDE_HEADERS"?

nohal, 2018-07-06:
I have no problem including `-DINCLUDE_HEADERS` option, I just think it is a waste of effort that if used unfortunately satisfies only one of the supported platforms.
It's only real effect, if someone decides to base their plugin development on the (wrong anywhere but on Linux) assumption that the header will be in some universally known location, will be just generating build problems we now don't have.

leamas, 2018-07-06:
Basic argument: It's wrong to install libraries without the corresponding, updated header available.
Build problems could be handled by installing all headers in $prefix/include/opencpn - this way, nothing will be included without an explicit *-I*. Or, more drastic, by handling the header as part of the docs.
I'll try to make a PR (later) - let's see where it goes...

nohal, 2018-07-06:
Well, OpenCPN is not a library and we are still speaking about one single header file that, although developed by the core program authors, represents the plugin (which is a library), not the core program...

bdbcat, 2018-07-06:
@leamus...
IMO, your basic argument rejected for released runtime packages..
The include file performs no function for the user at installation, runtime, or anytime later.
I agree that devs have other requirements, one of which is to keep track of API definition files.

leamas, 2018-07-07:
The thing seems to be some confusion about the roles for this package: runtime, developer, base for downstream packaging. Since we for practical reasons only will make one package, I think it might make sense to fulfill all these roles using configurations conditionals.

Note that the PR contains all versions of the API, this could/should probably be trimmed. Also, if I knew which tags which are actual releases the documentation in the header could be improved.

leamas, 2018-08-05:
As of now this issue is redundant, covered by PR #1097;  closing.

closed 2018-08-05 by leamas