OpenCPN/OpenCPN #1107: Units set in options are not saved when closing OpenCPN

Erik1, 2018-07-22:
The units set in options are not saved when closing OpenCPN. Everytime OpenCPN 4.8.4 starts the units are set to kilometers, km/h and meters for distance, speed and depth. It would be nice if the settings would be saved, so it can start with distance in nautical miles and speed to knots and depth in meters.

rgleason, 2018-07-23:
Erik, I believe this is required to meet the standards for Inland ECDIS. It is a specified requirement for this interface.
Please see the documentation.
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:inland_ecdis&s[]=inland&s[]=ecdis

rgleason, 2018-07-23:
Also see
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:faq#why_aren_t_my_settings_saved_properly_when_i_close_opencpn

This issue should be closed, I believe. It does not belong here.

nohal, 2019-01-19:
Inland ECDIS specification requires this behaviour.

closed 2019-01-19 by nohal