OpenCPN/OpenCPN #1108: Documentation: Missing source, privacy breaches.

leamas, 2018-07-22:
Pushing the opencpn source through debian toolchain reveals s large number of warnings similar to

    E: opencpn source: source-is-missing data/doc/opencpn/manual/toolbar_buttons/ais/ais-mob-simple-track.id.opencpn_3Aopencpn_user_manual_3Atoolbar_buttons_3Aais.jpg.detail.html line length is 291 characters (>256)

    W: opencpn: privacy-breach-generic usr/share/opencpn/doc/opencpn/files/chartsymbols/catzoc.id.opencpn_3Aopencpn_user_manual_3Acharts_3Avector_display_tab.png.detail.html [<script type="text/javascript" src="https://oss.maxcdn.com/html5shiv/3.7.2/html5shiv.min.js">] (https://oss.maxcdn.com/html5shiv/3.7.2/html5shiv.min.js)

Together, these issues makes it hard to package the documentation in a linux distro, it must probably be removed (replaced with a link?) in a proper packaging.IMHO, these warnings/errors reflects real problems which need to be dealt with.

Attaching excerpts from debian build log:

[doc.log.gz](https://github.com/OpenCPN/OpenCPN/files/2217372/doc.log.gz)

rgleason, 2018-08-13:
This github issue/ tracker item regards the dokuwiki documentation, and is probably caused by external links or links to local files, but I am not sure what is required to fix this problem. 

We normally export the current version of the wiki to html  (using a local server and copy) and then hand taylor the result. This is a significant effort so it is not done for intermediate versions.

I understand that the problem is derived from a local linux installation which has downloaded and installed the documentation package, which is then being packaged in a debian toolchain.  
Can you explain further what the specific problems are?  
what is causing privacy-breach-generic usr/share/opencpn.....?

Is fixing this achievable?

rgleason, 2018-08-13:
If I have the beginning of the line correct, from the log file, there are these kinds of errors which are links to missing files.  Here is the last successful line with the first failure following:

```
4.8.4/debian/opencpn/usr/share/opencpn/doc/menubar/ais.html
-- Installing: /home/mk/OpenCPN/opencpn-4.8.4/debian/opencpn/usr/share/opencpn/doc/menubar/tools.html
-- Installing: /home/mk/OpenCPN/opencpn-4.8.4/debian/opencpn/usr/share/opencpn/doc/menubar/navigate.html
-- Installing: /home/mk/OpenCPN/opencpn-4.8.4/debian/opencpn/usr/share/opencpn/doc/menubar/help.html
W: opencpn: privacy-breach-generic usr/share/opencpn/doc/opencpn/files/chartsymbols/catzoc.id.opencpn_3Aopencpn_user_manual_3Acharts_3Avector_display_tab.png.detail.html [<script type="text/javascript" src="https://oss.maxcdn.com/html5shiv/3.7.2/html5shiv.min.js">] (https://oss.maxcdn.com/html5shiv/3.7.2/html5shiv.min.js)
W: opencpn: privacy-breach-generic usr/share/opencpn/doc/opencpn/files/chartsymbols/catzoc.id.opencpn_3Aopencpn_user_manual_3Acharts_3Avector_display_tab.png.detail.html [<script type="text/javascript" src="https://oss.maxcdn.com/respond/1.4.2/respond.min.js">] (https://oss.maxcdn.com/respond/1.4.2/respond.min.js)
W: opencpn: privacy-breach-generic usr/share/opencpn/doc/opencpn/files/os/linux.id.opencpn_3Aopencpn_user_manual_3Agetting_started_3Aopencpn_installation_3Adedicated_marine_os.png.detail.html

```

`E: opencpn source: source-is-missing data/doc/opencpn/manual/ba_serveur_2-910db.id.opencpn_3Aopencpn_user_manual_3Aadvanced_features_3Anetwork_repeater.jpg.detail.html line length is 291 characters (>256)`

leamas, 2018-08-13:
Hi! Some quick notes from a sailing yacht at 65° N... 
The missing "files" directory seem to be a separate issue. If there is time, I will try to run a linkchecker on the docs, it sound like a great idea...

However, the messages are *not* about missing links. As I understand it, it boils down to:
  - Some toolchain messages are about the missing sources, in this case the wiki. In general, linux packaging wants to be given the sources and then run whatever tools applicable to get the distributed result. This is to make it possible to others to change the software (here: docs). In a better world there would a script or other tool which could automagically convert the wiki to completeted docs. Not only to satisfy  the debian tools, but because it would certainly make maintenance easier. Or?'
  - Other messages are about privace breaches. Without looking into the sources, in general this is about some javacript (probably just referenced) which calls home. It should not be

rgleason, 2018-08-13:
leamas, I am not a linux guy really..Win. Have used it. So I am going to have to learn about this by watching.

1.  "linux packaging wants to be given the sources and then run whatever tools applicable to get the distributed result. "   -I think (heck, I know!!)  in trimming docs down left docs out.

2. "In a better world there would a script or other tool which could automagically convert the wiki to completeted docs"  - We are using a script of sorts php programs, from Dokuwiki, siteexport plugin and dw2pdf from within dokuwiki.  It is written in PHP.  We have to have a local server setup and run the programs to be able to fully complete the documentation conversion to html.  Then because it is so big it has to be be edited and manipulated.
I have created index files that can be used to create the documentation needed, but for some reason siteeexport is not following the instructions properly.
Dave is too busy and involved for this to be a distraction. I am not program handy enough to fix it. I can help. It always lands on our laps at bad times, and Kathy B bails us all out with her editing. It would be very nice to get ahead of this this iteration, somehow.

rgleason, 2018-08-13:
PS, If you'd like access to the Wiki, I can make you a manager, if you can PM me your email. The siteexport plugin is accessed there and will do some smaller exporting, but takes ages and fails if you give it a the entire document, memory errors or something,  so we have to do that locally.

leamas, 2018-08-13:
well... frankly, given all other packaging the problems  I'd suggest to just remove the docs from the debian package and replace them with a link to the website for now.  

That said, thanks for explanations!

If we really want to fix this we might want to file the fundamental issue that is not possible to build the docs in an open process (giving me access doesn't really help here - what's needed is an open process which can be replicated by build robots and devs).

leamas, 2018-08-14:
At a second thought...

The first step in a more open process would be to publish the wiki markdown sources in the git tree. Do you think this is possible and maintainable? 

The next step would be to rebuild the documentation from these sources. I *think* it might be possible, but certainly needs investigating. Pandoc, http://pandoc.org/, seems interesting. Should work out of the the box for mediawiki input, but seemingly cannot parse docuwiki.  Perhaps some conversion hack could do it.?

In any case, getting the sources into git is an important step.

rgleason, 2018-08-14:
So after that process, what would we do?  Where does the manual reside? Where does the website point?  What would the process be whenever we have to release?

This might work depending on the amount of work. I am not at all prepared to reconstruct the manual by hand in yet another format, and I would have a very hard time convincing the many volunteers that also worked on the latest change to Dokuwiki which is still seems recent.

.

leamas, 2018-08-15:
Now, this stuff is new to me and I'm just trying ideas. The idea I'm trying to explore is that the website is where it is, no changes. 

However, the wiki sources should be in a git repo. This should be trivial (?), the docuwiki sources seems to be plain text files. This git repo could be a separate or  part of the overall opencpn tree.

Given that the git tree is public, we then need a way to rebuild the docs for those who download the sources and want a private copy. This needs to "work" in the sense that the documentation is useful, perhaps without being an exact copy of the website. Another way to address this might be to let the user install docuwiki and run a local instance displaying the docs from the sources.  

So, summing up:
  - Keep current wiki site as-is, including maintenance.
  - Put the wiki sources under git control in a public repo.
  - Investigate ways to rebuild or watch  the docs from the git sources for users  and packagers who wants an exact version of the docs  perhaps using links above

Thoughts?

transmitterdan, 2018-08-15:
If we only supply Debian packages for official releases as we have done, with the hand tailored HTML going into the package, what is wrong with that process going on forever?

OpenCPN runs on some pretty limited hardware like ARM systems. I would think it not a good idea to add to the bloat of these systems to accommodate a non-requirement of 99.9% of users.

leamas, 2018-08-15:
>  I would think it not a good idea to add to the bloat of these systems to accommodate a non-requirement of 99.9% of users.

I have no intention whatsoever to change the Debian packages built using cmake and Cpack. Fuill stop.

Now, for many reasons it would be great if we could make "official" packaging - that is, to make OpenCPN part of Linux enabling users to install it with the standard apt-get, dnf or flatpak (etc.).  If you don't share this goal, let's talk about that.

Also, an official packaging would definitely be split into several packages with the docs going into a separate package. I assume that the minimal install would actually be smaller than the cmake package. 

> what is wrong with that process going on forever?

The  problem is that users, without source, cannot modify the docs in a reasonable way. This is a standard FOSS requirement. After all, open-source requires publishing the source :)

At this point we are discussing adding the wiki sources to the git tree (or, perfectly fine, to put them into a separate repo). I see no path where these sources goes into the cmake packages unless explicitly configured.  So, no-one besides developers and packagers are affected.

However, if we could find a path to rebuild the docs from the sources we could make the maintenance and release process much easier. If nothing else works, it might at least be possible to just ship the wiki sources and let users view them on a local docuwiki instance (docuwiki is packaged in "all" distros).

rgleason, 2018-08-16:
I realize that If a copy could be made on github, it might be easier for you to change it and fix the problems, while keeping a list of how to fix them, so we can get those changes back into the Dokuwiki!

I have written the Dokuwiki forum here  https://forum.dokuwiki.org/thread/16158

leamas, 2018-08-16:
To me, the problem here is the same as backing up and restoring the site which is described in https://www.dokuwiki.org/faq:backup This in turn boils down to that you need ssh  access to https://opencpn.org/docuwiki so you can copy the files (as described in the backup link). If you don't have this, we need to know who does.

rgleason, 2018-08-16:
Well, we do all have access to the html files that are a part of every installation. Those could be put in a github repository.

Why do you see it as the same problem as backup and restoration?
I do have access.

leamas, 2018-08-16:
> I do have access.

Does this mean that you can  ssh into opencpn.org and access the files which the docuwiki instance displays i. e., the data/ directory if I got it right?

rgleason, 2018-08-16:
For the next release it might be advantageous to have the documentation available in smaller packages.
1. Basic Documentation with the package.
2.  Downloads on the Opencpn Downloads Documentation Page for 
a. [Offline User Manual - 83 pages](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toc_offline_user_manual)
b.  [Offline Plugins Extensions- 46 pages](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toc_offline_plugins_manual)
c. [Offline Advanced Features - 17 pages](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toc_offline_advanced_features) 
 d [Offline Charts Extension -5 pages (may be too small for this)]9https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toc_offline_user_manual_charts)

These are all TOC Table of Content pages we've created.
Here is a [TOC for all documentation](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toc)

The problem I would like to solve is getting the SiteExport Plugin to actually complete from the opencpn server. (It works locally.)  This would make things easier and more public for everyone.
Once it actually completes, there users would be able to create their own documentation packages. Or select from ones already available.

rgleason, 2018-08-16:
You have not answered my questions.

leamas, 2018-08-16:
These? 

> Some of my questions about duplicating content, danger of divergence, and ambivalence of two instances and defined process, have not been addressed.

As far as we are just publishing the source I see no problem whatsoever.  If/when the sources are available in a git repo, another user basically could just clone that repo, install docuwiki and have an exact copy of the site.

Divergence is a feature, not a problem. This is the idea with open source, that users can modify the software (and docs) according to their own requirements. If the change are good, they will send the changes as patches so we can merge them (if we want to).

For now, I'm focused on publishing the sources. The next step, to create the html pages, is another story which might or might not work. With the sources in git you can at least easily try the export process on another host where you have better control over both hardware and software (and without disturbing the official site).

rgleason, 2018-08-16:
> @rgleason: Back to our thread... would it be possible to put the wiki sources in a git repo without too much hazzle?

Ok,, I guess you mean the dokuwiki files themselves?  That's why you say its the same problem as backup.  

The answer is Yes.  --Sorry a little thick there.  But I need some review by others of what we are doing and want a plan from you as to how you intend to use it.

rgleason, 2018-08-16:
>  With the sources in git you can at least easily try the export process on another host where you have better control over both hardware and software (and without disturbing the official site).

Very true and consistent insofar as the use of github. I think I can do it, but I need some affirmation and review by others.

I could make a new repository "OpenCPN Manual" under my git but I think Dave would want it more central. I can do the download and upload if it is to my repository.

rgleason, 2018-08-16:
Here is an [offline.sh ](https://www.dokuwiki.org/tips:offline-dokuwiki.sh)which uses the indexmenu plugin, which we use extensively. I believe the TOC above would work if this works.

rgleason, 2018-08-16:
Here is an example with debian
https://forum.dokuwiki.org/thread/15440

Which raises the question, how does this copy/backup work with respect to security and passwords? We do have manager/editors who have restricted access to edit, add, change and delete. Also with regard the the admin security. Is it hackable?

leamas, 2018-08-16:
> I could make a new repository "OpenCPN Manual" under my git but I think Dave would want it more central. I can do the download and upload if it is to my repository.''

In the end, yes. But for now we just need some testing ground, and it could IMHO basically be anywhere as long as it works.

> Which raises the question, how does this copy/backup work with respect to security and passwords? We do have manager/editors who have restricted access to edit, add, change and delete. Also with regard the the admin security. Is it hackable?

Good catch. We cannot publish permissions, passwords, users etc. We need to exclude this info from the git repo, and make  sure that a replicated copy works  without any restrictions. It will typically run on a local docuwiki without web access, so that should be fine.  If a user wnats to publish it, she will need to define her own permissions and users.

rgleason, 2018-08-16:
I've written Dave, and asked about the security and passwords on the dokuwiki forum under server. Hopefully they will tell me which files to look for.

Unless we come up with a better solution, there must be a clear understanding that PR and changes to the Github version are just that. They will not be transferred be me or anyone automatically to the official OpenCPN Manual.  These changes if needed, will have to be done by Manager/Editors of the official Opencpn Manual and we are not going to be monitoring the changes to the github version in detail.

In fact, it may get overwritten at the time of the next release. So any important changes really must be made in the opencpn.org version by manager/editors!

leamas, 2018-08-16:
So, if you are using the auth:plain backend adding *conf/users.auth.php* to .gitignore should make the trick with respect to users and passwords.

Likewise, adding *conf/acl.auth.php,* to .gitignore should remove all ACL:s on the replicated repo if I git it right.

rgleason, 2018-08-16:
I could make you and Dave collaborators so that you can accept PR and update that repository yourself.

leamas, 2018-08-16:
That would be nice... but I would be careful exercising that right. My immediate plan is to clone the repo once it's available and see what can be done with my "own" copy.

rgleason, 2018-08-16:
> So, if you are using the auth:plain backend adding conf/users.auth.php to .gitignore should make the trick with respect to users and passwords.
Ok, we are using auth:plain but we would be creating a repository, then making it local, then downlooading via FTP all the files. If I delete that file conf/users.auth.php will dokuwiki work? What do I do with it? gitignore is a github thing. 

Likewise, adding conf/acl.auth.php, to .gitignore should remove all ACL:s on the replicated repo if I git it right.
So would I simply delete conf/acl.auth.php 

Then copy all the files into the local repository and then push to origin?

Still need Dave's input.

leamas, 2018-08-16:
>In fact, it may get overwritten at the time of the next release. So any important changes really must be made in the opencpn.org version by manager/editors!

Fair enough. That's how open-source development works - those who owns the project merges, rejects or ignore PR's as they please.

rgleason, 2018-08-16:
Leamus would you like to be a editor/manager of Opencpn Dokuwiki? I would need your email.. via the forum or whatever.(not here) to send the PW. It may be useful, but I realize you have a different focus.

leamas, 2018-08-16:
>So would I simply delete conf/acl.auth.php Then copy all the files into the local repository and then push to origin?

I would do something like:
    
    $ cd /wherever/docuwiki/data
    $ echo conf/acl_auth.php > .gitignore
    $ echo conf/users.auth.php  >> .gitignore
    $ git add .
    $ git commit -am "Initial release"
    $ git ls-files    # check that acl_auth.php and users.auth.php are not listed
    $ git remote add origin https://my-repo-address
    $ git push --set-upstream origin master

rgleason, 2018-08-16:
>  If you want to save some space you should at least backup the following directories:
> 
>     data/pages - contains your current pages
>     data/meta - contains meta information about your pages (like who created it originally, who subscribed to it, …)
>     data/media - contains your current media (images, PDFs, …)
>     data/media_meta - meta data for the media
>     data/attic - all the old versions of your pages
>     data/media_attic - all the old versions of your media
>     conf - the configuration settings
> 
> You can exclude the cache, index, locks, and tmp directories. 
> 

Dokuwiki Package Manager layout is a little different.
https://www.dokuwiki.org/install:debian_ubuntu_extras

rgleason, 2018-08-16:
Thanks Leamas, that is going to really help me, as I am not a native speaker of git yet and this is a little different. I can do routine stuff.

I have backedup and will download, then expand.
Need to hear from dokuwiki precisely and Dave before I make the repos.

leamas, 2018-08-16:
> Thanks Leamas,

You are welcome :)

> I have backedup and will download docuwiki, then expand.

Which is fine while testing. However, once it works I'd suggest you run the same commands directly on opencpn.org, so the files used by the main docuwiki instance is under git control. This creates an easy path if you want to merge changes proposed by others.

Running the commands above will not affect any files. That said, testing on a separate tree before messing with the real source is of course good.

leamas, 2018-08-16:
> If you are backing up anything confidential (e.g., private pages, or configuration files that include account passwords), you should set the backup namespace to be viewable only by administrators, or delete your backup files immediately after creating them.

What we do here is that we list them in .gitignore. This means that no backup is taken on these files, so no-one can see them.

rgleason, 2018-08-16:
There are more files, I believe. See https://forum.dokuwiki.org/post/62083;nocount
What I don't know is if I should just delete those files and dokuwiki will recreate them or if I should remove all content associated with user names and PW.

Also how to distinguish between private files and public is a problem.
Never done this so it is different.

leamas, 2018-08-16:
> There are more files, I believe. See https://forum.dokuwiki.org/post/62083;nocount. What I don't know is if I should just delete those files and dokuwiki will recreate them or if I should remove all content associated with user names and PW.

The .dist files could certainly be removed. This extension is commonly used for templates. The other files you should _not_  delete - just make sure they not goes into the git repo. That is what .gitignore is about. As long as they are not in the repo they will not leave the machine when you push.

> Also the basic configuration needs to be set to be open to users to edit.

As I understand the docs this will be the case since the replicated repo will lack any ACL records

> Also how to distinguish between private files and public is a problem. Never done this so it is different.

I have no idea what data which could be considered private - I havn't seen it...

> This is getting involved. Not going to get it done this morning.

Hey, take your time. In Sweden we say "there is no cow on the ice", the meaning to be understood by anyone. We have a mutual understanding what we are about to do; that's a great step forward.

rgleason, 2018-08-16:
There is a problem with this plan I think. 
There seems to be 500mb and there are php files. Aren't these considered to be binary executables?
Git does not like that!

rgleason, 2018-08-16:
Holey Cow!
This is going to use up the 1gb bandwidth and storage limit and require purchase of a data package.
https://help.github.com/articles/about-storage-and-bandwidth-usage/

I have a free opensource public account.
large binary files are to be uploaded under releases.
I don't want to upload this large a TAR file to Releases.

leamas, 2018-08-16:
There is a problem with this plan I think.
> There seems to be 500mb 

That's a lot... and sort of requires an explanation.  If your page count is OK the total documentation is about 150 pages; that is about 3MB per page. It doesn't make sense.

If you log in to the server, move to the data directory and run

    $ du -sh *

What's the output (assuming a linux host...)?

> and there are php files. Aren't these considered to be binary executables?

Storing executable php files in a git repo should not be a problem.

splitbrain, 2018-08-16:
Hi guys, DokuWiki author here. I came across this issue via some threads  @rgleason posted in the DokuWiki forum. I think you're on the wrong track here and should back up a few steps again. Here's what I understand:

* @leamas is packaging your software for Debian, that part seems to work fine
* this Debian package should also include the documentation for your software
* your documentation is maintained within a DokuWiki now, but used to be in a Drupal system
* there used to be some way to export HTML from the Drupal system that was then included in the Debian package
* that way is broken (and maybe always was)
* @leamas needs a new way to include the docs into the package

What I don't understand is how the suggestion to copy or sync the DokuWiki based documentation to a github repository would solve the above problem.  It seems @leamas was under the assumption that DokuWiki uses Markdown syntax and maybe thought if he had access to the source he could simply compile the HTML from markdown and having that in a github repo would make that easier...

However DokuWiki syntax is **not** Markdown. So I don't see any advantage of copying the docs to github. It will only give you headaches of syncing content changes back from github to the wiki and solves none of your initial problems.

What you guys need to figure out is a way to export the docs from the wiki into a format that you can ship with your software. I assume static HTML would be the way to go. There are a few DokuWiki plugins that may help you with that. You could also have a look at the XMLRPC API offered by DokuWiki.

However based on that @rgleason is not a very technical person and @leamas is lacking access to and experience with your DokuWiki install, I feel like @leamas' suggestion to not ship the docs and instead just link to the wiki might be the easier solution for all involved.

rgleason, 2018-08-17:
Dear Andreas
Thanks for clarifying this and replying. Yes,I am not terribly technical, but I agree with figuring this out first, and I do not intend to create a new repos without that plan, Dave's ok, and I do not want to exceed or disobey github rules/standards.

> static  HTML would be the way to go. 

Yes this is what we use.

From Dokuwiki, Opencpn Manual, we have been using Site Export Plugin to create static html which is packaged with the program. Site Export Plugin does not work well on our server, and terminates and fails without completing properly and it takes a long time. It works better on a local serve, but does  not seem to export our TOC's (created using IndexMenu -which we use extensively), in the manner we expect. so we get into extensive manual editing, under duress to get the release out. 

I wish Site Export worked better, but I will look into XMLRPC API.  This is something we should solve

Thank you.
Rick

leamas, 2018-08-17:
I left my box downloading the opencpn website over the night. It's about 2GB and still counting. The command i used was 
    wget -r -nH https://opencpn.org/OpenCPN/wiki/docuwiki

Looking at the toplevel directory three directories are > 1MB:
   
    1.3G	flyspray
    19M	OpenCPN
    700M	wiki

l attach the 100 biggest files found here. One obvious example are the last ones which seems to be charts in .7z format. Are the files corresponding to these urls part of the repo? 

[biggest-files.txt.gz](https://github.com/OpenCPN/OpenCPN/files/2296119/biggest-files.txt.gz)

I assume the flyspray urls are not significant. Or?

nohal, 2018-08-17:
What are we trying to do here exactly? Can you explain me what real benefit it will have for the user or the project?
There is a lot of stuff on the opencpn.org website that has absolutely nothing to do with the manual, is not at all part of it and definitely does not have to be included in any repository related to the dokuwiki content.
The "big files that seem to be charts in .7z format" are the calibrated US pilot charts, for most of the users currently replaced by https://github.com/chartcatalogs/pilot_kaps and installed using the chart downloader plugin.

leamas, 2018-08-17:
> What are we trying to do here exactly? 

We are exploring ways for users (notable packagers) to rebuild the documentation from sources. From a packaging perspective, we need to distribute sources rather than the result - this is the standard FOSS mechanism.

My first reaction to this was to just drop the documentation from the packages and ship a pointer to the website instead.

However, I decided to explore if there was a way to rebuild the docs. If the documentation was in git there would be a way to handle this both for packagers and for upstream work (test instances etc.)

Now, given all the if and buts, I'm again leaning to my first view: drop the documentation from the packages. It was a try, perhaps worth a bigger effort, but there are certainly more important tasks.

nohal, 2018-08-17:
If we drop the docs and ship a link to the website, both the users without any internet connections and the ones connected via satellite links will sure start jumping 3 meters high out of happiness. A total no go.
I still have an inevitable feeling we are trying to resolve a problem that does not exist. Yes, the HTML exported from docuwiki is not exactly beautiful, but that's all that is happening here.

The docuwiki is the "git" for the OpenCPN documentation. It also allows people with limited technical skills to contribute to the manual.
I also am not exactly happy even thinking I would have to implement the documentation generation in the build process on Windows.

leamas, 2018-08-17:
> f we drop the docs and ship a link to the website, both the users without any internet connections and the ones connected via satellite links will sure start jumping 3 meters high out of happiness.

yes, I wrote the same besides you colourful language ;)

> I still have an inevitable feeling we are trying to resolve a problem that does not exist. 

The problem is simple: The generated docs are not acceptable in a debian (fedora...) package just because they are generated. Linux packages are built from source. This is obvious when it comes to programs, but the same applies to docs. 

Possible work-around: A download link for the static HTML pages on the website?

rgleason, 2018-08-17:
> There is a lot of stuff on the opencpn.org website that has absolutely nothing to do with the manual, is not at all part of it and definitely does not have to be included in any repository related [...]

The pilot charts were a part of the the Drupal wiki. We continued that and then Kathi added them to the website I believe.  Perhaps we should just link to the website and delete them.

rgleason, 2018-08-17:
The documentation can be downloaded from this page
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started

If you think it would be best to move this to an opencpn..org/downloads/documentation page I would support that.  
Then Leamas can reference or link to it more easily.

I would also support providing the documentation for all the external plugins as a package, etc.

rgleason, 2018-08-17:
Pavel
> So, from an "official" packaging view there are two alternatives: 
> - Drop the docs, ship link to opencpn website.
> - Ship the sources and rebuild the status html pages as part of package build process on e. g. debian builders.
> 

Why not separate the documentation and have a last message with a link as part of the install?

1. Presumably the user has an internet connection as they are installing.
2. Also have a note at the top of the downloads page that documentation is separate.
3. If they are just downloading the program without installing, they would read the message hopefully.

Then have a downloads/documentation section.

It would simplify the process a lot I think, because documentation could follow release!
The release would not have to be stalled by documentation prep!
The documentation could be fully updated and be updated on the website separately and following the release. There are often things that need to be added to the documentation at the last minute, and yet we are also exporting to static html and then editing.

leamas, 2018-08-17:
> The item that will cause us the most effort is GDAL. We have a nearly verbatim part of the GDAL source tree embedded in our tree, since the GDAL top-level library API does not satisfy our needs. 

There has been some water under the bridges since that took place (I have seen at least public parts of discussions on this). Having patched the library is a perfectly sane reason to bundle it, and I don't think  this should be a blocker today (Fedora has been even more strict on this, but even there this should not be a problem)

I have tried to collect things to get it packaged in #1113. It's a huge list, but doable. For now, the real blocker is the licensing which needs an update (PR in #1100, but have an pending update)

rgleason, 2018-08-17:
I have a local [working copy of the wiki now on Windows](https://forum.dokuwiki.org/post/62102;nocount)
I am going to try some of the SiteExport techniques locally, to see if our 

- [OfflineOpenCPN Manual  TOC 87pages](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toc_offline_user_manual)
- [Offline Extended Plugins TOC 46 pages](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toc_offline_plugins_manual)

Table of Contents is working properly on a local machine.

bdbcat, 2018-08-17:
leamas...

Got a link to the discussion of "libraries in release" ?

Trolls:  I think Pavel will agree. Reading the rules without reading the code, and not thinking about real-life user needs, does not match up with my objectives for the Project.

Dave

I need direction on which commits to take on #1100.

leamas, 2018-08-18:
This is my bad from the beginning, but let's try to re-focus on the original issue in this bug:  can we all agree on handling  this by letting users of packaged versions download static manuals form the website?

In this context, I'd appreciate if the download link was a bit more  visible as proposed by @rgleason: 

> f you think it would be best to move this to an opencpn..org/downloads/documentation page I would support that. Then Leamas can reference or link to it more easily.

leamas, 2018-08-18:
@bdbcat  wrote:
> Got a link to the discussion of "patched libraries in release" ?
No. But lets move this to #1113, will restart discussion there.

bdbcat, 2018-08-18:
I am in general agreement. Let us get to specifics:

1. for linux, the doc is already in a separate installable package.  linux users can install it like any other package, or omit from the base install if desired.

2. For other packages, we could strip the docs from the downloadable installer, if we want.  Most Windows users would need a real installer to later get the (maybe updated)docs, and install them in the right place.  Otherwise, they would likely have multiple confusing copies.

3.  What exactly would the "more prominent" link on the website point to?  How would a user use it?

leamas, 2018-08-18:
> for linux, the doc is already in a separate installable package. linux users can install it like any other package, or omit from the base install if desired.

Yes, the cmake/CPack packages. But for users of an official package this won't work for reasons described above. Flatpack packaging would also be simpler if we could exclude the docs.

> For other packages, we could strip the docs from the downloadable installer, if we want. 

For me, this could be either way. I note that @rgleason  seems to prefer stripping it, basically decoupling the docs release cycle from the main package for better or worse. 

> What exactly would the "more prominent" link on the website point to? How would a user use it?

I'm not a native speaker, so the nuances are perhaps not what I intended. Anyway, I envision:

  - A link which can be kept stable e. g. like https://opencpn.org/download/manual-latest.
  - A simple page in the distributed package with a link to the online docs and some instructions how to download and use the off-line ones.
  -  That the downloadable manual is visible at https://opencpn.org/OpenCPN/info/downloads.html

leamas, 2018-08-20:
@bdbcat  Clarification: #1100 is rewritten and ready, see notes in bug.

bdbcat, 2018-08-21:
OK, time to finish this.

1.  We have added a link to the website to download standalone HTML doc files, zipped.  It is not especially current, but that depends on the wiki plugin tool development, proceeding elsewhere.  Not a factor for this discussion.

2.  On the distributions:

a.  Standard Distributions will all include the embedded documentation.  All distros, always.
Logic:  Users often need access to docs in remote places.  If they have not forethought to download the separate doc zip from elsewhere, and they have not good internet access, they are stuck.  We must think for them....

b.  If there are files in the linux distro that have issues with privacy, licensing, etc.  which make them not appropriate for a debian package, let us identify them, one-by-one, and fix or remove them,

So, let us see a PR that fixes 2(b) above, and move on.

Dave

leamas, 2018-08-21:
> We have added a link to the website to download standalone HTML doc files, zipped.

Great. It's not ideal, but perhaps the best we can do, because:

>  If there are files in the linux distro that have issues with privacy, licensing, etc...

This is basically each and every HTML file, see the build log in top comment. The privacy issue could probably be patched, it's just to exclude some javascript (which "probably" doesn't break anything badly). But the main issue is that the files are generated, and a proper linux packaging doesn't accept generated files - it want the sources, and a process to produce the docs from them. 

And this is the path we have tried and abandoned.

Let's leave this bug open for now. With the licensing and the zipped manual it's possible to create a working package. The main issue now is #1124. I will drop a question on the ML on this, it's a possible blocker. If you have the time to review what I have written there it would be really nice.

All this said, if we could make the docuwiki sources (only those) in a usable git repo it would create a usable testing ground for the document generation.  Low prio, but IMHO not zero.

[2 more replies]

closed 2018-08-21 by leamas