OpenCPN/OpenCPN #1124: Packaging: Bundling issues

leamas, 2018-08-15:
Source contains some bundled libraries. These are required to build on Windows and possibly also MacOS, but is a problem on linux distros where the system libraries should be used.  Together with the licensing (#1100) this is a major packaging headache (#1113).

This is partly resolved by using system libraries where they exist. However, here are pending issues

  - Debian needs a source tarball without the bundled libs. Fixes in #1100, but not complete.

Packages to be identified and used if available (and excluded from copyright file):

  - texcmp/lz4 is from https://github.com/lz4/lz4. It's packaged in Debian and Fedora.
  - texcmp/squish is available in Debian.
  - The wxsvg library is not identified, probably because it's not in Debian. It does exist in other distros, notably Fedora, and should be used when available. Prior to packaging for Debian, the wxsvg dependency needs to be packaged
  - The garmin/jeeps is from gpsbabel which is available in Debian and Fedora. It should be used if found at the system level, unless it's patched.

Some bundled libs could probably be left as-is:
 
  - glshim is from https://github.com/lunixbochs/glshim. Not available in Debian or Fedora, could perhaps be left.
  - mygdal is from the Gnu CPL. This is specifically designed to be bundled.
  - iso8211 seems to lack established upstream.

bdbcat, 2018-08-22:
Leamas...
Ok, thanks for the research.  Points:
1.  We should proceed with the easy stuff identified above.  I will look favorably on PRs which implement those changes.
2.  Having done some or all of (1), we should re-engage with a debian sponsor.
3.  Let us separate the flatpack effort into a new thread, since it is in some ways easier to accomplish in the short term, and we do not require the same level of auditing.

Regarding debian:
Recalling the previous interaction with debian mentor,  I will just state that I shall have little patience with mentors that disapprove of runtime structure (e.g. private directory locations), or implementation content (e.g. icon bitmap format selection).  This should not be their business.  Their business should be licensing, and encouraging reasonable use of available and useful dependent packages. If there is no defined upstream nor functional (dependent) debian package, we shall use what we need to, subject to compatible license (e.g. GDAL, NME0183, jasper, etc)..  If that means we can have no official debian package, so be it.

transmitterdan, 2018-08-22:
Amen brother!

leamas, 2018-08-22:
Actually, my plan is to drop a message on the debian mailng list to get their view. I have prepared some unbundling PRs, but we need to know there is a reasonable way to actually fix this if  we should undertake this effort. Also, the mailing list has more eyes on it than a single reviewer. Stay tuned.

rgleason, 2018-08-22:
Alec, thank you for your perseverance. It is not a simple task!

leamas, 2018-08-22:
I fully respect the difficulties. Let's see what's possible given the overall time and skills available.

leamas, 2018-08-22:
>  Let us separate the flatpack effort into a new thread, since it is in some ways easier to accomplish in the short term, and we do not require the same level of auditing.
Well, there is a PR in #1115. Discuss there?

leamas, 2018-08-23:
Hm... debian-devel thread at https://lists.debian.org/debian-devel/2018/08/msg00333.html.

Basically, I think that  since we can refer to Paul Wise and Sean Whitton in our position that bundling might be motivated we could proceed. It's certainly not a carte blanche, we should still do our homework with the low hanging fruits. But it's a recognition of that bundling sometimes is necessary.

leamas, 2018-08-23:
So, let's start. Filed ITP bugs for opencpn and the dependencies:

opencpn: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=907065
wxsvg: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=907068
unarr: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=907070
wxcurl: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=907071

In this bug the focus is the dependencies. Had to file opencpn to be able to refer to it.

leamas, 2019-04-03:
Package is now in the new queue [1]. Things might pop up in this final review, keeping bug open.

[1]  https://ftp-master.debian.org/new.html

leamas, 2019-09-07:
After six months we got a auto-reject from the ftp-masters. Sigh. Package resubmitted, no idea when it will be processed

rgleason, 2019-09-07:
Very Sorry about that Leamas. Hope they gave their reasons.

leamas, 2019-09-07:
Basically, they don't.  

We can't do anything to affect this, so let's stick to what we can do instead.

rgleason, 2019-09-07:
This is perseverance!  Thank you.

leamas, 2019-11-16:
Package is accepted. Here are still bundling issues, but they are by definition solved "enough" to be accepted into Debian. Closing.

closed 2019-11-16 by leamas