OpenCPN/OpenCPN #1126: Crash issue when connected to radar

bladedemon, 2018-08-21:
When I connected my radar's output of ARP table to OpenCPN via serial port, after a short time (5 to 10 minutes, depending on the number of tracked targets) the program hard-crashed. I had turned on the email report feature, but to no avail, as the OS (Windows 7 Home Premium) reported that the program stopped working. I tried to turn on/off various features to pinpoint the cause, and it turned out to be the radar input.

The Windows report (this happened to Windows 7 Home and to Windows 10 Professional) named the module StackHash_XXXX as the problem (the 4 letter code changed, from 1780, to 80bb, to 573e).

The problem persisted from version 4.2, to 4.8, to 4.8.6 .

The OpenCPN logfile didn't yield any errors. I tried to change the source of the radar tracked targets by re-routing from the serial port to a TCP port but the problem persisted. The program runs smoothly only when the radar input is disabled completely.

The reported exception code is c0000005 and the exception data is 00000008 (that is a constant between all the crash reports). The excpetion offset varied (10ff21ff, 10ffffff).

The radar in question is a BridgeMaster, and it outputs the table of tracked targets in NMEA 183 format every 2 seconds. OpenCPN accepts the input, but after a while the crash occurs.

bdbcat, 2018-08-21:
We need a copy of the NMEA ARP table datastream fed to OpenCPN, long enough to induce a crash.
Can you capture this and post here?

Dave

bladedemon, 2018-09-05:
Greetings and sorry for the delayed reply, I tried to collect some data.
The first file is the NMEA data collected with the NMEA Recorder plugin, up to the crash. (nmea_log_opencpn7.txt)
The second is the NMEA data collected with the same plugin, but I had also connected a NAVRAD radar (it is the auxiliary radar) along with the main Bridgemaster radar (nmea_log_opencpn8.txt)
[nmea_log_opencpn7.txt](https://github.com/OpenCPN/OpenCPN/files/2352340/nmea_log_opencpn7.txt)
[nmea_log_opencpn8.TXT](https://github.com/OpenCPN/OpenCPN/files/2352342/nmea_log_opencpn8.TXT)
For the first case, the crashed module was StackHash_573e, offset 10ff16ff, exception code c0000005 and exception data 00000008.
For the second case, the crashed module was StackHash_e98d, offset ff010104, exception code c0000005 and exception data 00000008.
I may be paranoid but exception code c0000005 is for access violation, and this thing strongly reminds me of a buffer overflow. I know that the StackHash module is not in the OpenCPN code, so it might be out of your scope, but still....
On regard of this, I ran OpenCPN on my laptop (Ubuntu 16.04) and the application also crashed, albeit later (my laptop has 8GB memory and Pentium i7 2.3GHz) but I couldn't extract the first error report, on account on being third party software. I ran it via gdb and I got the following listing and error:
[LinuxGDBError.TXT](https://github.com/OpenCPN/OpenCPN/files/2352379/LinuxGDBError.TXT)
I intend to install some tools (like ProcessHacker and such) to try to get a better understanding.
Thanks in advance.

rgleason, 2021-04-23:
Shouldn't this be closed by now?  Or is it useful somehow for confirming future bugs? (In which case, it is still available when closed)

douwefokkema, 2021-04-23:
Rick, your suggestion to close this seems fine. No one has ever complained about this again. Can you close it?

closed 2021-04-23 by rgleason