OpenCPN/OpenCPN #1131: Documentation: Document generation is not open

leamas, 2018-08-26:
There is no way for external users to modifying the documentation sources. Practically, this leads to two related problems:

  - There is no way for users to provide documentation patches (like for code). 
  - Linux distros generally frowns upon generated stuff and often refuses to package it, basically 
    for the same  reason - users cannot modify in any meaningful way.

This is really the underlying, still actual  problem in  #1108 (closed after a work-around).
@rgleason, @bdbcat

nohal, 2018-08-26:
Then the linux distros are free to ship OpenCPN without any documentation and link to the web (Which we natively support in case the documentation is not available) in case they consider it a better solution.
The users provide "patches" by editing the wiki. I really don't see where the real difference is if we will implement it based on some git workflow, which will just be more complicated for them, but yields the same result: People being able to contribute to the docs.

rgleason, 2018-08-29:
There are 50 editors of the wiki. The previous open drupal wiki was subject to hacking.  Document generation is open, but is not easy.  Single page pdf are easy, full document is very difficult (which is a technical hurdle).  Use of Epub seems to be working pretty well, not seamless, but much better.

rgleason, 2018-08-30:
There is a new Epub Book format for all of the [OpenCPN User Manual 4.8.6.epub](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started) which can be downloaded. (20mb)

rgleason, 2018-09-01:
There is no way for external users to modifying the documentation sources. Practically, this leads to two related problems:

    There is no way for users to provide documentation patches (like for code).
    Linux distros generally frowns upon generated stuff and often refuses to package it, basically
    for the same reason - users cannot modify in any meaningful way.

This is inaccurate. Known users are permitted to be editor/managers with full capabilities, via their username and a password. As a practical matter, we have perhaps 12 that are editing on a regular basis.

leamas, 2019-04-03:
Closing. Things have improved, and the remaining problems basically are wont-fix.

closed 2019-04-03 by leamas