OpenCPN/OpenCPN #1132: Lack of properly tagged releases in git / signed source tarballs.

MrElendig, 2018-08-30:
Several times now, last with 4.8.6, releases have been made on the download page[[1](https://opencpn.org/OpenCPN/info/downloadopencpn.html)][[2](https://web.archive.org/web/20180829175255/https://opencpn.org/OpenCPN/info/downloadopencpn.html)] without them being tagged in git.  Combined with a lack of (signed/check-summed) source tarball on the download page that makes packaging stable releases of opencpn infeasible. 
Linking to a branch[[2](https://github.com/OpenCPN/OpenCPN/tree/o486)] on GH is not good enough since that doesn't tell us exactly where in the history/from which commits the release was made.

[1]https://opencpn.org/OpenCPN/info/downloadopencpn.html
[2]https://web.archive.org/web/20180829175255/https://opencpn.org/OpenCPN/info/downloadopencpn.html
[3]https://github.com/OpenCPN/OpenCPN/tree/o486

tchiwam, 2018-08-31:
Holding for the version bump in our systems for the same reason.

greno4ka, 2018-10-24:
Ping?

bdbcat, 2018-10-27:
Hello...
I understand the issue.  The point at which branch o486 was forked was not tagged in master.  Subsequent changes were made in branch o486. As a result, the actual release content of o486 is not in master. It only exists in branch o486. So a back-tag of master will not help much today.

Branch o486 is tagged at the proper point, if that helps.

Dave

nohal, 2018-10-27:
Dave...

The tag does not have to be in master. What `o486` branch is missing is the last commit actually changing the version number and then the `v4.8.6` tag.

Pavel

nohal, 2018-10-27:
I have tagged it and created a release.

Closing until next time ;)

closed 2018-10-27 by nohal