OpenCPN/OpenCPN #1137: Pluginmanager loading same PI multiple times

RooieDirk, 2018-09-10:
Using linux with current master the pluginmanager(.cpp) is loading the same PI multiple times if running in portable mode. The directory search string  contains multiple dirs, and if there are multiple PI versions within those dirs is will remove the already loaded, and do a new loading for the version in the next directory.
`13:11:48 CEST: PlugInManager: plugins loading from ~/.local/lib/opencpn:/usr/local/lib/opencpn:/usr/lib/opencpn:/home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins
13:11:48 CEST: PlugInManager searching for PlugIns in location /home/dirk/.local/lib/opencpn
13:11:48 CEST:    Directory /home/dirk/.local/lib/opencpn does not exist.
13:11:48 CEST: PlugInManager searching for PlugIns in location /usr/local/lib/opencpn
13:11:48 CEST: PlugInManager: Loading PlugIn: /usr/local/lib/opencpn/libwmm_pi.so
**13:11:48 CEST: PlugInManager:  /usr/local/lib/opencpn/libwmm_pi.so**
              API Version detected: 108
              PlugIn Version detected: 101
13:11:48 CEST: WMM: WMM model data loaded from file /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins/wmm_pi/data//WMM.COF.
13:11:48 CEST: PlugInManager: Loading PlugIn: /usr/local/lib/opencpn/libchartdldr_pi.so
13:11:48 CEST: PlugInManager:  /usr/local/lib/opencpn/libchartdldr_pi.so
              API Version detected: 113
              PlugIn Version detected: 103
13:11:48 CEST: chartdldr_pi: /home/dirk/Charts
13:11:48 CEST: chartdldr_pi: /home/dirk/Charts
13:11:48 CEST: Opencpn language load for:  en_US
13:11:48 CEST: Loading catalog for:  opencpn-wmm_pi
13:11:48 CEST: Loading catalog for:  opencpn-chartdldr_pi
13:11:48 CEST: Loading catalog for opencpn core.
13:11:48 CEST: Getting DefaultSystemLocale...
13:11:48 CEST: PlugInManager searching for PlugIns in location /usr/lib/opencpn
13:11:48 CEST: PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libwmm_pi.so
13:11:49 CEST: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libwmm_pi.so
1**3:11:49 CEST: PlugInManager:  /usr/lib/opencpn/libwmm_pi.so**
              API Version detected: 108
              PlugIn Version detected: 101
13:11:49 CEST: WMM: WMM model data loaded from file /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins/wmm_pi/data//WMM.COF.
13:11:49 CEST: PlugInManager: Deactivating PlugIn: /usr/local/lib/opencpn/libchartdldr_pi.so
13:11:49 CEST: chartdldr_pi: DeInit
13:11:49 CEST: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libchartdldr_pi.so
13:11:49 CEST: PlugInManager:  /usr/lib/opencpn/libchartdldr_pi.so
              API Version detected: 113
              PlugIn Version detected: 103
13:11:49 CEST: chartdldr_pi: /home/dirk/Charts
13:11:49 CEST: chartdldr_pi: /home/dirk/Charts
13:11:49 CEST: Opencpn language load for:  en_US
13:11:49 CEST: Loading catalog for:  opencpn-wmm_pi
13:11:49 CEST: Loading catalog for:  opencpn-chartdldr_pi
13:11:49 CEST: Loading catalog for:  opencpn-wmm_pi
13:11:49 CEST: Loading catalog for:  opencpn-chartdldr_pi
13:11:49 CEST: Loading catalog for opencpn core.
13:11:49 CEST: Getting DefaultSystemLocale...
13:11:49 CEST: PlugInManager searching for PlugIns in location /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins
13:11:49 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libwmm_pi.so
13:11:49 CEST: PlugInManager: Loading PlugIn: /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins/wmm_pi/libwmm_pi.so
**13:11:49 CEST: PlugInManager:  /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins/wmm_pi/libwmm_pi.so**
              API Version detected: 108
              PlugIn Version detected: 101
13:11:49 CEST: WMM: WMM model data loaded from file /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins/wmm_pi/data//WMM.COF.
13:11:49 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libchartdldr_pi.so
13:11:49 CEST: chartdldr_pi: DeInit
13:11:49 CEST: PlugInManager: Loading PlugIn: /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins/chartdldr_pi/libchartdldr_pi.so
13:11:49 CEST: PlugInManager:  /home/dirk/Development/OpenCPN-Dev/OpenCPN/build/plugins/chartdldr_pi/libchartdldr_pi.so
              API Version detected: 113
              PlugIn Version detected: 103
13:11:49 CEST: chartdldr_pi: /home/dirk/Charts
13:11:49 CEST: chartdldr_pi: /home/dirk/Charts
13:11:49 CEST: Opencpn language load for:  en_US
13:11:49 CEST: Loading catalog for:  opencpn-wmm_pi
13:11:49 CEST: Loading catalog for:  opencpn-chartdldr_pi
13:11:49 CEST: Loading catalog for:  opencpn-wmm_pi
13:11:49 CEST: Loading catalog for:  opencpn-chartdldr_pi
13:11:49 CEST: Loading catalog for:  opencpn-wmm_pi
13:11:49 CEST: Loading catalog for:  opencpn-chartdldr_pi
13:11:49 CEST: Loading catalog for opencpn core.`

nohal, 2018-09-10:
And what should the "correct" behaviour be? Not loading it again at all? Loading just in case it is "newer"?

RooieDirk, 2018-09-10:
As I see it, in a portable mode, only the plugins in the 'portable' plugin location.

leamas, 2018-09-14:
hm... you should be able to work around this using the OPENCPN_PLUGIN_DIRS environment varable documentedi in the new manpage.

Perhaps it could be argued that this should be set to another default value in portable mode..."scratching head".

RooieDirk, 2018-09-16:
A proposal.`diff --git a/src/pluginmanager.cpp b/src/pluginmanager.cpp
index e0f0356..a6e20dc 100644
--- a/src/pluginmanager.cpp
+++ b/src/pluginmanager.cpp
@@ -131,6 +131,7 @@ extern int              g_chart_zoom_modifier;
 extern int              g_chart_zoom_modifier_vector;
 extern double           g_display_size_mm;
 extern bool             g_bopengl;
+extern bool             g_bportable;
 
 extern ChartGroupArray  *g_pGroupArray;
 
@@ -367,13 +368,19 @@ PlugInManager::~PlugInManager()
 
 bool PlugInManager::LoadAllPlugIns(const wxString &plugin_dir, bool load_enabled, bool b_enable_blackdialog)
 {
-#ifdef __linux__
-    const char* const envdirs = getenv("OPENCPN_PLUGIN_DIRS");
-    wxString dirs(envdirs ? envdirs : DEFAULT_PLUGIN_DIRS);
-    if (envdirs == 0  && dirs.Find(plugin_dir) == wxNOT_FOUND)
-        dirs = dirs.Append(_T(":") + plugin_dir);
-#else
     wxString dirs = plugin_dir;
+#ifdef __linux__
+    if ( !g_bportable )
+    {
+        const char* const envdirs = getenv("OPENCPN_PLUGIN_DIRS");
+        if (envdirs != 0) //There are environment dir settings, so use them
+            dirs =  wxString::FromUTF8(envdirs) ;
+        else
+            dirs = wxString::FromUTF8( DEFAULT_PLUGIN_DIRS );
+        
+        if ( dirs.Find(plugin_dir) == wxNOT_FOUND ) //Add plugin_dir to string if not there already
+            dirs = plugin_dir + _T(":") + dirs;        
+    }
 #endif
     wxLogMessage( _T("PlugInManager: plugins loading from ") + dirs);
     bool any_dir_loaded = false;
@@ -382,7 +389,10 @@ bool PlugInManager::LoadAllPlugIns(const wxString &plugin_dir, bool load_enabled
         wxString dir = tokens.GetNextToken();
         dir = ExpandWord(dir);
         if (LoadPlugInDirectory(dir, load_enabled, b_enable_blackdialog))
+        {
             any_dir_loaded = true;
+            break;
+        }
     }
     return any_dir_loaded;
 }
`
This would do:
In portable mode, only check the portable plugindir for plugins.
else If OPENCPN_PLUGIN_DIRS is set, use this, if not use the DEFAULT_PLUGIN_DIRS
As soon as a dir with plugins is found, stop searching for other directories. 

BTW The code as is in current master gives me a crash if trying to enable a plugin from the options window in portable mode.

RooieDirk, 2018-09-18:
> As soon as a dir with plugins is found, stop searching for other directories. _This might be problematic. There are usecases where users have plugins in more than one directory (e. g. , some packaged installs in /usr and some built locally in /usr/local.)

Yes, I do have three different versions installed, one in /usr, one in /usr/local and the one in the build directory. And in my usercase I would like to have each version its own set of PI's

> Do I understand you correctly that there is no crash in "normal" mode? Could you possibly create a stacktrace?

 No also in normal mode.
```
*** opencpn (wxWidgets 3.0.2) crashed ***, see backtrace!
0x822de1 in wxCrashPrint::Report() at crashprint.cpp:84
0x7fd0a3db290c in  at ??:0
0x7fd0a1e07390 in  at ??:0
0x7fd0a3ddafbe in wxEvtHandler::DoUnbind(int, int, int, wxEventFunctor const&, wxObject*) at ??:0
0x7fd07c9b18e2 in chartdldr_pi::OnCloseToolboxPanel(int, int) at ??:0
0x7526f7 in PlugInManager::CloseAllPlugInPanels(int) at pluginmanager.cpp:1746
0x715fbf in options::OnApplyClick(wxCommandEvent&) at options.cpp:6485
0x716b68 in options::OnXidOkClick(wxCommandEvent&) at options.cpp:6516
0x7fd0a3c541fe in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const at ??:0
0x7fd0a3dda6e7 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) at ??:0
0x7fd0a3dda7db in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) at ??:0
0x7fd0a3ddab8b in wxEvtHandler::TryHereOnly(wxEvent&) at ??:0
0x7fd0a3ddac13 in wxEvtHandler::ProcessEventLocally(wxEvent&) at ??:0
0x7fd0a3ddac75 in wxEvtHandler::ProcessEvent(wxEvent&) at ??:0
0x7fd0a4501de8 in wxWindowBase::TryAfter(wxEvent&) at ??:0
0x7fd0a3dda9e7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) at ??:0
0x7fd0a4362b57 in  at ??:0
0x7fd0a62fe1d4 in  at ??:0
0x7fd0a63189a6 in g_signal_emit_valist at ??:0
0x7fd0a631908f in g_signal_emit at ??:0
0x7fd09c869f35 in  at ??:0
0x7fd0a62fdfa5 in g_closure_invoke at ??:0
0x7fd0a630fafc in  at ??:0
0x7fd0a6318d5c in g_signal_emit_valist at ??:0
0x7fd0a631908f in g_signal_emit at ??:0
0x7fd09c868e79 in  at ??:0
0x7fd09c90faec in  at ??:0
0x7fd0a62fdfa5 in g_closure_invoke at ??:0
0x7fd0a631056e in  at ??:0
0x7fd0a63187f9 in g_signal_emit_valist at ??:0
0x7fd0a631908f in g_signal_emit at ??:0
0x7fd09ca2793c in  at ??:0
0x7fd09c90e284 in gtk_propagate_event at ??:0
0x7fd09c90e63b in gtk_main_do_event at ??:0
0x7fd0a5926c8c in  at ??:0
0x7fd09fc7d197 in g_main_context_dispatch at ??:0
0x7fd09fc7d3f0 in  at ??:0
0x7fd09fc7d712 in g_main_loop_run at ??:0
0x7fd09c90d697 in gtk_main at ??:0
0x7fd0a42eced5 in wxGUIEventLoop::DoRun() at ??:0
0x7fd0a3c96353 in wxEventLoopBase::Run() at ??:0
0x7fd0a436fcdd in wxDialog::ShowModal() at ??:0
0x51e0f6 in MyFrame::DoOptionsDialog() at chart1.cpp:5778
0x51e719 in  at chart1.cpp:4705
0x521247 in MyFrame::OnToolLeftClick(wxCommandEvent&) at chart1.cpp:4699
0x7fd0a3c541fe in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const at ??:0
0x7fd0a3dda6e7 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) at ??:0
0x7fd0a3dda7db in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) at ??:0
0x7fd0a3ddab8b in wxEvtHandler::TryHereOnly(wxEvent&) at ??:0
0x7fd0a3dda993 in wxEvtHandler::DoTryChain(wxEvent&) at ??:0
0x7fd0a3ddac75 in wxEvtHandler::ProcessEvent(wxEvent&) at ??:0
0x7fd0a4501de8 in wxWindowBase::TryAfter(wxEvent&) at ??:0
0x7fd0a3ddb6fa in wxEvtHandler::ProcessPendingEvents() at ??:0
0x7fd0a3c56317 in wxAppConsoleBase::ProcessPendingEvents() at ??:0
0x7fd0a42ccac2 in wxApp::DoIdle() at ??:0
0x7fd0a42ccbb3 in  at ??:0
0x7fd09fc7d04a in g_main_context_dispatch at ??:0
0x7fd09fc7d3f0 in  at ??:0
0x7fd09fc7d712 in g_main_loop_run at ??:0
0x7fd09c90d697 in gtk_main at ??:0
0x7fd0a42eced5 in wxGUIEventLoop::DoRun() at ??:0
0x7fd0a3c96353 in wxEventLoopBase::Run() at ??:0
0x7fd0a3c5b666 in wxAppConsoleBase::MainLoop() at ??:0
0x7fd0a3ce7f70 in wxEntry(int&, wchar_t**) at ??:0
0x4bb838 in main at chart1.cpp:930
Aborted (core dumped)
```

leamas, 2018-09-18:
>Yes, I do have three different versions installed, one in /usr, one in /usr/local and the one in the build directory. And in my usercase I would like to have each version its own set of PI's

Yes, but we shouldn't hardcode your usecase. Again: with OPENCPN_PLUGINS_DIRS set to a single directory, this is the only directory used for plugins.  Or at least should be. Could you please test this to get us some common ground?

BBL on the crash...

michael-kolmodin, 2018-09-19:
> Note the env setting is not used.

And this is the core issue, for sure. Looking at the code, I cannot see any path where the variable is defined but not used (can you?):

       #ifdef __linux__
        const char* const envdirs = getenv("OPENCPN_PLUGIN_DIRS");
        wxString dirs(envdirs ? envdirs : DEFAULT_PLUGIN_DIRS);
        if (envdirs == 0  && dirs.Find(plugin_dir) == wxNOT_FOUND)
            dirs = dirs.Append(_T(":") + plugin_dir);
      #else
        wxString dirs = plugin_dir;
      #endif
      wxLogMessage( _T("PlugInManager: plugins loading from ") + dirs

It's like it's not visible for the process running opencpn. What if you do a "export OPENCPN_PLUGIN_DIRS=..." instead?

leamas, 2018-09-19:
> Note the env setting is not used.  

And this is the core issue, for sure. Looking at the code, I cannot  see any path where the variable is defined but not used (can you?):    

     #ifdef __linux__     
     const char* const envdirs = getenv("OPENCPN_PLUGIN_DIRS");
     wxString dirs(envdirs ? envdirs : DEFAULT_PLUGIN_DIRS);
     if (envdirs == 0  && dirs.Find(plugin_dir) == wxNOT_FOUND)
          dirs = dirs.Append(_T(":") + plugin_dir);
     #else
     wxString dirs = plugin_dir;
     #endif   
     wxLogMessage( _T("PlugInManager: plugins loading from ") + dirs  

It's like it's not visible for the process running opencpn. What if you do a "export OPENCPN_PLUGIN_DIRS=..." instead?

RooieDirk, 2018-09-19:
Found something. the right syntax is
  `env OPENCPN_PLUGIN_DIRS="..."  opencpn` 
Then the env-var is used, and the env-var only.
Without env-var the DEFAULT_PLUGIN_DIRS + plugin_dir is used, but **all** directorys are searched for plugin one by one. And if a PI is found that was loaded before, that one is unloaded and loaded again. 

My thoughts for a portable mode use **only** the plugin_dir. Normal mode first look in the plugin_dir, and afterwards in DEFAULT_PLUGIN_DIRS, and if a PI is found that was loaded already, skip it. (Don't reload)(reloading only spills a lot of time and it is confusing what version was loaded at last)

RooieDirk, 2018-09-21:
> But: it should be consistent. I'd say that the expected behaviour would be that the first directory wins

If it should be consistent, then what is the reason to have a different behaviour on Linux compared to Windows??

leamas, 2018-09-21:
Because this is how things on linux is supposed to work. All sane linux (or really, unix) applications have separate locations for system (i. e., packaged), site and user configuration by default based in /usr, /usr/local and $HOME. 

From a practical point of view, users should be able to add and maintain plugins without having to mess with directories in /usr which are handled by package managers. Also, packaging systems like flatpak requires this to be in place for technical reasons (long story...)

Windows is certainly different, with no clear standardized way to handle site and/or user modifications (and no package manager).

Bottom line: *consistent* in this context is more about behaving like other linux applications than behaving the same way on all platforms.

bdbcat, 2018-09-21:
leamas...

Be careful here.  This sort of dogmatic insistence on the "linux" way is IMHO one of the reasons that linux struggles to gain market share among mainstream PC users.

So, we should arrange that things just work as expected out-of-the-box, with advanced options available to sophisticated users.  On all platforms.

leamas, 2018-09-21:
> So, we should arrange that things just work as expected out-of-the-box.

Agreed. And the ultimate out-of-the-box linux experience is a graphical or CLI install from the distro repos, like apt-get install  in the debian/ubuntu case.  Or a similar flatpak experience on almost all distros.

For this to work as expected we need to move local and user modifications out of the way for the package-manager controlled areas under /usr. It's not about dogmatism, it's just about being practical and using well-tested configuration concepts. Of course, it should also work as expected after a 'make install' - I have tried to have reasonable defaults set up for this.

The core issue here is an oversight: I didn't consider the "portable mode" usecase enough. Basically, we now have more buttons to push, but we need to define how to push them in this case. I'm pretty confident that OPENCPN_PLUGIN_DIRS does the job as for plugin loading, but a bit concerned how to handle XDG_DATA_DIRS (plugin data directory lookup). 

First step would be to verify that portable mode works with correct settings - next step would be to create a more streamlined UX.

RooieDirk, 2018-09-21:
After a bit of searching on the www it seems that the getenv() methode also works for windows and macOS. This code works for me but I can't test for Win an macOS. And I don't think it is bad that a environment var is also working for the portable mode.
```
bool PlugInManager::LoadAllPlugIns(const wxString &plugin_dir, bool load_enabled, bool b_enable_blackdialog)
{
char* envdirs = 0;
#ifndef __OCPN__ANDROID__
    envdirs = getenv("OPENCPN_PLUGIN_DIRS");
#endif    
    wxString dirs(envdirs ? wxString( envdirs ) : plugin_dir);
```For Android I don't consider it to be bad that you can't set the environment.

RooieDirk, 2018-09-22:
In portable mode you can set and use  the OPENCPN_PLUGIN_DIRS. And it works. Maybe even more important, I don't have a crash anymore after the option/plugin page.(Altough I have no clue why it did before) And yes it should be tested for Win and Mac.

leamas, 2018-09-22:
OK, this is great. However, you are still using the global data directories in this setup. Assuming that you are in the new directory described in [1], could you please try something like

    $ mkdir opencpn
    $ ln -s opencpn/plugins $PWD
    $ env OPENCPN_PLUGINS_DIRS=$PWD/plugins XDG_DATA_DIRS=$PWD ./lopencpn

and report back? 

[1] https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:portable_opencpn

rgleason, 2019-08-25:
Leamas, I've read this issue, which I had not seen, and may have some time in the next week or so to try this with Windows, to give you feedback.
I know you gave instruction before, but what is the link to download or compile this version of O. Also are the PI listed in the manager now, ready for installation? Or is this done through a new Installmanager PI?
IE is there some other resource to install in order to test?

leamas, 2019-08-25:
This issue should now be fixed.

The plugins should be available after building and running the PR. It's automagical, behind the scenes plugins are downloaded from https://bintray.com/leamas/OpenCPN/plugins#files

The PR does not contain any specific Installmanager PI. Installation/updates are part of main OpenCPN.

Note that the issue here is "long time ago". However, if you have time to test the PR at https://github.com/OpenCPN/OpenCPN/pull/1390 it would  really interesting.

leamas, 2019-08-25:
ah... sorry, I answered the wrong question.

There is no specific version on this point, the new plugin loader is merged into 5.0.0.

rgleason, 2019-08-25:
Leamas, is there a link to the actual commit? Sorry for being so thick about this, but perhaps I can just download the appropriate appveyor created file, but I have a hard time believing it is the 5.0 release version.

leamas, 2019-08-26:
> but I have a hard time believing it is the 5.0 release version.

But that is how it is -- 5.0.0 already implements the new plugin loader which loads from different locations depending on platform and also respects  OPENCPN_PLUGINS_DIRS. Note that this issue was filed long before 5.0.0 was out.

As I see it, the we already have a work-around using OPENCPN_PLUGINS_DIRS. The remaining issue is to possibly change the default value when using portable mode.

rgleason, 2019-08-26:
Thanks for explaning. How will the new gihub ci action affect your work, if at all?
Would it be used?

leamas, 2019-08-26:
I don't follow you here. Do you refer to the Appveyor/Travis builds? In that case I don't see how  they are related...

leamas, 2019-08-26:
>  and the answer is probably "no".

Right. That said,  thanks for the pointer; I wasn't aware of this and it certainly looks interesting.

There are actually more interesting services out there for ci builds like CircleCI and also deployment like cloudsmith. It would be great to have two parallel pipelines for builds + deployment since these services seems far from stable, mostly from a financial perspective.

leamas, 2019-08-27:
> wouldn't github action be more seamless and convenient when it becomes available

Perhaps, we should certainly give it a try. However, the ci-deployment tools are not really critical, there are several alternatives which seems OK besides what we have. The overall design of the plugin system is agnostic to the toolchain used, it's happy as long as  plugins are available on a URI. 

I don't see that the current tools (travis/appveyor + bintray) is a problem as such.  Do you?

transmitterdan, 2020-05-31:
I think this release, as it is now, will completely render portable mode useless.  Portable mode was designed so users could have multiple instances with different personalities or even different versions of the program run simultaneously on a single computer.  Also it is so a user can run from the thumb stick without installing anything from whence it got its name.  Both use chases have always been supported in the past AFAIK.  Now, I fear neither case works at all.

rgleason, 2020-05-31:
That is unfortunate. Have you done some trials?

bdbcat, 2020-05-31:
Dan...
Details, please?

bdbcat, 2020-06-01:
Dan...
I have committed a patch which should correct portable mode plugin loading on Windows.
Is this the nature of your statement:
" Now, I fear neither case works at all."

Dave

transmitterdan, 2020-06-02:
Dave, thanks for looking at this.  I think this will now be ok.  I'll run some tests to be sure.

transmitterdan, 2020-06-03:
Dave,
The patch for Windows is working. Now in portable mode the plugin manager does not crawl all over the computer looking for plugins. Thanks...

rgleason, 2021-04-23:
@RooieDirk  @TDan  can this be closed now?  Thanks

transmitterdan, 2021-04-24:
Fixed in latest master.

closed 2021-04-24 by transmitterdan