OpenCPN/OpenCPN #1142: TCP connection to AIS sources does not reconnect on connection loss

lnitram, 2018-09-16:
It seems that opencpn does not reconnect to a tcp source if the connection gets lost.

To reproduce:
- Start opencpn
- Setup connections and connect to AIS receiver which acts in TCP server mode
- Switch off tcp server
- Switch it on again
=> Ships don't get updated anymore, are marked as lost and then deleted.

Also the TCP export does not seem to reconnect once the destination is not reachable for a short time.

transmitterdan, 2018-09-17:
OpenCPN attempts to connects to the server. Once a connection is made the connection will wait for the server to send messages until a preset timeout. So if you simply wait long enough (about 60 seconds on windows) a new connection can be made.

However, it is better in my opinion to use UDP which is a connectionless protocol and does not require any timeout/reconnect sequence.

In any case, there is no bug in OpenCPN's TCP connection system.

lnitram, 2018-09-17:
ok, I tested again:
- Set up new connection for AIS (tcp, address in my network, right port)
- Applied and ships appear
- Switched off AIS source in the network
- Switched it on again
- Confirmed using netcat that NMEA is available

=> in Opencpn all ships disappeared after a while and did not come back. After disable the connection and enable it again it works

=> The 60 seconds timeout seems not to work on Ubuntu
=> TCP connection does not detect that the source is not available or does not restart it

If this is not a bug, then I suggest to improve the tcp connection handling. I will run some more tests over the next days to get it confirmed.

UDP is not an option for me.

transmitterdan, 2018-09-23:
The timeout may be different on different IP stacks.  I don't think OpenCPN can do anything about this.  The failure message comes from the TCP server not OpenCPN.  The server cannot open the initial communications port because it is already occupied.  It has to wait until the port closes before it can open again.  At least that's how it works on my Linux->Windows system.  But eventually you can restart the server and OpenCPN will connect to it.

nohal, 2019-01-19:
FWIW I'm not able to reproduce this issue.

nohal, 2019-03-12:
There's no new info and the issue is still not reproducible here, I will likely close it at next sweep as such.

lnitram, 2019-03-12:
I will try to reproduce it again and will let you know. Hope to find some time today.

lnitram, 2019-03-12:
Tested with 4.99.1408, all works now as it should. No idea what happened when I noticed the issue. AIS receiver is still the same, just the opencpn version is newer. Sorry for the false alert and thanks for the time you spent on this. I will close it now.

closed 2019-03-12 by lnitram

ve7mjc, 2020-11-04:
We are experiencing this behavior with current builds of OpenCPN. (5.0.0 build 9065270 and 5.2.4)

I agree that the underlying sockets may be operating predictably and as designed - and that we can expect this behavior to be different between platforms and Operating Systems. 

I recommend we wish-list the option of a receive data timeout or similar which could force a socket close and reconnect if data is not received within a specified window. I see this as a fairly critical issue as there is no indication to the user that they are no longer receiving time-sensitive AIS contact information. So far this does not appear to be a "once in a while problem" as I am having users report to me that a number of different builds between Windows and MacOS are sporadically losing their connections and not reconnecting even after many hours of timing out AIS contacts. Our use-case is of course not simply a TCP connection to localhost but that of a centralized AIVDM TCP server accessed across a network.

I will take a look at the source code but I may not be the best as I am not familiar with many of these libraries and build environment.

Hakansv, 2020-11-04:
Are you sure this is a O problem and not the AIS device? 
I've a TCP connection to a server that sometimes is restarted several times a day. O is reconnected all times.
BTW; you can't have more watchdogs for AIS messages since we don't know if or when any ship will transmit. The target lost indication and finally deletion is the best we can do. There's no, what I know, "I'm alive" message transmitted from any AIS.

Did you try to use UDP instead? As of T-Dan's advice it's a better alternative for this kind of network traffic.

ve7mjc, 2020-11-04:
We have confirmed this is not an AIS device problem (TCP server with continuous AIVDM messages in this case) as we are observing a BrokenPipe at the TCP server side when the drop occurs. We verified with tcpdump as well as examined verbose server logs.

We have also shut down and restarted the TCP server while clients were connected and found that frequently clients do not reconnect after many hours. (Windows 10 and OSX with recent builds in this case)

Correct, there is no concept of an AIS receiver "alive" indication to my knowledge. Our use case does not expect any particular MMSI station to be active on the channel but our region is saturated with marine traffic and we are assured "traffic" on AIS at least 50 messages per second. Many AIS transponders also push NMEA strings such as own-ship or other information at least once per second. Ironically we filter these out of our TCP stream to the clients.

Pushing UDP to a dynamic number of desktop and Android is unfortunately not viable. I had considered writing and deploying a cross-platform bridging adapter alongside each user which would aggressively maintain a TCP connection and forward AIVDM messages to localhost UDP but this is not ideal - especially for Android.

I have been examining the source code and found that there appears to be receive data watchdog functionality. I will do a deeper dive here and report back.

This is an incredible application. The code base is astounding and a huge benefit to the community.