OpenCPN/OpenCPN #1144: build: External linker flags are discarded by cmake setup.

leamas, 2018-09-16:
As of today, user-defined linker flags using either the environment variable LDFLAGS or -DCMAKE_SHARED_LINKER_FLAGS=... on the command line are discarded by code setting CMAKE_SHARED_LINKER_FLAGS to absolute values instead of adding to the possibly already defined set of flags. 

The problem is present on all builds paths including Windows, MacOS and Linux. On linux, it's a packaging problem since all distros passes a default set of linker flags which should be used in all builds.

leamas, 2018-10-02:
It's not only linker flags, also CFLAGS  is affected (output cleaned up).  This means that  e. g., optimization level cannot be controlled be the user building opencpn.

    $ git grep CFLAGS
    CMakeLists.txt:#  IF NOT DEBUGGING CFLAGS="-O2 -march=native"
    plugins/chartdldr_pi/cmake/PluginConfigure.cmake:#  IF NOT DEBUGGING CFLAGS="-O2 -march=native"
    plugins/dashboard_pi/CMakeLists.txt:#  IF NOT DEBUGGING CFLAGS="-O2 -march=native"
    plugins/demo_pi_sample/CMakeLists.txt:#  IF NOT DEBUGGING CFLAGS="-O2 -march=native"
    plugins/demo_pi_sample/cmake/PluginConfigure.cmake:#  IF NOT DEBUGGING CFLAGS="-O2 -march=native"
    plugins/grib_pi/CMakeLists.txt:#  IF NOT DEBUGGING CFLAGS="-O2 -march=native"
    plugins/wmm_pi/cmake/PluginConfigure.cmake:#  IF NOT DEBUGGING CFLAGS="-O2 -march=native"

nohal, 2018-10-02:
I would say this is a designed behaviour. But understand that it is not understood by the packagers.

leamas, 2018-10-02:
Could you please expand a little here? Having a default set of flags is of course necessary, but I cannot see the point  in not being able to override it. Here is certainly something I don't understand (no surprise,  after all I'm a packager ;) )

In particular, do you see problems with the patch in #1144?

nohal, 2018-10-02:
No, I do not see any explicit problem with this patch. I do see possible problems with packagers using untested flag sets in good faith they improve stuff though...

nohal, 2018-10-02:
They are good. Until applied to the texture compression stuff.

leamas, 2018-10-02:
OK... which one is the problem?

nohal, 2018-10-02:
`O2`, as far as I remember it kills the performance on Linux significantly.

nohal, 2018-10-02:
Of course O2 is fine and even now used for most of the other OpenCPN code. What I really would like to avoid is "fixing" stuff that is not broken to make packagers feel happy and than have to fix the damage it did which we could have avoided.

leamas, 2018-10-02:
Hm... I see, there is a -O3. I could easily append this, overriding the O2, and I sure will. Thanks for input!

leamas, 2018-10-02:
BTW: I have no idea of "fixing" things here, it's about finding a balance between the kind of overall rules which all distributions have and the specific application needs.  Debian requires all packages to apply the standard flags, but I see no problems in using -O3 as required on top of that. That said, I have some downstream homework to do here, for sure. And without your input I would certainly have missed this one.

leamas, 2018-10-02:
And... After some thinking I don't really see the problem with the -O3 flags. If cmake just is somewhat sane -O3  et. al. is appended to the set of flags initiated by CFLAGS. And since the last flag wins, it should not be a problem.

closed 2019-01-21 by leamas