OpenCPN/OpenCPN #1146: Tracking works poorly on android when device is sleeping or app is in background

stepcut, 2018-09-17:
I typically use opencpn on my phone/android tablet while sailing. If I start tracking and then switch to another app, it will stop tracking after a while.  Also, when the phone goes to sleep it seems to record very little data.

This is hugely annoying as I often want to check weather and radar apps, and I don't always have the screen on.

I assume that both these issues are related to the OS not giving the application the time slices and data that it needs to track in the background. I'm guessing that tracking needs to be implemented as a background Service.

https://developer.android.com/guide/components/services

nohal, 2018-09-18:
What you are using has very little to do with OpenCPN on Android as it is now - it's a ripoff of a very old beta. Either build from source or get the official prebuilt package from https://play.google.com/store/apps/details?id=org.opencpn.opencpn

stepcut, 2018-09-18:
Paid. I see the 'Do not sleep' option as well as 'Enable continuous GPS tracking even while paused/asleep', so I assume things will work better.

I wish it was easier to figure out that the paid version is the 'official' version.

closed 2018-09-18 by stepcut