OpenCPN/OpenCPN #1159: Link to current OpenSeaMap KAP exports in documentation/wiki

taruti, 2018-10-10:
Linking to 
https://wiki.openstreetmap.org/wiki/KAP-charts_from_OpenSeaMap
might make sense. These work out of box with OpenCPN. 
These can be nice for e.g. street maps near harbours.

nohal, 2019-01-10:
This is for @rgleason and other wiki editors.
It is a little unfortunate these are available over FTP and compressed with 7-zip, otherwise the best would of course be to integrate them directly to the chart downloader...

rgleason, 2019-01-16:
I have added this resource to the Chart Sources page
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:charts:chart_sources#openseamap_for_opencpn

Thank you for advising Taruti

Nohal, if they were compressed differently would they work with Chartdownloader? 
If it were to work with ChartDownloader it would be much easier for users, really kind of seamless.

I have not used these charts yet, but will try Caribbean.

nohal, 2019-01-16:
Yes, they would of course work with the chartdownloader if they were compressed using xz or zip.

rgleason, 2019-01-16:
Taruti, is it possible to provide one of those formats, and perhaps add that to the the download tables for Pavel?