OpenCPN/OpenCPN #1161: User Interface question.

Sad7Dayz, 2018-10-18:
hello.
I don't know when i'm Developing.
I want to create add OpenCPN Toolba.  because it will put the plugin that it was Developed.
I wonder where is it.
[image]

nohal, 2018-10-18:
I am afraid I do not understand your question at all... Please try to rephrase it or show us some of your code so that we understand what you are trying to do.

Sad7Dayz, 2018-10-23:
I'm sorry. My English is poor.
As shown above I want to make the same toolbar.  and Source is not open.

nohal, 2018-10-23:
I still do not understand. The OpenCPN source is open (And if you got a plugin from someone without access to it's source, he violated the GPL license), you are looking at it in this repository. Several plugins creating toolbar icons are part of it, for example at https://github.com/OpenCPN/OpenCPN/blob/master/plugins/grib_pi/src/grib_pi.cpp#L136 the Grib plugin inserts it's icon into the toolbar.
If you want to create a whole toolbar, look at https://github.com/jongough/ocpn_draw_pi/blob/master/src/ODToolbarImpl.cpp - But it of course is not 1 line of code as above anymore.

closed 2019-01-19 by nohal