OpenCPN/OpenCPN #1167: Debian 9/gtk3: Assert failures when starting opencpn

leamas, 2018-10-29:
When starting ocpn on Debian there are a multitude of assertion errors as described in http://www.cruisersforum.com/forums/f134/opencpn-rc-4-1-1301-release-158718-8.html, like
    ../src/gtk/colour.cpp(207): assert "IsOk()" failed in Blue(): invalid colour

The  wxsvg patch described in the same article resolves this problem. This leaves me with three (two?)

    src/gtk/window.cpp(5059): assert  "increment > 0" failed in isScrollIncrement
    
before ocpn is successfully started.The traceback points to the function AddPage() in options.cpp and it's very last line

      m_pListbook->DeletePage(parent + 1);
    
This is on ocpn version 4.8.6 with multiple patches,mostly from master.  I can reproduce the same problem using the ubuntu ppa sources with the only patch CMakeLists-Fix-FTBS... from current master to make it build.

OTOH, rebuilding the same sources against gtk2 (without the patch) shows no assert errors on startup. So, this is about gtk3.

Thoughts?

leamas, 2018-10-31:
The wxsvg patch is accepted upstream:  https://sourceforge.net/p/wxsvg/discussion/424987/thread/ee22f2bacf/?limit=50#b50a

leamas, 2018-10-31:
Without more input here the conclusion is that despite the gtk3 hooks in the build script, gtk3 isn't realli in shape as of 4.8,6. As for packaging, this means I revert to using the gtk2 backend.

However, in a longer perspective (5.0?) staying on gtk2 doesn't seem appropriate. With gtk 4 under way, the support for gtk2 will be certainly become shaky.

The gtk2 backward compatibility is a mess :(

Leaving bug open as a 5.0/gtk3 issue.

leamas, 2018-11-14:
Upstream wxWidgets bug: http://trac.wxwidgets.org/ticket/17702

reneherrero, 2019-09-29:
Has this moved forward?  Is there a way to enable gtk3 in the build?

reneherrero, 2019-09-29:
I did some digging around, found the `OCPN_FORCE_GTK3` option and tried doing `cmake -DOCPN_FORCE_GTK3=ON ../` (most of my experience is with autotools, please don't shoot).

It seems to pickup the gtk3 libs and compiles correctly, but when I run it I get a seg fault.  Normal?  Is this work in progress?  Or is this something on my end?
```
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.24.5
-- Found GTK3: /usr/include/gtk-3.0;/usr/include/at-spi2-atk/2.0;/usr/include/at-spi-2.0;/usr/include/dbus-1.0;/usr/lib/x86_64-linux-gnu/dbus-1.0/include;/usr/include/gtk-3.0;/usr/include/gio-unix-2.0;/usr/include/cairo;/usr/include/libdrm;/usr/include/pango-1.0;/usr/include/harfbuzz;/usr/include/pango-1.0;/usr/include/fribidi;/usr/include/atk-1.0;/usr/include/cairo;/usr/include/pixman-1;/usr/include/freetype2;/usr/include/libpng16;/usr/include/gdk-pixbuf-2.0;/usr/include/libmount;/usr/include/blkid;/usr/include/uuid;/usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include  
-- Building against GTK3...
```
All gtk2 apps look like crap when using the Adawait Dark theme in Gnome...

```
rene@prometheus:~$ opencpn 
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
14:11:06: Device: 0: HDA Intel PCH: ALC3266 Analog (hw:0,0)
14:11:06: Device: 1: HDA Intel PCH: HDMI 0 (hw:0,3)
14:11:06: Device: 2: HDA Intel PCH: HDMI 1 (hw:0,7)
14:11:06: Device: 3: HDA Intel PCH: HDMI 2 (hw:0,8)
14:11:06: Device: 4: HDA Intel PCH: HDMI 3 (hw:0,9)
14:11:06: Device: 5: HDA Intel PCH: HDMI 4 (hw:0,10)
14:11:06: Device: 6: sysdefault
14:11:06: Device: 7: hdmi
14:11:06: Device: 8: lavrate
14:11:06: Device: 9: samplerate
14:11:06: Device: 10: speexrate
14:11:06: Device: 11: pulse
14:11:06: Device: 12: upmix
14:11:06: Device: 13: vdownmix
14:11:06: Device: 14: default
14:11:06: Device: 0: HDA Intel PCH: ALC3266 Analog (hw:0,0)
14:11:06: Device: 1: HDA Intel PCH: HDMI 0 (hw:0,3)
14:11:06: Device: 2: HDA Intel PCH: HDMI 1 (hw:0,7)
14:11:06: Device: 3: HDA Intel PCH: HDMI 2 (hw:0,8)
14:11:06: Device: 4: HDA Intel PCH: HDMI 3 (hw:0,9)
14:11:06: Device: 5: HDA Intel PCH: HDMI 4 (hw:0,10)
14:11:06: Device: 6: sysdefault
14:11:06: Device: 7: hdmi
14:11:06: Device: 8: lavrate
14:11:06: Device: 9: samplerate
14:11:06: Device: 10: speexrate
14:11:06: Device: 11: pulse
14:11:06: Device: 12: upmix
14:11:06: Device: 13: vdownmix
14:11:06: Device: 14: default
14:11:06: Device: 0: HDA Intel PCH: ALC3266 Analog (hw:0,0)
14:11:06: Device: 1: HDA Intel PCH: HDMI 0 (hw:0,3)
14:11:06: Device: 2: HDA Intel PCH: HDMI 1 (hw:0,7)
14:11:06: Device: 3: HDA Intel PCH: HDMI 2 (hw:0,8)
14:11:06: Device: 4: HDA Intel PCH: HDMI 3 (hw:0,9)
14:11:06: Device: 5: HDA Intel PCH: HDMI 4 (hw:0,10)
14:11:06: Device: 6: sysdefault
14:11:06: Device: 7: hdmi
14:11:06: Device: 8: lavrate
14:11:06: Device: 9: samplerate
14:11:06: Device: 10: speexrate
14:11:06: Device: 11: pulse
14:11:06: Device: 12: upmix
14:11:06: Device: 13: vdownmix
14:11:06: Device: 14: default
14:11:06: Device: 0: HDA Intel PCH: ALC3266 Analog (hw:0,0)
14:11:06: Device: 1: HDA Intel PCH: HDMI 0 (hw:0,3)
14:11:06: Device: 2: HDA Intel PCH: HDMI 1 (hw:0,7)
14:11:06: Device: 3: HDA Intel PCH: HDMI 2 (hw:0,8)
14:11:06: Device: 4: HDA Intel PCH: HDMI 3 (hw:0,9)
14:11:06: Device: 5: HDA Intel PCH: HDMI 4 (hw:0,10)
14:11:06: Device: 6: sysdefault
14:11:06: Device: 7: hdmi
14:11:06: Device: 8: lavrate
14:11:06: Device: 9: samplerate
14:11:06: Device: 10: speexrate
14:11:06: Device: 11: pulse
14:11:06: Device: 12: upmix
14:11:06: Device: 13: vdownmix
14:11:06: Device: 14: default
Segmentation fault
``` 
App runs ok with gtk2, but I see things like this: (last two plugins don't show the text unless I select them)
[image]

leamas, 2019-09-30:
>  but when I run it I get a seg fault. 

In which circumstances?

> App runs ok with gtk2, but I see things like this: 

For me it looks perfectly sane in using both gtk2 and gtk3.

Building on fedora 29,  wxGTK3-3.0.4-4. Fedora does not ship gtk2 libraries, instead there are compat packages seemingly on top of gtk3. Sorry, I have no info on them.

I had to use `-DOCPN_WXWIDGETS_OPTIONS="--unicode=yes"` to make it build

leamas, 2019-09-30:
> For me it looks perfectly sane in using both gtk2 and gtk3.

Actually I got it wrong. I cannot build against GTK3...

reneherrero, 2019-09-30:
> In which circumstances?

Right on startup. The UI never even opens up.  Dies here: `14:11:06 CDT: Creating MyFrame...size(1344, 756)  position(321, 71)`

> Actually I got it wrong. I cannot build against GTK3...

It builds for me.  I started to get confused as to what needs to be on the machine (I installed so many libs I'm not sure what's being used) so I restarted yet again except this time using Docker and Gitlab so I can keep track (and remember the next time I need to do it): https://gitlab.com/sevenseas/opencpn/pipelines

Here's how I'm trying to isolate the issue:
1. I just created another branch called gtk3
2. I removed the gtk2 libs and replaced them with the gtk3 libs from the Dockerfile
3. I also replaced the wxwidgets library with one that seems to be made for gtk3
4. cmake didn't complain and the build is in-progress

I build a docker image with all the dependencies, push it to the Gitlab registry of the project, then clone OpenCPN in an instance of that image and do the build.

Don't know if this is optimal, but it works for me.  Idea is to make a Debian package for Buster once I got it all ironed out and install that instead of building from source on each machine.  And like I said, I always forget what I did when too much time has gone by...  I'm guessing you can also do this with with travis ci and docker hub.

reneherrero, 2019-09-30:
Build passed with these dependencies (no gtk2): https://gitlab.com/sevenseas/opencpn/blob/gtk3/Dockerfile

And if you look here you'll see what's below: https://gitlab.com/sevenseas/opencpn/-/jobs/308200831
```
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.24.5
-- Found GTK3: /usr/include/gtk-3.0;/usr/include/at-spi2-atk/2.0;/usr/include/at-spi-2.0;/usr/include/dbus-1.0;/usr/lib/x86_64-linux-gnu/dbus-1.0/include;/usr/include/gtk-3.0;/usr/include/gio-unix-2.0;/usr/include/cairo;/usr/include/libdrm;/usr/include/pango-1.0;/usr/include/harfbuzz;/usr/include/pango-1.0;/usr/include/fribidi;/usr/include/atk-1.0;/usr/include/cairo;/usr/include/pixman-1;/usr/include/freetype2;/usr/include/libpng16;/usr/include/gdk-pixbuf-2.0;/usr/include/libmount;/usr/include/blkid;/usr/include/uuid;/usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include  
-- Building against GTK3...
```

I'm guessing it should also work on Fedora.

taruti, 2019-10-01:
@reneherrero 
Could you run opencpn with gdb to get a backtrace from the segfault:
1) Run `gdb opencpn`
2) Type `run` at the (gdb) prompt.
3) Wait till opencpn segfaults
4) Type `bt` at the (gdb) prompt.
5) Copy-paste that output here.

leamas, 2019-10-02:
OK, I can now build it even if the gtk2  compat libraries are installed. It seems to be all about the alternatives setup for wx-config. For Fedora, my missing steps was

    sudo dnf install wxGTK3-devel.x86_64
    sudo alternatives --install /usr/bin/wx-config wx-config /usr/lib64/wx/config/gtk3-unicode-3.0
    sudo alternatives --set wx-config /usr/lib64/wx/config/gtk3-unicode-3.0

Starting on the regular wayland desktop gives a crash at startup in wxGlCanvas; this is #1166.

After switching to X11 opencpn starts and shows initial window. Not tested anything besides this.

leamas, 2019-10-02:
Basically, this means that the problem described in the heading is fixed, at least for me. Which means that I should close this bug. 

Which I will unless someone feels strongly about keeping it open. Anyone?

reneherrero, 2019-10-02:
@taruti Here you go:
```
(gdb) bt
#0  0x0000000000000002 in  ()
#1  0x00007ffff7a4800b in XQueryExtension ()
    at /lib/x86_64-linux-gnu/libX11.so.6
#2  0x00007ffff1946862 in  () at /lib/x86_64-linux-gnu/libGLX.so.0
#3  0x00007ffff19428e5 in glXQueryVersion ()
    at /lib/x86_64-linux-gnu/libGLX.so.0
#4  0x00007ffff76ffe55 in wxGLCanvasX11::GetGLXVersion() ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#5  0x00007ffff7700455 in wxGLCanvasX11::ConvertWXAttrsToGL(int const*, int*, unsigned long) () at /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#6  0x00007ffff7700b68 in wxGLCanvasX11::InitXVisualInfo(int const*, __GLXFBConfigRec***, XVisualInfo**) () at /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#7  0x00007ffff7701b5f in wxGLCanvas::Create(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&, int const*, wxPalette const&) ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#8  0x00007ffff7701d03 in wxGLCanvas::wxGLCanvas(wxWindow*, int, int const*, wxPoint const&, wxSize const&, long, wxString const&, wxPalette const&) ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_gl-3.0.so.0
#9  0x0000555555a5df35 in glTestCanvas::glTestCanvas(wxWindow*)
    (this=0x5555562d7bb0, parent=0x555556365960)
    at /usr/include/wx-3.0/wx/gdicmn.h:258
#10 0x00005555558ff285 in OCPNPlatform::BuildGLCaps(void*)
    (this=<optimized out>, pbuf=0x5555562d6bb0)
```

leamas, 2019-10-03:
hm... is this no wayland or X11?

reneherrero, 2019-10-03:
That's with Wayland.

I tried starting opencpn with X11 using `GDK_BACKEND=x11 opencpn` and I get that split screen again (except this time I have all the libs installed).

[image]

Here's the cmake output for completeness sake:
```
rene@prometheus:~/Source/github.com/OpenCPN/OpenCPN/build$ cmake -DOCPN_FORCE_GTK3=ON ../
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- cmake version: 3.13.4
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- *** Build Architecture is amd64
-- Setting C++11 standard via cmake standard mechanism
-- Default compiler options:
-- CMAKE_CXX_FLAGS: 
-- CMAKE_CXX_FLAGS_DEBUG: -g
-- CMAKE_CXX_FLAGS_MINSIZEREL: -Os -DNDEBUG
-- CMAKE_CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_CXX_FLAGS_RELWITHDEBINFO: -O2 -g -DNDEBUG
-- *** Staging to build opencpn 5.0.0 ***
-- *** Package will NOT include documentation ***
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.24.5
-- Found GTK3: /usr/include/gtk-3.0;/usr/include/at-spi2-atk/2.0;/usr/include/at-spi-2.0;/usr/include/dbus-1.0;/usr/lib/x86_64-linux-gnu/dbus-1.0/include;/usr/include/gtk-3.0;/usr/include/gio-unix-2.0;/usr/include/cairo;/usr/include/libdrm;/usr/include/pango-1.0;/usr/include/harfbuzz;/usr/include/pango-1.0;/usr/include/fribidi;/usr/include/atk-1.0;/usr/include/cairo;/usr/include/pixman-1;/usr/include/freetype2;/usr/include/libpng16;/usr/include/gdk-pixbuf-2.0;/usr/include/libmount;/usr/include/blkid;/usr/include/uuid;/usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include  
-- Building against GTK3...
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found OpenGL....
--     GL Lib: /usr/lib/x86_64-linux-gnu/libGL.so/usr/lib/x86_64-linux-gnu/libGLU.so
--     GL Include: /usr/include
--     Revised GL Lib: /usr/lib/x86_64-linux-gnu/libGL.so
-- Found wxWidgets: -L/usr/lib/x86_64-linux-gnu;-pthread;;;-lwx_gtk3u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_core-3.0;-lwx_baseu-3.0;-lwx_gtk3u_webview-3.0 (found version "3.0.4") 
-- Found wxWidgets webview add-on
-- Found wxWidgets...
--  wxWidgets Include: /usr/lib/x86_64-linux-gnu/wx/include/gtk3-unicode-3.0;/usr/include/wx-3.0
--  wxWidgets Libraries: -L/usr/lib/x86_64-linux-gnu;-pthread;;;-lwx_gtk3u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_core-3.0;-lwx_baseu-3.0;-lwx_gtk3u_webview-3.0;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_gl-3.0
--  Revised wxWidgets Libraries: -L/usr/lib/x86_64-linux-gnu;-pthread;-lwx_gtk3u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_core-3.0;-lwx_baseu-3.0;-lwx_gtk3u_webview-3.0;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_gl-3.0
-- 
-- *** Staging to build opencpn  ***
-- *** Build type: RelWithDebInfo
-- *** Will install to /usr/local  ***
-- GarminHost is enabled.
-- Found Gettext: /usr/bin/msgmerge (found version "0.19.8.1") 
-- selected config gtk3-unicode-3.0

-- Found OpenGL....
--     GL Lib: 
--     GL Include: /usr/include
-- Crash reporting enabled
-- Using Linux crash reporting
-- Checking for module 'sqlite3'
--   Found sqlite3, version 3.27.2
-- Found Sqlite: /usr/include  
-- Using systrem SQLite library version 3027002.
-- Found BZip2: /usr/lib/x86_64-linux-gnu/libbz2.so (found version "1.0.6") 
-- Looking for BZ2_bzCompressInit
-- Looking for BZ2_bzCompressInit - found
[... 65 lines cut ...]
-- chartdldr_pi: building with system tinyxml
-- *** Added plugin: /home/rene/Source/github.com/OpenCPN/OpenCPN/plugins/chartdldr_pi
-- 
-- *** Staging to build dashboard_pi ***
-- *** Added plugin: /home/rene/Source/github.com/OpenCPN/OpenCPN/plugins/dashboard_pi
-- *** Staging to build grib_pi ***
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so (found version "62") 
-- Could NOT find Jasper (missing: JASPER_LIBRARIES JASPER_INCLUDE_DIR) 
-- Using bundled jasper library...
-- *** Added plugin: /home/rene/Source/github.com/OpenCPN/OpenCPN/plugins/grib_pi
-- Build type: RelWithDebInfo
-- *** Staging to build wmm_pi ***
-- Found OpenGL...
--     Lib: /usr/lib/x86_64-linux-gnu/libGL.so/usr/lib/x86_64-linux-gnu/libGLU.so
--     Include: /usr/include
-- Found OpenGL...
--     Lib: /usr/lib/x86_64-linux-gnu/libGL.so/usr/lib/x86_64-linux-gnu/libGLU.so
--     Include: /usr/include
-- *** Added plugin: /home/rene/Source/github.com/OpenCPN/OpenCPN/plugins/wmm_pi
-- *** Package will include GSHHS basechart level: NONE ***
-- *** Package will NOT include tide and current data ***
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rene/Source/github.com/OpenCPN/OpenCPN/build
```

reneherrero, 2019-10-03:
FYI: I tried a gtk3 build again on another machine to see if this could be a graphics card issue.  Same thing happens with both the NVidia Quadro P1000 Mobile and the Intel 620

leamas, 2019-10-03:
> That's with Wayland

OK, that means that the crash is #1166. It is a known, wxWidgets  bug.

Closing this bug, referring the crash to #1166.

closed 2019-10-03 by leamas