OpenCPN/OpenCPN #1170: master window path search is broken

did-g, 2018-11-06:
Hi,
On windows : is not a valid path separator...

> 20:13:53: PlugInManager: plugins loading from C:\Program Files (x86)\OpenCPN\plugins
> 20:13:53: PlugInManager searching for PlugIns in location C
> 20:13:53:    Directory C does not exist.
> 20:13:53: PlugInManager searching for PlugIns in location \Program Files (x86)\OpenCPN\plugins
> 20:13:53:    Directory \Program Files (x86)\OpenCPN 5.2.11105\plugins does not exist.
> 
Bug introduced in 
4b68cc6bd7795007c13d4a894c7de07d1f2c0484
ab1d56dbc9e2851a1d145a424733e09521d97903

Is multiple path for Windows make sense?

Regards 
Didier

nohal, 2018-11-06:
Not at this moment, but it probably will if we at some moment implement some sort of a plugin manager, which will have to place stuff to some directory writable for the user.

leamas, 2018-11-08:
Ah... will look into this. The intention is to have single directory on Windows for now,  but still split it on ':' which is plain wrong on Windows. Stay tuned....

leamas, 2018-11-08:
@did-g: Could you please try the enclosed patch?

[0018-pluginmanager-Windows-load-path-bugfix-1170.patch.gz](https://github.com/OpenCPN/OpenCPN/files/2562239/0018-pluginmanager-Windows-load-path-bugfix-1170.patch.gz)

did-g, 2018-11-09:
Hi
@leamas 
Didn't test but look at it, should be fine.
There's two places in linux only code path using ":", they could use PATH_SEP

leamas, 2018-11-09:
> There's two places in linux only code path using ":", they could use PATH_SEP

Indeed, and considering future windows use of multiple load directories  they probably should. Good catch!  Attaching an updated patch. 

However, to close this issue we would need some testing, and I don't have any windows boxes at hand for this (I could of course make a plain test, but the build chain is challenging for a linux guy ;) ) 

[0018-pluginmanager-Windows-load-path-bugfix-1170.patch.gz](https://github.com/OpenCPN/OpenCPN/files/2564949/0018-pluginmanager-Windows-load-path-bugfix-1170.patch.gz)

did-g, 2018-11-09:
I haven't windows either but O works file with wine and you can:
- use appveyor  for each PR in
https://ci.appveyor.com/project/bdbcat/opencpn/builds/xxx/artifacts there's
a window installer, if you create your own account I think you can use RDP
and run O in Appveyor.

- use mingw best with mxe, there's rough edges though, and current master
is not 100% compatible, can build 32 or 64 bits versions. It's the reason
why I found this one, I have C: in config file but wine starts from Z: and
suddenly no more plugins.

leamas, 2018-11-09:
OK... at a glance, this is still more work than I'm willing ti undertake. OTOH, you found the bug and have reviewed the patch. Basically, we both seems confident it will work. I will make a PR with the patch later.

Hakansv, 2018-11-09:
I also noticed Didier's initial note some days ago. It's though working wo the "C:\"  but the log message is annoying. 
I can build and test any patch on Win10 if you like.   
Håkan

leamas, 2018-11-09:
> I can build and test any patch on Win10 if you like.

@hakansv; If you could build, test and report back that would be very helpful, actually.

Hakansv, 2018-11-10:
The above exception is introduced here:
It seems some changes have recently been made to either of: myConfigFileDir  or  GetPathSeparator()
Now we "suddenly" have double path separators in Windows? See this debug from styles.cpp:1147
[image]
Any bells ringing?
Håkan

leamas, 2018-11-10:
@hakan: hm...
Perhaps you could open a separate issue for this one. It seems unrelated to thecore issue here (?).
Doing  so, perhaps you could try to do a git bisect to find out the offending commit if the build isn't to painful?

leamas, 2018-11-10:
@hakan: continuing on this issue: I attach the complete downstream packaging patch series against 4.8.6 . Although it's linux-centric I don't think it should break too many eggs when used on Windows (almost all of the patches are rebased from master). If you have time to verify that the bug is gone using this series we could close this issue.
[patches.tar.gz](https://github.com/OpenCPN/OpenCPN/files/2568283/patches.tar.gz)

Hakansv, 2018-11-12:
Sorry for the lag.
After some local workaround to be able to run and test this patch:
OK, the patch works in Windows as well. 

My above error seems related to some Journeyman_flat style issue. Not relevant for this issue. Closing is OK for me.

leamas, 2018-11-12:
@did-g: PR in place, perhaps we should close this bug?
Thanks for spotting and reporting my bugs!

closed 2018-12-20 by did-g