OpenCPN/OpenCPN #1184: MUI branch - segfaults opening dialog Options->Display->Advanced->Options

transmitterdan, 2018-12-06:
When trying to open the OpenGL Options dialog box i get a segfault.  If backtrace would be helpful I can create it.  If it is a known issue I can wait for a fix.

System: Ubuntu 18.04.1 LTS running in VirtualBox, Host system: Windows 10

tai-fun, 2018-12-06:
I can see the same problem on Win10 but not on Kubuntu 18.04/64.
Error report sent on Win10.

Hakansv, 2018-12-06:
I'm on Win10 and VS 2107 can't reproduce neither in Release nor Debug mode.
Probably not related but there is a file path change committed for this function yesterday.

Hakansv, 2018-12-06:
For info I've built present MUI on a RPi 3b+. No problem to open the OpenGL Options dialog box.
My system for info:
raspex@raspberry:~$ uname -a
Linux raspberry 4.14.76-v8+ #1 SMP PREEMPT Tue Oct 16 18:45:20 CEST 2018 aarch64 aarch64 aarch64 GNU/Linux
raspex@raspberry:~$ cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.10
DISTRIB_CODENAME=cosmic
DISTRIB_DESCRIPTION="RaspEX - Ubuntu Cosmic Cuttlefish"
NAME="Ubuntu"
VERSION="18.10 (Cosmic Cuttlefish)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.10"
VERSION_ID="18.10"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=cosmic
UBUNTU_CODENAME=cosmic

bdbcat, 2018-12-08:
Thomas and/or Tdan...
@Thomas:
The crash report you sent already was somehow scrambled and unusable.

I have made a fresh build for testing today:
https://www.dropbox.com/s/yck4fkw601y0pkg/opencpn_4.99.1207_setup.exe?dl=0

Please generate a crash again with 1207, if possible. Allow the report to be uploaded to my server.
A backtrace would be useful, if possible.
Thanks
Dave

transmitterdan, 2018-12-08:
Dave,

My Windows build does not crash.  But my Linux build will crash.  Backtrace here:

[backtrace.txt](https://github.com/OpenCPN/OpenCPN/files/2659450/backtrace.txt)

tai-fun, 2018-12-08:
Dave
New Crash report sent.
As before  the crash came when opening the GL dialog and then ticking the
"cache" box.
But, when I change "OpenGL" in the ini file from 0 to 1, and then restart,
all goes well. No problem to tick the "Cache".

This is the GL section from the latest log file:
9:10:22 AM: OpenGL-> Renderer String: Intel(R) HD Graphics
9:10:22 AM: OpenGL-> Version reported:  4.0.0 - Build 10.18.10.4276
9:10:22 AM: OpenGL-> Detected Intel renderer, disabling stencil buffer
9:10:22 AM: OpenGL-> Texture rectangle format: de1
9:10:22 AM: OpenGL-> glGenerateMipmap unavailable
9:10:22 AM: OpenGL-> Detected Windows Intel renderer, disabling
Vertexbuffer Objects
9:10:22 AM: OpenGL-> Vertexbuffer Objects unavailable
9:10:22 AM: OpenGL-> Using Framebuffer Objects
9:10:22 AM: OpenGL-> Using FBO Stencil buffer
9:10:22 AM: OpenGL-> Using Depth buffer clipping
9:10:22 AM: OpenGL-> Not Using compression
9:10:22 AM: OpenGL-> Minimum cartographic line width:  1.0
9:10:22 AM: OpenGL-> Minimum symbol line width:  1.0

Thanks
Thomas

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

transmitterdan, 2018-12-08:
It is definitely easy to trigger the crash with OpenGL off.  Since this MUI version seems not to have the software gl option it runs really slow with OpenGL off.  I think a race condition is responsible for the crash.  The options dialog takes a while to close as it does a lot of fiddling with chart data bases and the like.  So for example if you try to shut down O right after closing the options dialog it will crash because the globals are getting deleted but the options dialog is still trying to use them.  It has probably always been this way but for some reason drawing in non-OpenGL is much slower in MUI than in master.

tai-fun, 2018-12-08:
> Fixed.
> Simple problem. O was calling GL driver to detect some capabilities, but
> the driver context had not been setup since GL mode was not active.
> Probably been that way forever...

The fix works on W10.
Thanks

closed 2019-01-10 by transmitterdan