OpenCPN/OpenCPN #1186: impossible position

AartK, 2018-12-14:
Sometimes an AIS track seems to be "endless". This error is caused by the AIS receiver inside the AIS transponder. On the screen it is displayed as an impossible position. Something like 91N or 181E. Can that be filtered out?

RooieDirk, 2018-12-17:
Aart, do you mean your own position (from a AIVDO sentence) or an AIS target?
For an AIS target I have some code in testing that is stopping target tracks to do jumps of more as 6 Nm.
As for an AIVDO sentence there is a check already for impossible posittions
`if( ( lon_tentative <= 180. ) && ( lat_tentative <= 90. ) ) // Ship does not report Lat or Lon "unavailable"`
Do you have some sample nmea data??

nohal, 2019-03-12:
I still haven't seen this with many many hours of AIS data feed, so unless a data feed that demonstrates the actual issue appears, will be closed on the next sweep.

rgleason, 2021-04-23:
Time to close this please.

closed 2021-04-23 by rgleason