OpenCPN/OpenCPN #1212: Infinite loop during shutdown

nohal, 2019-01-10:
since 0da43424fc267eb81b954adb0bb4ea5cb6873391 if the `MarkInfoImpl` dialog has been open during the session, application never terminates at least on macOS Mojave.
As I'm currently working on #1210 which touches this code again, will try to fix it there myself.

Hakansv, 2019-01-10:
It could be this I've seen this on Win10 as well. I've sometimes need to kill O from program manager to be able to start a new session.

nohal, 2019-01-10:
I have found the cause already, should be 100% reproducible if the platform is affected.

Hakansv, 2019-01-10:
Confirmed. Every time that dialog has been open I've to kill afterwords. Not all sessions sees that dialog though, I'll keep it down until fixed.
Thanks

transmitterdan, 2019-01-11:
This function ```MarkInfoDlg::OnNotebookPageChanged``` appends every tide station in the world to m_comboBoxTideStation.  Is this really the intended behavior?  I would hate to have to scroll through that list looking for a particular station.  Maybe the original intention was to only append stations less than some distance from the mark so as to only show nearby stations?

transmitterdan, 2019-01-11:
I found also several wxAsserts in this module.  If you are interested I made a branch [infiniteLoop](https://github.com/transmitterdan/OpenCPN/commit/aeaa771191e859c9c2e9bce3841a5991a13d3b0a) in my repository show how to get rid of the asserts and also I limited the number of tide stations to 100 in the markinfo dialog page Extended tab.

commit 1361cd3d (2019-01-19): Fix #1212
closed 2019-01-19 by nohal