OpenCPN/OpenCPN #1225: Bugs found during testing of OpenCPN 4.99.1301 on Debian 9.6

stygge-ulv, 2019-01-20:
During testing of OpenCPN 4.99.1301 on Debian 9.6 I have come across the following bugs:

1. When in OpenGL mode (and possibly non-OpenGL) changing vector chart settings regarding which attributes to display do not always occur as expecte upon hitting Apply or OK button. Sometimes after loading a template they appear.  Erattic behaviour and not always reproducible it seems.

2. In certain areas of the Danish 0eSENC chart in the area called "Limfjorden" charts dont display correct when zooming out while displaying two neighboring charts with differenc scales.  I have attached two images that show the behaviour.  One image is at one zoom level where all render ok, whereas the other image is zoomed out one level using the dash/minus key, which render half the image brown.  Disabling quilting bring back the waterway in the browned-out area, but that only introduce problems in other areas that dont display right.  Using vector chart detail scaling setting of 5 reduce the problem, at the expense of some information cluttering.

3. While in the area shown in the attached images, turning OpenGL off and then back on result in an error, the log of which is also attached as images.

4. Using the oeSENC plugin to download charts from ocharts crashed OpenCPN consistently every time, and only way to get charts into OpenCPN was to do it manually after a manual download.

[image]
[image]
[image]
[image]

stygge-ulv, 2019-01-20:
I have found another issue.  Not a big thing but annoying:

When using one of the key menus such as "M" for measurement or "D" for details level, upon leaving this menu through "Esc" key, input focus for the main window is gone and subsequent attempts to use a single key command fails until the canvas get input forcus again by clicking in it or some such.

stygge-ulv, 2019-01-21:
> "When in OpenGL mode (and possibly non-OpenGL) changing vector chart settings regarding which attributes to display do not always occur as expecte upon hitting Apply or OK button."
> 
> Specifics, please? Are you using the "CanvasOptions" menu (3-bar, lower right), or the global settings dialog?
> 
> Thanks
> Dave

I will investigate to see if I can find some reproducible errors.  Unfortunately the problems I saw were not reproducible, and were often a prelude to some more serious error wherer OpenCPN behaved erratic or crashed altogeter.

rgleason, 2021-04-14:
Close as there is a new version?

bdbcat, 2021-04-28:
Closing, inactive

closed 2021-04-28 by bdbcat