OpenCPN/OpenCPN #1227: BUG: OpenCPN crash during import of oeSENC chart from ochart.org

stygge-ulv, 2019-01-21:
During attempt to refresh/import my danish oeSENC chart (which was manually downloaded and installed)  using the oesenc 2.7 plugin OpenCPN consistently crash with a core dump.
Both OpenCPN and oeSENC plugins were downloaded from github yesterday and compiled with cmake option -DCMAKE_BUILD_TYPE=Debug.  I have included a backtrack trace of the core dump using the GNU debugger.  Hope this can help alleviate this bug. 
Platform. Debian 9.6 on Intel CPU

gdb /usr/bin/opencpn ./core
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/opencpn...(no debugging symbols found)...done.
[New LWP 19361]
[New LWP 19558]
[New LWP 19559]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `opencpn'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00005650a6f0a10b in ?? ()
[Current thread is 1 (Thread 0x7fb8b8b8da40 (LWP 19361))]
(gdb) bt
#0  0x00005650a6f0a10b in ?? ()
#1  0x00005650a6f0a1a2 in ?? ()
#2  0x00005650a6f2bba3 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
#3  0x00005650a6f2aa83 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() ()
#4  0x00005650a6f2869a in std::__shared_ptr<wxCurlBase, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() ()
#5  0x00005650a6f2bc59 in std::__shared_ptr<wxCurlBase, (__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_ptr<wxCurlBase, (__gnu_cxx::_Lock_policy)2>&&) ()
#6  0x00005650a6f2aac2 in std::shared_ptr<wxCurlBase>::operator=(std::shared_ptr<wxCurlBase>&&) ()
#7  0x00005650a704853d in wxCurlBaseThread::SetURL(wxString const&, std::shared_ptr<wxCurlBase>) ()
#8  0x00005650a704837f in wxCurlBaseThread::SetURL(wxString const&) ()
#9  0x00007fb8a1d6404e in doDownload(oeSencChartPanel*, int) () from /usr/lib/opencpn/liboesenc_pi.so
#10 0x00007fb8a1d6c2f2 in shopPanel::doDownloadGui() () from /usr/lib/opencpn/liboesenc_pi.so
#11 0x00007fb8a1d6b731 in shopPanel::OnButtonInstall(wxCommandEvent&) () from /usr/lib/opencpn/liboesenc_pi.so
#12 0x00007fb8b402a40e in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#13 0x00007fb8b41afea5 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#14 0x00007fb8b41aff9b in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#15 0x00007fb8b41b034b in wxEvtHandler::TryHereOnly(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#16 0x00007fb8b41b03d3 in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#17 0x00007fb8b41b0435 in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#18 0x00007fb8b48ee2f8 in wxWindowBase::TryAfter(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#19 0x00007fb8b41b01a7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#20 0x00007fb8b474f7d7 in ?? () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#21 0x00007fb8b81f5f75 in g_closure_invoke () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#22 0x00007fb8b8208053 in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#23 0x00007fb8b8210bdc in g_signal_emit_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#24 0x00007fb8b8210fbf in g_signal_emit () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#25 0x00007fb8b62030c5 in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#26 0x00007fb8b81f5f75 in g_closure_invoke () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#27 0x00007fb8b820795c in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#28 0x00007fb8b8210bdc in g_signal_emit_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#29 0x00007fb8b8210fbf in g_signal_emit () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#30 0x00007fb8b6202029 in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#31 0x00007fb8b62a77bc in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#32 0x00007fb8b81f5f75 in g_closure_invoke () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#33 0x00007fb8b820837d in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#34 0x00007fb8b821067f in g_signal_emit_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#35 0x00007fb8b8210fbf in g_signal_emit () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#36 0x00007fb8b63bf8ac in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#37 0x00007fb8b62a5f84 in gtk_propagate_event () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#38 0x00007fb8b62a633b in gtk_main_do_event () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#39 0x00007fb8b6819cbc in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#40 0x00007fb8b84837f7 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#41 0x00007fb8b8483a60 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#42 0x00007fb8b8483d82 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#43 0x00007fb8b62a53b7 in gtk_main () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#44 0x00007fb8b46d9e65 in wxGUIEventLoop::DoRun() () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#45 0x00007fb8b406c27b in wxEventLoopBase::Run() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#46 0x00007fb8b475cbcd in wxDialog::ShowModal() () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#47 0x00005650a6c7fd0f in MyFrame::DoOptionsDialog() ()
#48 0x00005650a6c77df4 in MyFrame::DoSettings() ()
#49 0x00005650a6c76938 in MyFrame::OnToolLeftClick(wxCommandEvent&) ()
#50 0x00007fb8b402a40e in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#51 0x00007fb8b41afea5 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#52 0x00007fb8b41aff9b in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#53 0x00007fb8b41b034b in wxEvtHandler::TryHereOnly(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#54 0x00007fb8b41b0153 in wxEvtHandler::DoTryChain(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#55 0x00007fb8b41b0435 in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#56 0x00007fb8b41b0ef3 in wxEvtHandler::ProcessPendingEvents() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#57 0x00007fb8b402cbff in wxAppConsoleBase::ProcessPendingEvents() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#58 0x00007fb8b46b9e0a in wxApp::DoIdle() () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#59 0x00007fb8b46b9ef3 in ?? () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#60 0x00007fb8b84836aa in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#61 0x00007fb8b8483a60 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#62 0x00007fb8b8483d82 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#63 0x00007fb8b62a53b7 in gtk_main () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#64 0x00007fb8b46d9e65 in wxGUIEventLoop::DoRun() () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#65 0x00007fb8b406c27b in wxEventLoopBase::Run() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#66 0x00007fb8b4031776 in wxAppConsoleBase::MainLoop() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#67 0x00007fb8b40bdcd0 in wxEntry(int&, wchar_t**) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#68 0x00005650a6c6782a in main ()
(gdb)

bdbcat, 2019-01-21:
Corrected in github oesenc_pi master now.
Thanks

stygge-ulv, 2019-01-21:
Please let me know if the kind of debug trace like the one I sent is helpful in finding the problem.  I could not get it working properly with symbols and code snippets so I was wondering if it is of any use ?

nohal, 2019-01-22:
Yes, it is useful, usually the best way to isolate the problem.

closed 2019-01-22 by nohal