OpenCPN/OpenCPN #1239: OCPN 4.99.1326 Bugs found on Windows 10 x64 (opengl ON)

bc38, 2019-01-31:
1. GRIB plugin. In the two Canvas Layout mode, the display of the direction of the wind arrows and particles does not work correctly.
2. When you turn on the Touch Screen mode, the width of the route line and track increases. For the route I agree, but for the track it is better to take from the opencpn.ini parameters. selection of the track line width is needed in the track settings.
3. When the chart is moved (ENC / CM93), lines remain on the screen.
4. White frames when switching to night mode. The problem with version 4.8. White frames disappear when switching to full screen mode and back or minimize to a window and maximize to full screen.
5. The colors of the night mode for the indication line of the active Canvas Layout do not change.

Video
https://www.dropbox.com/s/lzgggbiln0il0pn/Video.mp4?dl=0

bdbcat, 2019-02-01:
3.  The display artifacts are related to the OpenGL configuration of your machine. We see this on some systems, rarely.  Not seen by me in some time.  Is OCPN 4.8.8 the same?  I guess not, or we would have heard about it.
Please post the section of OpenCPN logfile that describes OpenGL on your PC.

Thanks
Dave

bc38, 2019-02-01:
> 1. The display artifacts are related to the OpenGL configuration of your machine. We see this on some systems, rarely.  Not seen by me in some time.  Is OCPN 4.8.8 the same?  I guess not, or we would have heard about it.
>    Please post the section of OpenCPN logfile that describes OpenGL on your PC.
> 
> Thanks
> Dave
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/2820165/opencpn.log)

Today I will write down a new video with notes

bdbcat, 2019-02-04:
I see the video, thanks.  Problems are now clear.
I am most concerned about the grey lines at top and bottom of screen while dragging.  This may be caused by the specific GL adapter you use. (GeForce) I tried a correction for this case in the latest github master.  Please try this, and tell us how it works.
Thanks

bc38, 2019-02-04:
> Thanks

The strips are gone!

bc38, 2019-02-05:
Bugs fixed! Thank!
Dave, with the GRIB plugin, the problem is not completely solved.
Particle map friezes from the cursor position
Also strange indication appears OwerZoom
https://www.dropbox.com/s/hxqink0lsxx2vwe/GRIB.mp4?dl=0

closed 2019-02-12 by bc38