OpenCPN/OpenCPN #1250: Opencpn 4.8.8 & 4.8.4, Debian Stretch & Buster text field bug.

archviceroyofspackton, 2019-02-03:
What happens: Text can be entered into text fields.  However backspace, delete, the cursor keys, ctrl-C, ctrl-V etc. do not work.  The only way to alter text is by highlighting it with the mouse and overtyping or highlighting, right click and select delete.

What I expect to happen: Text boxes to behave like a normal textbox.

Where it happens: NMEA debug window filter field, detail fields for waypoint editor.

My system: Panasonic CF-52 toughbook (i5-540M), Fresh install of Debian stretch and buster.  This is the same irrespective of which ubuntu package I use (zesty, xenial, artful etc.) 

Other symptoms: There are no relevant errors present when opencpn is run from the terminal, just some complaints about ALSA.

leamas, 2019-02-14:
Hm... are you able to rebuild a fresh debian from git sources at https://salsa.debian.org/debian-gis-team/opencpn ? It would be interesting to see if this is a packaging problem, the salsa package is a completely different beast  packaging-wise.

If not, I can rebuild them for you. Just let me know if you are interested. 64-bit debian stretch, right?

leamas, 2019-02-16:
FWIW, I cannot reproduce this with "semi-official" packages built from  sources above on stretch/VirtualBox

did-g, 2019-03-17:
Hi,
It seems to be worse with 4.99 and wx 3.0, text fields don't work properly and updating route dialog waypoints can kill OpenCPN:
Xubuntu 18.04 
- select a waypoint row.
- click on the speed field.
- enter some value.
- click on another row while the cursor still in speed field.
- O is killed.

Work with wx 3.1.x github branch

commit f6934194 (2019-03-18): Disable in-place editing of route properties data view cells on wx3.0 due to it being buggy (Closes #1250)
closed 2019-03-18 by nohal