OpenCPN/OpenCPN #1259: Beta test code cutoff.

bdbcat, 2019-02-07:
All devs...
I plan to add a "PreBeta0" tag to master today.  This represents the cutoff line for code which will be included in the first Beta release.
We can of course continue to patch.  But the focus will shift to a wide-scale Beta test, with only clearly regression-safe patches merged as required.
So, this is a last call for code...

nohal, 2019-02-08:
I will have a "biggish" and "intrusive" patch touching the z-order hell no sooner than on Sunday (Shall it not break non-macOS too badly). But am not convinced anything has to wait for it...

bdbcat, 2019-02-08:
OK, thanks.  We will push ahead now.

transmitterdan, 2019-02-08:
@bdbcat 

I believe there is a bug regarding the flag m_bFollow getting out of sync with the ship icon in the MUI bar.

When O starts, the global m_bFollow flag is initialized false. Then this line:  ```m_bFollow = pcc->bFollow;``` in ```void ChartCanvas::ApplyCanvasConfig(canvasConfig *pcc)``` sets the follow flag according to the value obtained from the persistent configuration.  But the MUI bar does not yet exist so it cannot align the ship icon image with the follow state. In fact it doesn't even try to do it.

Steps to reproduce on Windows when GPS is moving the ship:

1) Edit opencpn.ini and set all instances of ```canvasbFollow`=1``
2) Start O and it will show a green ball on the MUI ship icon and canvas is in GPS follow mode.
3) Click somewhere on the canvas with the mouse.  The canvas will recenter but the MUI ship icon does not change state. The canvas is still following the ship but the ship is not centered.
4) Click the MUI ship icon.  It changes from green ball to red circle.  Canvas centers on the ship.  But canvas is no longer following the ship.

I think this is not the expected behavior.

bdbcat, 2019-02-08:
tdan:
    3\. Click somewhere on the canvas with the mouse.  The canvas will recenter but the MUI ship icon does not change state. The canvas is still following the ship but the ship is not centered.

This is new function.  One can drag the chart to change the relative bFollow position on the chart.  However, if you drag the chart such that the ship goes off-canvas, then the mode reverts to no-follow.
Dave

leamas, 2019-02-09:
hm... isn't this the time when we should fork a release branch with the described policy? In this way, there is a main development branch for further work, and a clean bug-fix only place for the release.

nohal, 2019-02-09:
Maybe in some other project, we historically do it the same way as wxWidgets - prereleases are tagged in master and the maintenance branch is created at the stable release.

leamas, 2019-02-09:
OK, fair enough.

closed 2019-02-26 by nohal