OpenCPN/OpenCPN #1265: PluginManager constructor crash due GetPrimaryCanvas()==null

taruti, 2019-02-13:
This occurs when starting opencpn under undefined behavior sanitizer. Apparently GetPrimaryCanvas returns NULL, which of course causes the chained GetNextContextMenu call to crash.

```
/.../OpenCPN/src/pluginmanager.cpp:365:86: runtime error: member call on null pointer of type 'struct ChartCanvas'
```

```
@ src/pluginmanager.cpp:363 @ PlugInManager::PlugInManager(MyFrame *parent)
    s_ppim = this;

    MyFrame *pFrame = GetParentFrame();
    if(pFrame)
    {
        m_plugin_menu_item_id_next = pFrame->GetPrimaryCanvas()->GetNextContextMenuId();
        m_plugin_tool_id_next = pFrame->GetNextToolbarToolId();
```

Just checking that GetPrimaryCanvas is not null avoids the crash, but not sure whether there is a deeper underlying bug. Does not occur without undefined behavior sanizer.

bdbcat, 2019-02-13:
Good catch.

This actually ends up as a static method call, but the code is obscure.
Fixed in github.

taruti, 2019-02-13:
Thanks for fixing this!

closed 2019-02-13 by taruti