OpenCPN/OpenCPN #1269: Compiling with Visual Studio 2019 Preview

transmitterdan, 2019-02-19:
In case anyone is adventurous and wants to compile with VS 2019 I have used it to build a working executable of current master branch. There is one "trick". During the Cmake V3.14 configuration process add the option -A win32 to the Cmake comand line.  Otherwise, Cmake now defaults to assumes it is targeting the same architecture as the build machine which is probably x64 so the Cmake configuration step falls over. Some new warnings have appeared.

In the "good news" department Microsoft kept linkage compatibility between VS2019 and VS2017.  Plugin binaries built with VS2017 will work with O built with VS2019.

In the "useless for now" news department I have also been able to build an executable of O for a Linux target on Microsoft "Bash for Windows". So now it's possible to build for Linux and Windows without resorting to multiple computers or virtual machines.  Have not been able to get O to run correctly on "Bash for Windows" due to X issues.  But the Linux binary runs without crashing.  A really handy feature is the ability to debug using Visual Studio while running on a target Linux platform. Visual Studio is much easier to use than gdb -tui.

Hakansv, 2019-02-19:
Dan, your in the front as usual. Nice info! 
This is the future path to the final death of XP support, I suppose.
For a native Windows user like me it sounds interesting to be able to target a Linux platform in that way. I'm not sure all are convinced? :)

leamas, 2019-02-19:
Great to hear that we can compile using the gcc toolchain under the Bash for Windows (BfW) environment. On the other hand, this is as expected, running compilers  is a core objective for BfW if I get this right.

Of course, there is more the one way to skin a cat. This is what I read from the Ocpn website:

> MinGW is Minimalist GNU for windows. There is a free compiler (gcc) which may be used to compile OpenCPN to run on windows. It can be installed from windows, or used as a cross compiler under linux. The run-time performance of OpenCPN is identical to the visual studio compiled version, the compiler produces more informative error messages, and uses less disk space. 

I don't know the status of this this. It would be nice if it worked also these days which would mean we could cross-compile a windows version under linux.

did-g, 2019-02-25:
@leamas 

> I don't know the status of this this. It would be nice if it worked also these days which would mean we could cross-compile a windows version under linux.

It does work, I'm using it with mxe for building a window 64bits version, but you can't ask other dev to keep an environment  for it and unless we are able to configure appveyor or travis for CI it'd bit rote quickly.
Last one is 1625377ec23d992003967d9f7b6c0d6c4a872748
```
+
+IF(WIN32)
+  ADD_CUSTOM_COMMAND(TARGET ${PACKAGE_NAME}
+    POST_BUILD
+    COMMAND mv "${CMAKE_BUILD_TYPE}/opencpn.pdb" "${CMAKE_BUILD_TYPE}/${PACKAGE_NAME}-${PACKAGE_VERSION}.pdb")
+ENDIF(WIN32)
+

```
No pdb with mingw so it should be 
IF(MSVC) rather than IF(WIN32)
not sure if trying to keep it working is worth the trouble.

BTW there's an issue with WX 3.1 and wxProgressDialog on windows, could be a wx bug or a dead lock with thread
it's brutal with climatology_pi (progress dialog is half drawn, grabs climatology focus and so on) but it's also used in grib_pi , even if showing very rarely.

If someone with a Window toolchain can look at it, checking on OSX would be nice too.

leamas, 2019-02-25:
@did-g hm... I have looked at an rather aggressive update of the cmake files for post 5.0.0; no idea if it will take me to a sane place.  Keeping/enhancing the mingw cross build then  needs to be part of this. 

It should be possible to configure Travis to do also a cross build,...As a  starter, I guess you are in the best position do do this, right?

BTW: shouldn't those things with wxProgressDialog and climatology_pi be filed as bugs?

nohal, 2019-02-25:
Unless we are able to have CI build using mingw, supporting it is a no-go at least for me as I simply have way too many environments to maintain already now.

rgleason, 2019-02-25:
Perhaps Sean will be able to help with these problems. He is a big proponent of mingw

leamas, 2019-02-26:
hm... we have actually hijacked this bug for mingw discussions. Partly my bad, sorry.  Let's move the mingw discussion to #1276 and leave this bug coping with builds based on Visual Studio.

transmitterdan, 2019-07-22:
Another issue is in CMakeLists.txt at about line 1084 in this line:

```    set(CMAKE_EXE_LINKER_FLAGS ${CMAKE_EXE_LINKER_FLAGS} /DEBUG)```

but it should be

```    set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} /DEBUG")```

CMake will insert a semicolon between strings in a set statement.  That will be a syntax error in later versions of Visual Studio.

leamas, 2019-07-22:
Perhaps it would be better to do it  properly while we are on it using using `list(APPEND CMAKE_EXE_LINKER_FLAGS  "/DEBUG")`instead. 

There is still a lot of work to do with this build code...

transmitterdan, 2019-07-24:
I think that will still put the ```;``` between the strings which is effectively the same as what happens now.  Thus it doesn't help.  I think you have to force the space between the strings.

What is a mystery to me is why this ```/DEBUG``` flag is needed at all. Cmake knows when it is building a debug version.

leamas, 2019-07-24:
>  I think you have to force the space between the strings.

Indeed, I stand corrected.

> What is a mystery to me is why this /DEBUG flag is needed at all

Also, it's done twice (around lines 481 and 1112). Besides this, it just seems to be the common case to add this flag.

transmitterdan, 2019-12-31:
CmakeLists.txt is now working with VS2019.

closed 2019-12-31 by transmitterdan