OpenCPN/OpenCPN #1276: mingw: No CI builds => no support

leamas, 2019-02-26:
There seem to be an interest to support the mingw build chain given the opportunities it provides.

However, as stated in #1269, in particular https://github.com/OpenCPN/OpenCPN/issues/1269#issuecomment-467051253, we cannot support the mingw build without working CI builds. 

Question the becomes how this could be accomplished.

leamas, 2019-02-26:
Creating reproducible builds in e.. g. Travis could perhaps be done:

  - By creating a dependency packet similar to the the buildwin tarball and build using
    that and native cross-compilers on a Debian Travis VM.
  - By using the mxe Makefile "from scratch"
  - By making the build in Fedora/CentOS using packaged compilers and libraries. These distros
    packages (almost?) all deps already.  Fedora s not directly supported by Travis but  according to
    https://derekweitzel.com/2016/05/03/building-centos-packages-on-travisci it should be possible
    to run it using docker. Even if all libraries are packaged(?) at least wxWidgets must be updated
    to 3.0, though.

Just my 5 öre...This is definitialy not for 5.0.

nohal, 2019-02-26:
As the assigned milestone says... Anyway, we can accept a PR implementing another toolchain on Travis at any moment, if it at least works.

did-g, 2019-03-01:
As someone who is actually building it I'm not sure it's worth the trouble. 
On the other hand an Android travis target for plugins...

leamas, 2019-03-02:
@did-g: hm... when you build using mingw, are you using the buildwin libs or something else?

did-g, 2019-03-03:
@leamas not I don't use buildwin libs but I'm using mingw for 64bits and there's no 64 bits version of them.

my notes on howto build :

[mxe.txt](https://github.com/OpenCPN/OpenCPN/files/2923031/mxe.txt)
I haven't used it from scratch on a new install so it's likely to fail.

leamas, 2019-03-03:
hm...thanks for sharing!

I have just made some  experiments  using the fedora toolchaiin.  Don't know where it will take me, besides some need to patch the build files.  The good is that almost all mingw deps aire alrady packaged in fedora. The bad is that it's fedora-centric. The ugly is the some strange linkage errors.

Hard to say which way (mxe or fedora) to prefer. I might have some time to investigate, though.

leamas, 2019-03-07:
Still extremely experimental, but i have a working  Travis builds based fedora 28 and mingw. It actually also builds the installer package, so that's a bonus compared to Appveyor. I haven't done any tests whatsoever.

This is on top of some massive and aggressive patches to the bulid system  which are far from complete. if it ever becomes possible to merge, it 's certainly after beta freeze.

The work is available in my clone: https://github.com/leamas/opencpn, in the the *ideas*  branch. The branch is not stable, if anyone is interested use for example the tag _ci-test-5_ which I don't plan to touch.

leamas, 2019-03-13:
The mingw Travis build is part of #1298.

PR #1335 (2019-04-09): Build: cleanup and mingw patches
Second part of #1298. Reorganizes the build targets, getting rid of the 2.8 style EXTRA_LIBS in favor of regular link commands and creates a working mingw build.

At his point the focus is on the build chain -- the mingw artifacts are not tested in any way.

IMHO, we need to let the dust settle after #1300 before merging this PR.

EDIT: Closes #1276
closed 2019-04-09 by nohal