OpenCPN/OpenCPN #1277: linux ships bell

bdbcat, 2019-02-28:
leamas...
I think this one is yours....
https://opencpn.org/flyspray/index.php?do=details&task_id=2498

On linux, portAudio. synchronous playing.  Some trouble with 

size_t SoundFileLoader::Get(void* buff, size_t length);

The member m_next never gets reset at the end of the wav data.  So sound will only ever play once.
Thanks
Dave

leamas, 2019-03-04:
Moving discussion from closed PR #1287

> Leamas...
> Better.
> Something still not quite right about Mac, but I will investigate. Builds OK, obviously.
>  But sounds are repeated/garbled.
> Dave

Hm... I looked at that code, my gut feeling is actually bad. First of all, the sequence  

     MacSound snd;  
     snd.playSound( path );  

is not a SystemCmdSound in any way  Secondly, does it block?  SystemCmd assumes that the command blocks, and if not we should have the  trouble you describe. Have you any pointer to the MacSound API? Google is not really friendly  on this subject...

Assuming PlaySound is non-blocking, there are still many ways out of this:
  - Writing a proper MacSound backend shouldn't be a big deal, but then we need some way to define  a callback when sound is completed.
  - There seem to be reasonably maintained portaudio OSX port (?)
  - afplay should be ubiquotious on OSX (?)

bdbcat, 2019-03-04:
leamas...
We need to find a way.
The alternative SystemCmdSounds (e.g. afplay) always interject some latency which is annoying to the user. We might not care about this, but users will worry about alarm sounds that have unpredictable latency. And of course, it makes the bells sound tough to get right. I prefer to think of SystemCmdSounds as a fall-back method if nothing else is found.
Don't know if portaudio implementation for modern Mac works, or not.  That may be a way...

leamas, 2019-03-04:
>The alternative SystemCmdSounds (e.g. afplay) always interject some latency 

Right...

> Don't know if portaudio implementation for modern Mac works, or not. That may be a way

I feel somewhat helpless here without proper Mac HW. That said, for what I can see portaudio is available using Homebrew so a test should reasonably simple (some CMakeLists patching should be trivial) 

I really wish I could be of more help...

keesverruijt, 2019-03-04:
Just butting in — I asked a friend and he suggested the AVAudioPlayer APIs in AVFoundation. I have never used them but online searches shows that this seems to have a simple API of the correct abstraction level.

leamas, 2019-03-04:
hm...thanks for sharing!  

That is https://developer.apple.com/documentation/avfoundation/avaudioplayer, right?

Looking at that API I see no synchronous  or callback capabiltiies. The closest is a function returning the estimated time to play a loaded sound which could be used  with a timer.  But that value is in seconds which might not be fine-grained enough. 

From my horizon, a portaudio test would be the first bet. Easy to say when I cannot do it  myself...

bdbcat, 2019-03-05:
leamas...
Looks like wxSound works fine on the Mac.  Implemented now in HEAD. Tested on physical hardware.
What's left:  I think we need a new backend implementation for Windows direct "PlaySound()" API call, currently hiding in SystemCmdSound.cpp

leamas, 2019-03-05:
> Looks like wxSound works fine on the Mac. 

Wow, that was a surpise...

> What's left: I think we need a new backend implementation for Windows direct "PlaySound()" API call

That shouldn't be that complicated using for example OcpnWxSound as a template. Reading the docs, I get impression that the sound file must be available in PATH in order to be used.  Doable, but strange. Perhaps I get it all wrong.

> currently hiding in SystemCmdSound.cpp

This isn't pushed, right?

bdbcat, 2019-03-05:
Leamas...

Yes, this is pushed already, as a proof of concept hack.  Needs correct implementation.

SystemCmdSound.cpp:44

leamas, 2019-03-05:
Oops,  missed that  one. If you want me too, I could certainly make a proper backend based on that. Or do you want to do it yourself?

bdbcat, 2019-03-05:
If you have the time, please implement this.
Thanks

leamas, 2019-03-05:
OK, I actually have som cycles available. Stay tuned...

leamas, 2019-03-05:
done: #1288

closed 2019-03-13 by bdbcat