OpenCPN/OpenCPN #1290: RPM package generation broken

leamas, 2019-03-06:
As heading says: On fedora 27 (and probably any RPM-based platform) *make package* fails. Attaching  logs.

At a glance, it looks like a CPack bug. The generated spec file contains

    %files
    %files -f %{name}.lang
    %files -f %{name}-chartdldr_pi.lang
    %files -f %{name}-dashboard_pi.lang
    %files -f %{name}-grib_pi.lang
    %files -f %{name}-wmm_pi.lang
    %defattr(-,root,root,-)

This is clearly in violation of the RPM spec  syntax rules since it basically tries to define multiple %files sections for the one and only default package -- it just doesn't make sense.

Here is also a warning about a not-installed license.txt 

I'm not that used to this situation, normally there is just one lang file per package. This would be the case also here if dropping each plugin in a subpackage, which seems to be the natural solution. That said, a single %files line with multiple "-f" options seems possible.

In the end, I'm not really concerned about this. If working RPM generation is required, a "real" spec file with some docs could be used to build proper rpms from github using release tags and/or  snapshots. RPMs built this way would have higher quality.

 And, for the RPM world (as well as most other Linuxes) flatpak is probably the way forward.

[build.log.gz](https://github.com/OpenCPN/OpenCPN/files/2934454/build.log.gz)
[opencpn.spec.gz](https://github.com/OpenCPN/OpenCPN/files/2934455/opencpn.spec.gz)

leamas, 2019-03-06:
To be clear: My suggestion is to close this bug as wont-fix.

SethDart, 2019-03-06:
we should use subpackages for each embedded plugin but it doesn't work fine with cmake automated packaging (at least, last time I checked)
I'm using manual spec files for all rpm I'm building for Fedora/EPEL.
I can publish those somewhere if someone is interested

leamas, 2019-03-09:
> we should use subpackages for each embedded plugin
Right.

That said, question is if we should support RPM at all in the long run. As it is, "support" means a CI job, probably Travis, producing rpms. There are two problems, each with a reasonable solution:

  - We cannot really add the work to build rpms to the regular builds. It just takes too long itme. The 
    proper solution would be to chain builds so that packages was built when OCPN was rebuilt. T
    Travis does nto support this, but id does support daily jobs. Such a job could check if there is
    a new OCPN version and rebuild if required.

  - Travis only supports Ubunut VMs. However, I have seen the light and learned how to run a 
    Fedora docker container inside the VM and build in that.

TL; DR; it should be possible to setup a separate github project which builds RPMS from OCPN sources and this verifies the build chain. Possible is not the same as clever, though.

leamas, 2019-03-11:
I just checked 4.8.6, and it exhibits the same error. Ergo:  since there has been no bug reports on this issue the rpms generated this way are not used.  Instead, packagers like @SethDart  and jcollie (at COPR) makes the downstream packaging which IMHO is just fine.

My gut feeling is that we should remove the attempt to let cmake/cpack build rpms since they are unusable. Proper downstream packaging is the way to go, be it at opencpn.org or somewhere else. And, removing the RPM stuff from CMakeLists would make it much more maintainable.

leamas, 2019-03-25:
@SethDart : If you have a working 5.0 spec file I could use that to build some CI jobs which rebuilds OCPN rpm, packages nightly or so as a first step to remove the broken cmake/cpack RPM generation.

dominig, 2019-04-23:
For info.
Even if I also use RPM as the packager for OpenSUSE, I do not care of rpm generation provided by OpenCPN.
I have my own package creation done directly from the git repo by the OpenSUSE CI (called OBS).
So let be clear, abandoning rpm generation, would not be an issue for OpenSUSE.

leamas, 2019-04-23:
Indeed. And you are not alone, the RPM generation was broken long time ago (at least in 4.8.6, havn't cared to check older releases). So, the generation as of  #1298 removed, and  presumably everybody is happy with that.

Thanks for input!

Closing.

closed 2019-04-23 by leamas