OpenCPN/OpenCPN #1294: MBTiles reported scale still needs improvements

nohal, 2019-03-08:
- The NOAA sets are reporting scale 1:70M, which makes them fall bellow pretty much all the world-wide sets, including Baikal's `WorldMap.mbtiles` (1:10M), which may be considered at the better end of the available sources.
- NOAA update sets are reporting much larger scale than deserved, but their main problem is that they should be blended into the base tileset. This probably deserves a separate issue.

bdbcat, 2019-03-08:
We could declare the reported scale of MBTiles to be based on the max zoom, instead of min zoom as we do now. A quick test looks OK for Baikal world and NOAA together.

Please try these changes for your favorite sets, before I commit it.

mbiltes.cpp line 532
      m_Chart_Scale = OSM_zoomScale[m_maxZoom];

line 1023
    // Do not render if significantly underzoomed
    if( VPoint.chart_scale > (20 * OSM_zoomScale[m_minZoom]))
        return true;

LeaseOnLife, 2019-03-08:
This is better! I still have O sometimes selecting the wrong mbtiles set in overlapping situations, but for that I can live with the 'hide' chart functionality.
Dirk

nohal, 2019-03-08:
@bdbcat I am happy, all the sets I consider reasonable quality seem to work fine so far.

closed 2019-03-08 by nohal