OpenCPN/OpenCPN #1309: All processors go to 100% on Raspberry PI 3/3+

andyrbarrow, 2019-03-12:
Running fine on RPI 3+ until incoming NMEA stream includes AIS and/or GPS data. Then processor (according to top) goes over 360% and stays there. Turning off NMEA stream brings CPU back to reasonable levels. Happens when using either MBTiles or KAP files.
[image]

nohal, 2019-03-12:
What is the configuration of your connections? Are you sure there is no data loop?

andyrbarrow, 2019-03-12:
It's on the bench right now, with a GPS puck and an Daisy AIS hat. I'll look at my KPlex configuration to ensure again that no loops exist.

andyrbarrow, 2019-03-12:
No loops

nohal, 2019-03-12:
We would like to see your connections setup and few seconds of the data stream coming.

nohal, 2019-03-12:
The screenshot also looks like you are using OpenGL, what driver?

did-g, 2019-03-12:
If available you can also run
perf top

andyrbarrow, 2019-03-12:
Just to be sure, I disconnected the serial AIS line and the USB GPS puck from KPLEX and routed them directly into OpenCPN. 

I disabled GL.

Same result.

Sorry, I can't get perf top to run.

nohal, 2019-03-12:
If you connect just the GPS? We have never seen or heard of this behavior as far as I can tell. What plugins are enabled? May I really see the connection settings?

andyrbarrow, 2019-03-12:
Here are the connection settings with both devices connected. I'll disconnect the AIS hat in a moment. I set both to priority 4 to see what effect it would have.
[image]

nohal, 2019-03-13:
Simply disabling the connection should be enough as OpenCPN will simply not use the port at all, there is no need to disconnect the device physically.

andyrbarrow, 2019-03-13:
Disconnected the AIS hat, pretty much same result. Unplugged the GPS puck, CPU usage went down to reasonable levels. Plug the GPS puck back in, CPU usage starts climbing, and then is back to 350%+. Crazy...

nohal, 2019-03-13:
`perf` is not available at all for your system or just not installed (I think it should be in `linux-perf` package on Raspbian, unless you are using some bizarre kernel that does not have it available)?

andyrbarrow, 2019-03-13:
linux-perf installed, but when I try to run it I get an error saying that the 4.19 version is required. Let me give it another shot.

andyrbarrow, 2019-03-13:
Sorry, despite searching, I can't find a raspbian linux-perf package for the 4.19 kernel.

nohal, 2019-03-13:
Yes, this seems to be happening on Raspbian. I wonder what sense does it make to keep releasing the bleeding edge kernels without their profiler built...
Anyway, when you previously mentioned using `kplex`, how was it configured? Was the incoming connection to OpenCPN still serial or TCP/UDP? I would like to discard the serial code (Which is about a year "new" and we haven't heard any similar reports about it, but still newer and less tested than the rest)...

bdbcat, 2019-03-13:
Andy...
Not clear what is going on here.  
I reproduced the environment almost exactly with real HW (rPI3, USB -serial Dongle, live data, GL, MBTiles charts).
And I see opencpn CPU load as about 5% statically, falling to about 3% if the ownship is offscreen.

For ideas:
1.  Does it change if the ship position is moved offscreen, and AIS is shutdown?
2.  Please post some of your NMEA debug stream, if possible.
Dave

andyrbarrow, 2019-03-13:
Okay, I did a completely clean Raspbian install, updated firmware, and installed OpenCPN 4.99. No problem with CPU utilization.

So, I guess I'm looking for something in OpenPlotter. When I get time, I'll do a completely clean OpenPlotter install and replace OpenCPN and see what happens. Thanks all for your attention and suggestions - I'm happy to close this issue.

closed 2019-03-13 by andyrbarrow