OpenCPN/OpenCPN #1310: Take over the snap

nohal, 2019-03-12:
Finish the transition of the Snap package to the project

leamas, 2019-03-17:
Do we need some priorities here? There is snap (i. e., this bug), broken RPM.s (#1290) and flatpak (#1115). Given that we have Debian packages one way or another all these issues lo a large extent addresses the same non-Debian linux users.

From my perspective:

  - Drop the RPM package generation. It's just broken, and nobody uses it. 
  - Make a decision to prioritize either snap or flatpak.,  they are overlapping, 

Supporting either flatpak or snap at the OCPN upstream should be enough and a better use of limited resources. 

And,  whichever system which would get a lower priority t  would probably be done downstream anyway. The only reason I havn't published the flatpak package at flathub is that it's still unclear if you want it to be puiblished upstream instead.

nohal, 2019-03-17:
The snap already exists, is not being maintained by the author and occupies the upstream name. The least damage will be done if we take it over, not that I'm any happy doing it...

leamas, 2019-03-17:
> The snap already exists, 

Which could motivate to focus on it and downplay the flatpak package.

> is not being maintained by the author and occupies the upstream name.

Is the maintainer cooperative? Otherwise, it might become tricky to take over the name...

....

nohal, 2019-03-17:
> Which could motivate to focus on it and downplay the flatpak package.

You are the expert here, for me both are just weird attempts to address a problem I personally don't have.

> Is the maintainer cooperative? Otherwise, it might become tricky to take over the name...

Yes, we are talking to him, the thing just did not happen yet for lack of time and priority

leamas, 2019-03-17:
For flatpak, I'm stuck -- I need input from you or Dave (see #1115).   

If we have a working snap package I could livei with that instead -- it basically solves the same problem for more or less the same set of users.

bdbcat, 2019-03-17:
All....
Here is some high-level direction:
1.  No alternative packaging systems will be available for O5 initial release.  This effort will continue as a WIP.
2.  Drop the broken RPM package build option.
3.  If flatpak is trivially completed, finish that. Otherwise, drop flatpak.
4.  Attempt to get the snap package moved to us.
5.  If he effort to get snap going proves intractable, revert to flatpak.
6.  For either option, if we need a location to host any resulting alternative packages, we can do that in the opencpn CDN.

Dave

leamas, 2021-04-13:
@nohal: The snap package is still on 4.8, and not that visible.  Of course, the outdated package is a problematic since it occupies the opencpn namespace. OTOH, there is a very visible flatpak package, and also a native Debian package.

Given the amount of open issues: could it be right to just drop this issue and close it?

nohal, 2021-04-14:
I think so, unless someone wants to actually maintain the snap (Which I certainly don't).

bdbcat, 2021-04-14:
Done.  No need for snap.  We support flatpak.

closed 2021-04-14 by bdbcat

leamas, 2021-05-16:
If this should surface again it is possible to claim the name:  https://forum.snapcraft.io/t/q-process-for-upstream-to-claim-snap-package-name/24429.