OpenCPN/OpenCPN #1313: Fullscreen mode: Window decorations hidden behind toolbars on macOS

nohal, 2019-03-13:
On macOS the global menu and window decorations slide down from the top of the screen in fullscreen mode but are unusable because OpenCPN toolbars stay in front of them.
[image]

bdbcat, 2019-03-13:
Pavel...
Do you mean "OCPN Full Screen" or "Green button" full screen.  I see different bahviour for the two.
Dave

nohal, 2019-03-13:
Dave...

I mean all the 3 colored dots in the left upper corner of a window. In fullscreen they slide from the top together with the menu and should be on top of the app content. In this case they are behind the toolbars (you see that the dark grey title, but it's whole left part is missing)

More apparent here (The red and yellow are not accessible because of our toolbar in this case, you can at least exit fullscreen):
[image]

bdbcat, 2019-03-13:
Pavel...
Experimenting.
On Sierra, I never get a wxMaximizeEvent when pushing the green button.  Works OK on gtk, though.
Make it difficult to detect this condition.
Tried various incantations of toolbar (wxFrame) properties. No help.
Still thinking.

bdbcat, 2019-03-17:
Corrected in https://github.com/OpenCPN/OpenCPN/commit/c44652ce992204e8b0c8b5c026782a33b69c8bbf

closed 2019-03-17 by nohal