OpenCPN/OpenCPN #1325: crash due to multiple canvas support?

seandepagnier, 2019-03-20:
[code]
(gdb) bt
#0  0xc00b24c0 in ?? ()
#1  0xaacd52a2 in ChartCanvas::SetViewPoint (this=0xab569000, lat=32.708217898096862, lon=-64.01071224437672,
    scale_ppm=0.0001393642685172381, skew=0, rotation=0, projection=1, b_adjust=true, b_refresh=true)
    at /home/rock/OpenCPN/src/chcanv.cpp:4995
#2  0xaacd4c0c in ChartCanvas::SetVPScale (this=0xab569000, scale=0.0001393642685172381, refresh=true)
    at /home/rock/OpenCPN/src/chcanv.cpp:4888
#3  0xaacd39ce in ChartCanvas::DoZoomCanvas (this=0xab569000, factor=1.0001, can_zoom_to_cursor=true)
    at /home/rock/OpenCPN/src/chcanv.cpp:4487
#4  0xaac6a7d6 in MyFrame::ProcessOptionsDialog (this=0xab441ca0, rr=8192, pNewDirArray=0xb429d4d8)
    at /home/rock/OpenCPN/src/chart1.cpp:6290
#5  0xaac69c30 in MyFrame::DoOptionsDialog (this=0xab441ca0) at /home/rock/OpenCPN/src/chart1.cpp:6014
#6  0xaac63674 in MyFrame::DoSettings (this=0xab441ca0) at /home/rock/OpenCPN/src/chart1.cpp:4849
#7  0xaac6268a in MyFrame::OnToolLeftClick (this=0xab441ca0, event=...) at /home/rock/OpenCPN/src/chart1.cpp:4455
#8  0xf67dd824 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
[/code]

m_singleChart is garbage. 
[code]
(gdb) p *m_singleChart
$6 = {_vptr.ChartBase = 0xac0f85f8, m_Chart_Scale = -1414639816, m_ChartType = CHART_TYPE_UNKNOWN,
  m_ChartFamily = CHART_FAMILY_UNKNOWN, m_FullPath = {static npos = <optimized out>, m_impl = {
      static npos = 4294967295,
      _M_dataplus = {<std::allocator<wchar_t>> = {<__gnu_cxx::new_allocator<wchar_t>> = {<No data fields>}, <No da\
ta fields>}, _M_p = 0xd10900e <error: Cannot access memory at address 0xd10900e>},
      _M_string_length = 2416642704, {_M_local_buf = L
[/code]
Nothing testing this in ChartDB::DeleteCacheEntry.   What if one canvas deletes the single chart used on the other??

Not sure why it would purge a chart being used so this could be a bigger problem even if the crash is fixed.

did-g, 2019-03-23:
Indeed
da82d3 remove all cache locking in single chart mode (chartdb.cpp).

did-g, 2019-03-23:
Made a PR with a dirty fix #1328 
For various reasons in single chart mode, charts aren't protected by reference counting and can be evicted from memory while displayed and referenced in canvas.

A proper fix could be:
get ride of m_singleChart

Single chart mode is a quilt mode with:
- a locked reference chart
- only one chart in the quilt, the reference chart
- piano with squared corners
- no limit on zoom
- no overlay?

Doing this would remove thousands lines of very, very convoluted code.

bdbcat, 2019-03-24:
did...
I like the idea.  Quilting was essentially patched onto the legacy single-chart mode, without really knowing where it would all end up.
I think your list defining single-chart mode is a good start.  There will be some (lots) of spaghetti to clean up, but now is a good time to start.
Dave

seandepagnier, 2019-03-24:
I think if it is done, it is good ensure the efficiency of single
chart mode is not lost.   It runs a lot faster than quilted mode
currently.

bdbcat, 2019-03-24:
This should be no problem.
The expensive part of quilt is, and always has been, the successive region calculations to build the stack.  Not needed if we include only the reference chart.
One complication will be that quilt respects the bounds declared by the coverage regions in the KAP header.  In single chart mode, we want to see all of the raster image, to read margin info, etc.  This is usually clipped by the KAP coverage regions.

rgleason, 2023-03-26:
Can this be closed?