OpenCPN/OpenCPN #1326: Crash in wmm_pi::SetPositionFix  | wxImage::InitAlpha()

leamas, 2019-03-21:
As heading says:

    #0  0x00007ffff3c76f74 in wxImage::InitAlpha() ()
        at /lib64/libwx_gtk2u_core-3.0.so.0
    #1  0x00007fffc00c48ce in wmm_pi::SetPositionFix(PlugIn_Position_Fix&) (this=0x35397d0, pfix=...)
        at /builddir/build/BUILD/OpenCPN-4.8.8/plugins/wmm_pi/src/wmm_pi.cpp:589
    #2  0x0000000000796405 in PlugInManager::SendPositionFixToAllPlugIns(GenericPosDatEx*) (this=0x115f800, ppos=ppos@entry=0x7fffffffd0b0)
        at /usr/include/wx-3.0/wx/dynarray.h:837
    #3  0x0000000000582226 in MyFrame::OnFrameTimer1(wxTimerEvent&) (this=0x1159fa0, event=...) at /home/mk/src/OpenCPN/src/chart1.cpp:7157
    #4  0x00007ffff35e3402 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /lib64/libwx_baseu-3.0.so.0
    #5  0x00007ffff35e3513 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /lib64/libwx_baseu-3.0.so.0
    #6  0x00007ffff35e3870 in wxEvtHandler::TryHereOnly(wxEvent&) ()
        at /lib64/libwx_baseu-3.0.so.0
    #7  0x00007ffff35e38fb in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
        at /lib64/libwx_baseu-3.0.so.0
    #8  0x00007ffff35e3991 in wxEvtHandler::ProcessEvent(wxEvent&) ()
        at /lib64/libwx_baseu-3.0.so.0
    #9  0x00007ffff35e372b in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
        at /lib64/libwx_baseu-3.0.so.0
    #10 0x00007ffff3551215 in wxTimerImpl::SendEvent() ()
        at /lib64/libwx_baseu-3.0.so.0
    #11 0x00007ffff3b43615 in timeout_callback ()
        at /lib64/libwx_gtk2u_core-3.0.so.0
    #12 0x00007ffff78833e1 in g_timeout_dispatch () at /lib64/libglib-2.0.so.0
    #13 0x00007ffff788291d in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
    #14 0x00007ffff7882ce8 in g_main_context_iterate.isra ()
        at /lib64/libglib-2.0.so.0
    #15 0x00007ffff7883012 in g_main_loop_run () at /lib64/libglib-2.0.so.0
    #16 0x00007ffff564ae5f in gtk_main () at /lib64/libgtk-x11-2.0.so.0
    #17 0x00007ffff3b31555 in wxGUIEventLoop::DoRun() ()
        at /lib64/libwx_gtk2u_core-3.0.so.0
    #18 0x00007ffff34b5021 in wxEventLoopBase::Run() ()

While certainly not a  solution, following patch walks around the crash:

    diff --git a/src/chart1.cpp b/src/chart1.cpp
    index c6cfe1085..53a07425a 100644
    --- a/src/chart1.cpp
    +++ b/src/chart1.cpp
    @@ -7154,7 +7154,7 @@ void MyFrame::OnFrameTimer1( wxTimerEvent& event )
 
             GPSData.FixTime = m_fixtime;
 
    -        g_pi_manager->SendPositionFixToAllPlugIns( &GPSData );
    +//        g_pi_manager->SendPositionFixToAllPlugIns( &GPSData );
         }
 
         //   Check for anchorwatch alarms                                 // pjotrc 2010.02.15

Fedora 29, on a Xorg display.

nohal, 2019-03-22:
What should I do to reproduce this? Except installing Fedora 29 to see the wxWidgets package it contains is more broken than any other, which we kind of already know, of course.

leamas, 2019-03-22:
The problem w Fedora is that they (we) live at the frontier, so there is no wxWidgets 3.x built on GTK2 available -- Fedora has moved to GTK3 since long. So,I have  rebuilt a  wxWidgets 3.0/GTK2 package called wx3GTK;  it is available at https://copr.fedorainfracloud.org/coprs/leamas/opencpn-mingw/.  

Yes, the name of the repo is weird, I know.

nohal, 2019-03-22:
I honestly miss the point here. If we have to build our wx packages here anyway, why don't we build wx3.1 against gtk3 and head to the inevitable future instead?

leamas, 2019-03-22:
Because as of now wxWidgets  doesn't seem to be ready for gtk3,  see #1166 (notably the upstream bug).

leamas, 2019-03-22:
Ah, that's not the bug, that's about Wayland. Please dont get distracted by that one, sorry.

TBH, I don't remember all details, but at some point I gave up getting Debian to work with gtk3. So, I have stuck to the same strategy for Fedora with the  overall idea to stick with GTK2 in this cycle  and do the inevitable switch in next release.

I am by no means sure that this is the proper approach is the right one...

leamas, 2019-03-25:
I'm closing this for now. There is no working RPM packaging we can use as common ground so anything happening on a RPM platform is hard to reproduce. 

There is of course *make install*,  but without  *make uninstall* support it's also hard to use in a reproducible way.

The general RPM issue is enshrined in #1290

closed 2019-03-25 by leamas