OpenCPN/OpenCPN #1329: 5.0.0 build failure in Mageia: 'wxWebView' does not name a type'

barjac, 2019-03-25:
Output below:
`In file included from /home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/include/AboutFrameImpl.h:28,
                 from /home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/src/chart1.cpp:83:
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/include/AboutFrame.h:64:3: error: 'wxWebView' does not name a type; did you mean 'wxListView'?
   wxWebView* m_htmlWinHelp;
   ^~~~~~~~~
   wxListView
In file included from /home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/src/chart1.cpp:83:
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/include/AboutFrameImpl.h: In member function 'virtual void AboutFrameImpl::m_btnBackOnButtonClick(wxCommandEvent&)':
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/include/AboutFrameImpl.h:44:60: error: 'm_htmlWinHelp' was not declared in this scope
     void m_btnBackOnButtonClick( wxCommandEvent& event ) { m_htmlWinHelp->GoBack(); m_btnBack->Enable(m_htmlWinHelp->CanGoBack()); }
                                                            ^~~~~~~~~~~~~
/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/include/AboutFrameImpl.h:44:60: note: suggested alternative: 'm_htmlWinLicense'
     void m_btnBackOnButtonClick( wxCommandEvent& event ) { m_htmlWinHelp->GoBack(); m_btnBack->Enable(m_htmlWinHelp->CanGoBack()); }
                                                            ^~~~~~~~~~~~~
                                                            m_htmlWinLicense
make[2]: *** [CMakeFiles/opencpn.dir/build.make:66: CMakeFiles/opencpn.dir/src/chart1.cpp.o] Error 1
make[2]: Leaving directory '/home/baz/rpmbuild/BUILD/OpenCPN-5.0.0/build'
make[2]: *** Waiting for unfinished jobs....`

Deps seem OK so I'm stuck - any help appreciated.

`-- Found wxWidgets: -L/usr/lib64;-pthread;;;-lwx_gtk2u_xrc-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_qa-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu_xml-3.0;-lwx_baseu_net-3.0;-lwx_baseu-3.0 (found version "3.0.4") 
-- Found wxWidgets webview add-on
-- Found wxWidgets...
--  wxWidgets Include: /usr/lib64/wx/include/;/usr/lib64/wx/include/gtk2-unicode-3.0;/usr/include/wx-3.0
--  wxWidgets Libraries: -L/usr/lib64;-pthread;;;-lwx_gtk2u_xrc-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_qa-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu_xml-3.0;-lwx_baseu_net-3.0;-lwx_baseu-3.0`

nohal, 2019-03-25:
I don't know Mageia and if it contains a package for `wxWebView`, which is what you are missing. The easy way around it if it does not is to comment out https://github.com/OpenCPN/OpenCPN/blob/master/CMakeLists.txt#L1059

closed 2019-03-25 by nohal

barjac, 2019-03-26:
Thanks - I have commented that line, the build progresses further now but we hit linking problems.
The full build log is here: http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.0.0-1.mga7.src.rpm/build.0.20190326183620.log

nohal, 2019-03-26:
This seems like `wxWidgets` in Mageia is also configured without support for AUI and OpenGL. These unlike the optional `wxWebView`, are hard requirements for OpenCPN.

nohal, 2019-03-26:
It looks like both AUI and OpenGL support are provided as [separate packages](https://madb.mageia.org/package/list/t_search/wx/release/cauldron/application/0) `libwx_gtk2u_aui3.0_0` and libwx_gtk2u_gl3.0_0 in something called Mageia Cauldron. In stable Mageia 6 AUI support is either missing or in something I don't see and OpenGL support seems to be in https://madb.mageia.org/package/show/name/libwxgtkugl3.0_0/release/6/application/0

SethDart, 2019-03-27:
I've the same problem with Fedora:
cmake set wxWidgets use libs: gl;net;xml;html;adv;aui;core;base;webview
But I get wxWidgets Libraries: -pthread;;;-lwx_gtk3u_xrc-3.0;-lwx_gtk3u_webview-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_qa-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_core-3.0;-lwx_baseu_xml-3.0;-lwx_baseu_net-3.0;-lwx_baseu-3.0
I had to force
SET( wxWidgets_LIBRARIES "${wxWidgets_LIBRARIES};-lwx_gtk3u_aui-3.0;-lwx_gtk3u_gl-3.0")
To be able to link properly

nohal, 2019-03-27:
The not found parts of wxWidgets even if present seem to be connected to very new cmake versions (I see it with 3.14.0) and the final solution should probably be something like this:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index fcec1191a..8df2a9e21 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -408,9 +408,9 @@ IF((_wx_selected_config MATCHES "qt-armv7"))
 ENDIF()

 IF((_wx_selected_config MATCHES "qt-armv7"))
-  SET(wxWidgets_USE_LIBS base core xml html adv aui)
+  SET(wxWidgets_FIND_COMPONENTS base core xml html adv aui)
 ELSE()
-  SET(wxWidgets_USE_LIBS net xml html adv aui core base webview)
+  SET(wxWidgets_FIND_COMPONENTS net xml html adv aui core base webview)
 ENDIF()

 OPTION (USE_GL "Enable OpenGL support" ON)
@@ -436,7 +436,7 @@ IF (ARCH MATCHES "arm*" AND (NOT QT_ANDROID) AND CMAKE_SYSTEM_NAME MATCHES ".*Li
     SET(OPENGLES_FOUND "YES")
     SET(OPENGL_FOUND "YES")

-    SET(wxWidgets_USE_LIBS ${wxWidgets_USE_LIBS} gl )
+    SET(wxWidgets_FIND_COMPONENTS ${wxWidgets_FIND_COMPONENTS} gl )
     add_subdirectory(src/glshim)

     SET(OPENGL_LIBRARIES "GL_static" "EGL" "X11" "drm"  )
@@ -455,7 +455,7 @@ IF(QT_ANDROID)
   SET(OPENGLES_FOUND "YES")
   SET(OPENGL_FOUND "YES")

-  SET(wxWidgets_USE_LIBS ${wxWidgets_USE_LIBS} gl )
+  SET(wxWidgets_FIND_COMPONENTS ${wxWidgets_FIND_COMPONENTS} gl )
   add_subdirectory(src/glshim)
 ENDIF(QT_ANDROID)

@@ -468,7 +468,7 @@ IF ((NOT OPENGLES_FOUND) AND (NOT QT_ANDROID))
   ENDIF(USE_GL)

   IF(OPENGL_FOUND)
-    SET(wxWidgets_USE_LIBS gl ${wxWidgets_USE_LIBS} )
+    SET(wxWidgets_FIND_COMPONENTS gl ${wxWidgets_FIND_COMPONENTS} )
     INCLUDE_DIRECTORIES(${OPENGL_INCLUDE_DIR})

     MESSAGE (STATUS "Found OpenGL...." )
@@ -529,14 +529,14 @@ IF(NOT QT_ANDROID)
   ENDIF(MSVC)

   IF(WIN32 OR APPLE OR QT_ANDROID)
-    FIND_PACKAGE(wxWidgets)
+    FIND_PACKAGE(wxWidgets COMPONENTS ${wxWidgets_FIND_COMPONENTS})
     if (wxWidgets_FOUND)
       message(STATUS "Found wxWidgets webview add-on")
       add_definitions(-DOCPN_USE_WEBVIEW)
     else ()
-      list(REMOVE_ITEM wxWidgets_USE_LIBS webview)
+      list(REMOVE_ITEM wxWidgets_FIND_COMPONENTS webview)
       message(STATUS "Could not find wxWidgets webview add-on")
-      FIND_PACKAGE(wxWidgets REQUIRED)
+      FIND_PACKAGE(wxWidgets REQUIRED COMPONENTS ${wxWidgets_FIND_COMPONENTS})
     endif ()
     IF(MSYS)
       # this is to convert msys to windows paths, and handle the missing /usr
@@ -1053,14 +1053,14 @@ IF(NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
     set(wxWidgets_CONFIG_OPTIONS ${wxWidgets_CONFIG_OPTIONS} --toolkit=gtk3)
     MESSAGE(STATUS "Building against GTK3...")
   ENDIF(GTK2_FOUND)
-  FIND_PACKAGE(wxWidgets)
+  FIND_PACKAGE(wxWidgets COMPONENTS ${wxWidgets_FIND_COMPONENTS})
   if (wxWidgets_FOUND)
     message(STATUS "Found wxWidgets webview add-on")
     add_definitions(-DOCPN_USE_WEBVIEW)
   else ()
-    list(REMOVE_ITEM wxWidgets_USE_LIBS webview)
+    list(REMOVE_ITEM wxWidgets_FIND_COMPONENTS webview)
     message(STATUS "Could not find wxWidgets webview add-on")
-    FIND_PACKAGE(wxWidgets REQUIRED)
+    FIND_PACKAGE(wxWidgets REQUIRED COMPONENTS ${wxWidgets_FIND_COMPONENTS})
   endif ()
   MESSAGE (STATUS "Found wxWidgets..." )
   MESSAGE (STATUS " wxWidgets Include: ${wxWidgets_INCLUDE_DIRS}")
```

but it won't make it to the 5.0 branch for now as I'm not able to assess the breakage it will cause elsewhere...

barjac, 2019-03-27:
Thanks for the patch - 5.0.0 is now building without error in Mageia Cauldron without the commented line mentioned earlier, and also installs and runs OK, so for us this seems fixed.
Thanks for your help. :)

bdbcat, 2019-03-27:
Pavel...
I wrote a FindwxWebView.cmake module to fix the Trusty build. Has not made it to github yet, still testing.
Looks like it will work.

hindsholm, 2019-04-02:
I can confirm that the patch provided by @nohal also works on ArchLinux, so it would be great to have it in the release.