OpenCPN/OpenCPN #1330: AIS target drawn in grey with no track but still moving on the canvas

transmitterdan, 2019-03-27:
I am seeing what I think is a bug of some kind in the AIS display.  In the below screen shot the vessel Juanita is moving and has a valid fix.  But the ship is drawn "attenuated", in grey color and has no track line drawn which means the vessel's last known fix has expired.  Obviously the fix is actually recent and valid as the vessel can be seen moving on the canvas.  This was observed after about 10 hours continuous running with several AIS targets.  Another instance of O5 was started and using the identical NMEA stream. Juanita was drawn in regular attenuated green/yellow as it should be.

Should a moving AIS target ever be colored grey?

[image]

transmitterdan, 2019-03-27:
I think I have tracked this down to this small section of code:

```
void AIS_Decoder::UpdateOneTrack( AIS_Target_Data *ptarget )
{
   if( !ptarget->b_positionOnceValid ) return;
    // Reject for unbelievable jumps (corrupted/bad data)
    if ( ptarget->m_ptrack->GetCount() > 0 )
    {
        AISTargetTrackPoint *LastTrackpoint =  ptarget->m_ptrack->GetLast()->GetData();
        if ( fabs( LastTrackpoint->m_lat - ptarget->Lat ) > .1  || fabs( LastTrackpoint->m_lon - ptarget->Lon ) > .1 )
        {
            // after an unlikely jump in pos, the last trackpoint might also be wrong
            // just to be sure we do delete this one as well.
            ptarget->m_ptrack->pop_back();
            ptarget->b_positionDoubtful = true;
            return;
        }        
    }
```
It was changed by this commit c388545f31969abb68ffbc1c0e8d864add9be794

I believe the issue is caused by the testing method of replaying an NMEA stream in a continuous loop. When the stream gets to the end and loops back most of the AIS targets "jump" position.  The jump causes UpdateOneTrack() to flag the new position as "doubtful".  Then it pops one track point off the track list and returns.  It keeps doing that until the track list is empty.

I think the solution is to pop the last (possibly corrupt) track point and then don't return.  Go ahead and add the current position report onto the end of track list.

Granted, this issue doesn't happen in real life that often but when it does happen it can take long time to clear the b_positionDoubtful flag.

bdbcat, 2024-02-27:
The proposed solution will surely fix the problem, when running a VDR playback.
But it will break the essential function of this code clause.
Imagine a single real AIS target reporting a bogus position, once.  The proposal would add that bogus position to the track, which is clearly not desired.
I see no easy solution here, without requiring the track code having intimate knowledge of the VDR playback state.  It might detect the restart of the VDR loop, and handle this special case accordingly.  This seems overkill.
I propose NOTABUG, WONTFIX disposition here.
Thoughts?

Hakansv, 2024-02-27:
I suppose we're talking about this feature:
[image]
If playing VDR it may be a good advice to switch it off.
For real AIS reception the symbol will be deleted when AIS ship time for mark as lost is expired if no speed less than the limt is received before that.
Agree to not a bug.

Hakansv, 2024-02-28:
@transmitterdan Sorry, to fast conclusions by the word "grey" instead of reading carefully. Thanks for your gentle correction instead of a "silly you" :)