OpenCPN/OpenCPN #1331: ENC Light Colors

rgleason, 2019-03-27:
Report from David Burch, Starpath:
I am noting some issues with light colors on ENC charts. What would be best here. Direct note to Dave Register or post a note in the forum?

There are yellow lights that are not getting their colors stated (so far i see this mostly on minor lights) but i also see a major light listed as yellow that is actually white.

Also the new v5 is still missing the lights and buoy labels. There is an option to turn them on, but it does not seem to do anything.
thanks —david

I don't know if this has been noticed/addressed.

nohal, 2019-03-27:
We need specific info about the objects that are supposed to be erroneous (Chart number, object query window screenshot), without it, I would say these issues seem to be operator error and are 
not present on my OpenCPN 5.0 system as you can confirm on the attached screenshot.
[image]

rgleason, 2019-03-30:
Pavel, I've written and asked David Burch to sign up for free github opensource account and to respond to this.

starpathnav, 2019-03-31:
I will come back to the issue of light color specifications in ENC, which is a different topic...  First may I ask about the easy one, and note the good news is you can see the labels, so i am doing something wrong, or missing some setting.  i have checked what i thought would be the obvious ones, ie turn on the labels, but i do not see the labels. Is there another setting that is required?  thanks.  attached is the same chart as shown above as i see it.
[image]

nohal, 2019-03-31:
May you attach your config file and logfile, please?

rgleason, 2019-03-31:
Also David,
Options > Vector Chart Display >
Chart Texts check National Text on Chart etc
Select  "Reset to Standard" which checks the text box.

starpathnav, 2019-03-31:
Rick, i had not done that, but just now did, and see no effect. Pavel, i will get the files shortly. I am just finishing up notes on the yellow-white light point i wanted to raise, which as noted is a different topic.
[image]

starpathnav, 2019-03-31:
Here are the log file and config (ini) file, and a pdf with notes on the question i have on another topic about the light colors. I apologize that these two different subjects were asked at the same time originally. 

[Also i should note that whatever i am doing wrong with the labels, i am doing wrong in both PC and Mac versions.]

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/3026264/opencpn.log)
[opencpn.ini.zip](https://github.com/OpenCPN/OpenCPN/files/3026268/opencpn.ini.zip)
[opencpn yellow v white lights.pdf](https://github.com/OpenCPN/OpenCPN/files/3026269/opencpn.yellow.v.white.lights.pdf)

nohal, 2019-03-31:
To see the "missing" text, go to the Toolbox and uncheck the "Important Text Only" checkbox
[image]

closed 2019-03-31 by nohal

starpathnav, 2019-04-01:
Thanks for that valuable note. On the Mac version this does turn on the labels. No luck yet with the PC version.  I tried complete uninstall and reinstall of v5 PC version and "important text only" was not checked, and no labels, then i checked it and no labels, and then unchecked again and no labels, rebooted and still the same.  It seems this "important text" button is a key factor, but there is something else involved that i am still missing.  I apologize for missing something that must be obvious to most users. 

[image]

starpathnav, 2019-04-01:
Thanks very much. That does the job. To show labels, we need three checks: (1) show text, (2) show labels, and (3) un check or leave unchecked  "Important text only."  Thanks again for tracking down my errors on this. I was thinking of chart text and object labels as different things.  I will add this to our support files on the program. I hope to add a video on use of ENC as soon as i can.

rgleason, 2019-04-01:
Thats good chart text is resolved. Perhaps more detail about the Enc light colors?
Otherwise can this be closed?

rgleason, 2019-04-01:
David, I just realized, that with dual screens and v50 this all can be controled from the "Chart Panel Options"
Pick "Standard" Display Category  and check the  Show Texts, Show Depths, and Bouy Light Labels!

IF you pick 'User Standard' Display Category you must have certain items checked in the OPTIONS menu.

starpathnav, 2019-04-01:
Yes, the view light/buoy  labels is resolved. Thanks for that.  This was my lack of knowledge on how the controls interact.  If i can figure how to help with the online manual, i will add a note of clarification on this, which i think would be worth it.  To show light and buoy labels requires 3 check boxes to be set properly; not just show labels. (One could maybe discuss the logic of that, but the key factor is knowing how to turn them on.)

As for distinguishing white and yellow lights in the  object query, that is not resolved.  Also now that i see the labels, these light colors are not resolved in the labels either.  The light color issue is illustrated in the attached pdf above.

nohal, 2019-04-01:
Are you running the latest CI build?

nohal, 2019-04-01:
The bugfixes do not get magically applied to your machine and we do not make a new release for each... If you want to test the latest automatic builds you can always wind it at https://ci.appveyor.com/project/bdbcat/opencpn/build/artifacts and at https://ocpnci.kalian.cz for macOS. They of course receive no testing at all and may be severely broken at any given moment.

starpathnav, 2019-04-01:
Fantastic!  That was fast.  Both lights mentioned, which are two different types, now show perfectly.  Thanks for that.  We will look for this in the next official build.

All aspects of this topic are now closed.

OpenCPN is very fortunate to have such wonderful support.

commit 99d158d8 (2019-03-31): Implement support for more light colors in chart object query dialog. Fixes #1331