OpenCPN/OpenCPN #1332: Crash on exit (macOS)

keesverruijt, 2019-03-28:
Usually O5 seems very well behaved but I guess I ran into a race condition today, exiting from O5 (the exact commit of the v5 release, but compiled locally for debugging.)

The stack trace is:

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libwx_baseu-3.1.dylib         	0x00000001022ee2e4 wxString::empty() const + 52 (string:1255)
1   libwx_baseu-3.1.dylib         	0x00000001022ee411 wxAppConsoleBase::GetAppDisplayName() const + 49 (appbase.cpp:242)
2   libwx_osx_cocoau_core-3.1.dylib	0x0000000101a5b23e wxLogGui::GetTitle() const + 974 (logg.cpp:239)
3   libwx_osx_cocoau_core-3.1.dylib	0x0000000101a5b5c3 wxLogGui::Flush() + 243 (logg.cpp:302)
4   libwx_baseu-3.1.dylib         	0x0000000102390a87 wxLog::FlushActive() + 87 (log.cpp:804)
5   libwx_baseu-3.1.dylib         	0x000000010237e8bb DoCommonPostCleanup() + 59 (init.cpp:430)
6   libwx_baseu-3.1.dylib         	0x000000010237e80b wxEntryCleanup() + 107 (init.cpp:463)
7   libwx_baseu-3.1.dylib         	0x000000010237edd9 wxUninitialize() + 57 (init.cpp:568)
8   libwx_baseu-3.1.dylib         	0x000000010237f2be wxInitializer::~wxInitializer() + 30 (init.h:109)
9   libwx_baseu-3.1.dylib         	0x000000010237eb95 wxInitializer::~wxInitializer() + 21 (init.h:109)
10  libwx_baseu-3.1.dylib         	0x000000010237eab6 wxEntry(int&, wchar_t**) + 454 (init.cpp:510)
11  libwx_baseu-3.1.dylib         	0x000000010237ebdc wxEntry(int&, char**) + 60 (init.cpp:519)
12  org.opencpn                   	0x00000001005bc916 main + 38 (chart1.cpp:972)
13  libdyld.dylib                 	0x00007fff6d5bded9 start + 1

```
This could very well be a bug in wxWidgets, given the stack.

nohal, 2019-03-30:
Are you able to reliably reproduce it? I can't remember ever seeing it...

keesverruijt, 2019-03-30:
No, I saw it just the one time. Let's just call it a fluke, I have since closed it 10+ times without error.

closed 2019-03-30 by keesverruijt