OpenCPN/OpenCPN #1337: wxGTK problems in route properties dialog

nohal, 2019-04-01:
- if I try to change the planned speed - double click to highlight the default speed (6) and type 5 expecting it to replace the six, instead it appends. Unable to delete with the delete key but backspace works.
- Can't select anything other than the hours in the departure time spinner. This does odd things such as displays blank after changes but is not consistent.

nohal, 2019-04-04:
`wxTimePickerCtrl` is unusable in GTK based environments on Linux (Bionic, the originally reported Mint), but works fine in eg. KDE. Seems not fixable and replacing the control with a custom textbox probably is the only chance to make it work everywhere.

closed 2019-04-22 by nohal

gromain, 2020-02-19:
The first part of this bug still exists for me in the latest git revision.

commit 2b9bb326 (2019-04-22): Use custom control for time on wxGTK

and improve planned speed control behavior.

Fixes #1337
commit 4f5593b2 (2019-04-22): Use custom control for time on wxGTK

and improve planned speed control behavior.

Fixes #1337