OpenCPN/OpenCPN #1338: Furuno Upload Route - 24 waypoints should be 30

PRBgithub, 2019-04-01:
I am new to OpenCPN - thank you for all the good work.

(1)
Furuno GPS accepts routes with max 30 waypoints in max 4 $GPRTE sentences each containing max 8 waypoints and max 80 characters long. So 30 waypoints would be 3x8+6 waypoints. Current upload uploads sentences of 6 waypoints and thus only a total of max 24 waypoints works. (multiplexer.cpp line 696 and beyond).
A warning should be issued if more than 24 (or better 30) waypoints are attempted, since nothing is uploaded at all when you try that.
(2)
Also route name - if present -  is not included in upload

See example in attached output to GPS, which leads to no route being uploaded at all.

(3) Very nice to have would be to be able to upload several selected routes, so longer routes can be split in sections while planning and uploaded as a set.

Note that Furuno GPS erases all stored waypoints and routes when uploading.

[OpenCPN log Furuno Route Upload.txt](https://github.com/OpenCPN/OpenCPN/files/3030270/OpenCPN.log.Furuno.Route.Upload.txt)

rgleason, 2021-04-14:
Furuno is long gone. Time to close this.

PRBgithub, 2021-04-15:
Hello Rick,
 
I appreciate all the work that is going into OpenCPN, and use it in combination with oeSENC charts. Thank you!!
 
Furuno still sells GPS, current models GP-33, GP-39 have same limitations.
In general upload to Furuno requires all the waypoints to be uploaded (with a unique 8 character waypoint name), followed by one or  more routes each specifying up to 30 waypoint names.
 
manual page 7-12 and 7-11 resp.:
"
Note: A route can contain max. 30 waypoints and GPRTE sentence for one route data may exceed 80 byte limitation. In this case, route data is divided into several GPRE sentences (max 4 sentences)….
"
 
Routes in Sweden (and also along the Dutch waddenkust) often contain many waypoints, because of narrow channels and dangerous rocks. Using the split route function is a work around to this issue – albeit a bit cumbersome. So if you consider this is nice-to-have only, issue can be closed. Can you do that or should I close it?
 
Best regards,
 
Philip Beekman
Floralaan West 304 
5644BP Eindhoven
 
h: +31 40 212 2792
m: +31 6 502 666 90
e: ***@***.***

rgleason, 2021-04-15:
I am redfaced!  Sorry. I was thinking of Fugawi Charts.(???)

commit 3f031871 (2021-04-17): Format Furuno route upload to allow 30 waypoints, plus name.  Closes #1338
closed 2021-04-17 by bdbcat