OpenCPN/OpenCPN #1343: RFE: Plugin loader

leamas, 2019-04-03:
Stating the obvious as an issue: Ocpn need a plugin loader which can install plugins from a central repository.

The reason is simple: The matrix of plugins x packaging schemes is not maintainable.  Universal plugin packaging can  only be done my opencpn code.

This is a huge task involving some kind of plugin metadata, distribution framework and actual code.  Filing this bug is not to say I'm about to implement it.

nohal, 2019-04-03:
Feel free to pick up https://github.com/nohal/piman_pi - I abandoned the idea as I would have to do it for pretty much all the plugins myself, which was a no go then and is a no go now (for me personally)

leamas, 2019-04-03:
> Feel free to pick up https://github.com/nohal/piman_p

Right. Will take a closer look later.

> I abandoned the idea as I would have to do it for pretty much all the plugins mysel

My gut feeling is that this is just too much work for one person. Perhaps if we could find  a way where more than one person works on it...

nohal, 2019-04-03:
> > I abandoned the idea as I would have to do it for pretty much all the plugins mysel
> 
> My gut feeling is that this is just too much work for one person. Perhaps if we could find a way where more than one person works on it...

It got **much** better nowadays with CI available for all the platforms, but still we were not even able to push all the plugin authors to do the minimal first step and start using it. Once that is in place, extending it with deployment in a special format is still quite some work, but not that insanely complex anymore....

rgleason, 2019-04-03:
Pavel,
I was not aware of this effort. Is this that big "software machine" that I thought you were building?   Kind of like Chartdownloader, but for Plugins?

What are the first steps required to start doing this?
1. CI Appveyor doing windows on all plugins
2. CI  .Travis doing linux and osx on all plugins
3. Establish a common versioning system and naming system.
4. Configure Travis and appveyor to push a tagged result to github release, so PIman can use the files.
5. Develop piman_pi to work with github  and CI.

Would it be simplest and best if all the plugins releases were in a central OpencpnPlugin github account?

How can we encourage and involve every plugin author in this effort? 

Provide a roadmap / example for them to follow which uses piman_pi.  Just one example, say IACfleet_pi?

Then I and others will help and work towards that goal.

Thanks
Rick

bdbcat, 2019-04-03:
Rick...
Not that complicated to start, especially if we focus the results on user activities.

We define an XML structure that describes the state of all plugins x platforms. State includes a current download location, wherever that might be, as well as name, version, core compatibility, etc.
Then we build a plugin with UI looking like current settings->Plugins, extended to allow one-click download and install (or update) of any available plugin, for the platform in use.

The main ongoing dev activity then becomes maintenance of the centrally located XML file.

Later, we can move to centralized "repo" of plugins, using CI, etc...  But this part is mostly for the conveniece of the devs.  User should see no difference in the UI.
Dave

leamas, 2019-04-04:
hm... perhaps a first step could be to define plugin metadata.  I have had a preliminary look at he piman_pi stuff [1].  I have also made some experiments [2]. With that +  Dave's  and Ricks's comments as a starting point, some thoughts/questions/ideas:

  - With a CI-based approach it might be better to have a central "repo" of XML files where build jobs can drop the results from start -- a central file has obvious drawbacks when it comes to plugin CI jobs.
  - Github might not be the best option for a central repo, a service like bintray.com seems more appropriate.
  - Wouldn't it be easier to handle each binary plugin as a separate entity with it's own metadata? To load all the files and show the relevant views for users shouldn't be that hard.
  - Could we simplify the piman model? In particular:
      - Drop the dependency stuff (one plugin depends on another plugin).
      -  Drop the declaration of data etc. and just install plugins using schemes like _/usr/share/opencpn/plugins/datadir_ -> _~/.local/share/opencpn/plugins/datadir_ and so on?
      - Use a simple version string instead of three different XML entities?
      - Drop the post-install command and message?
      - Drop the size element?

All in all:
  - A central repo of XML files at a suitable location like bintray.com.
  - Each XML file represents a build artifact
  - We limit data to
      -  Manifest version.
      - Name
      - Short and long description
      - Version, release  [date, git hash]
      -  API-version.
      - Platform (that is, os and os version).
      - Homepage/info URL [screenshot URL, icon url]
      - Download url.
      - Author and license [open source yes/no]

Thoughts?
      
[1] https://github.com/nohal/piman_pi/blob/f30526f7d98bb4cab3c2c757fb1eb394b64c7dd8/OpenCPNPluginManager.PluginManifest.xsd

[2] https://bintray.com/leamas/OpenCPN/plugins#files

bdbcat, 2019-04-04:
Leamas...
Nice start for the metadata definition.
My focus is more on the plugin runtime code than it is on automating the build and maintenance process.  I see one (large-ish) XML file (The Catalog) that contains clauses for every plugin that we know about, in all flavors. That XML can be hosted on github, with trusted dev write access for now.  Over time, we could create CI scripts that will automatically update the Catalog as suitably tagged builds succeed.  Maintaining the Catalog manually should be no more difficult than maintaining the opencpn.org download page links, which we already do now.  So once running, easy work.
Also, no reason to limit the individual pi xml data artificially.  It should contain the minimum required data to support the runtime download/update cycle on supported platforms, as well as anything required to support a future CI model.  But other stuff doesn't hurt.  For sure we won't get the definition right the first time, anyway...
Dave

leamas, 2019-04-04:
> My focus is more on the plugin runtime code

Right. And I can see that this becomes simpler with a single file.

> than it is on automating the build and maintenance process

My thinking around this more started with this process. And from this perspective the natural solution is to let each build produce a separate file. 

Taken together, this would mean that we keep both the single Catalog and the multi-file repo in our thinking. The Catalog is basically produced by merging the separate files in the repo. 

> Also, no reason to limit the individual pi xml data 

I'm certainly not a native English speaker. I'm just trying to find the bare minimum we can start with (and adding a manifest version so we can expand it over time).   Question is if this minimal version could work, if it's a sane way to start.

rgleason, 2019-04-04:
There are some great ideas here to solve various maintenance problems simply.
I've looked at Leamas two examples, one from Pavel (looks pretty complete) and leamas bintray plugin manager.
I like the inclusion of the API version required!
I also have some questions about how I would make these xml files if I were to be focused on a number of Windows plugins, which has been my focus, particularly if I am forking other repositories (generally trying to follow the parent repository, but not always) and uploading compiled windows plugins to my github Release directory.  Will the field defined allow me to do this with clarity? I don't think so.
I think there should be a primary github account with author and a secondary github account with OS maintainer or whatever.  Maybe it should be a secondary github Release directory.  I leave it to you guys to figure out what is really needed, but there are times when there is an OS plugin, maintained by someone other than the primary github account.  (I generally try to keep  the Windows version of Sean's plugins up to date).

Now with CI that situation may not need to exist... provided the primary account has appveyor.yml all set up to complete properly.  The key is to get it setup so it is seamless.

The other alternative is to get both appveyor.yml and .travis.yml working so that Windows, Linux and MacOS versions are all produced and used... I would be willing to work on that too.

leamas, 2019-04-04:
> PS: Now I see that Leamas has made separate xml for each of Mac, Linux and Windows of bdbcat's oesenc_pi. So I guess my concerns are not relevant.

They are actually produced from the same template, but yes, they are separate. The mechanics behind  is available at https://github.com/leamas/oesenc_pi

rgleason, 2019-04-04:
Leamas, I am not following the mechanics here, but I have not looked at it long enough. It seems like you are making a complete plugin machine here though.
I don't quite understand where cmake makes a new .travis file?  How does that work?

Which leads me to comment that all plugins will have to be updated to exact and current cmake versions, appveyor, and travis.

I see you moved opencpn.lib out of the windows build and named it explicitly with the API number, bravo. That should be very helpful, but I suppose cmake strips the api while running.

This is a little complex for me right now, but maybe I could walk through changing a plugin accordingly if you were to help me.  Does this all work now?   How does the bintray get created?

It seems like there are two goals here:
1. Easing User plugin selection and installation  (bdbcat suggestion)
2. Unifying and streamlining plugin compiling using CI and scripts.

How far have you gotten with this?

leamas, 2019-04-04:
> How far have you gotten with this?

Basically nowhere...

I'd suggest that you set up appveyor CI scripts in your windows plugins, using the Appveyor documentation -- it's rather straight-forward. At a later point you should be able to create some XML metadata in that job,  but  I don't think we are there yet.

leamas, 2019-04-04:
Perhaps we need some new bugs to sort this out. But besides the metadata we also need to define where plugins are installed since it need to in areas users have access to.  A table with missing entries:

               win32                         linux                           OSX
     data      $HOME/?                       ~/.local/share/opencpn/plugins  ?
     dll/so    $HOME/?                       ~/.local/lib/opencpn/           ?
     binaries  $HOME/?                       ~/.local/bin/                   ?

Also, do any plugins  define anything else than the dll/so-file, the plugin data and/or helper binaries?

taruti, 2019-04-07:
Is there a third party library that could do this? Seems like something countless other projects have encountered. An ad hoc solution could grow more and more complex (with e.g. signatures/checksums etc).

nohal, 2019-04-07:
https://sparkle-project.org / https://winsparkle.org but they are not exactly what we are looking for.

leamas, 2019-04-07:
First: it's a good question, for sure.

The problem is that it's not really a plugin manager we need. Plugin managers handles things like defining plugins, loading, unloading etc -- but we already have those parts. And here are so many wheels running anyway so we shouldn't add another one by messing with that code.

What we really need is the missing parts: Metadata definitions, a wxWidgets UI and download/install the plugin. I doubt we can use any existing framework for this,  but I'd love to be proven wrong.

rgleason, 2019-04-09:
I will try to use Pavel's appveyor and travis files as an example, but what about  opencpn's appeveyor and travis, but I am unsure which plugins\opencpn  versions are the most applicable and complete. 

- appveyor (for windows)
- .travis  (Osx & Linux?)
- ? (Android?)

Cmake files and subfiles are also hugely important, and I don't know which ones are the best examples to follow yet.  I notice that there is a versioning system for cmake.  I don't know when that started or this is a way for me to be using the most applicable or current version.

Some brief directions and instruction would assist and prevent needless spinning of wheels doing something improperly or wrong.

I would probably start with climatology_pi which is now in pretty good shape after Pavel's and Sean's work.  Is that a good example at this point for CMAKE, appveyor and .travis?  ...Maybe I could follow that?

leamas, 2019-04-09:
For now, we are more or less  stuck on my question in  my question "why"  in https://github.com/OpenCPN/OpenCPN/issues/1347#issuecomment-480578307. This is a core issue when defining how to handle plugins.

@bdbcat : Thoughts?

leamas, 2019-04-09:
As for appveyor/travis I presume you have registered yourself with them; that's the first part. 

@rgleason : For reference, there is also https://github.com/leamas/oesenc_pi. It also has some very experimental appveyor/travis setup. Much of it is related to metadata generation, and that one will never fly as it is. But, perhaps you could find useful bits there when it comes to just building the plugins, dunno.

rgleason, 2019-04-09:
leamas, 
I have appveyor and travis accounts. 
Is the oesenc_pi in your repository operational in that properly implemented, it will produce usable plugins?
Is the cmake and subdirectories to a current standard?  IE can I use it as an example?

I expect that some of this will be experimentation, / "learning"

Thanks.
Rick

leamas, 2019-04-09:
> it will produce usable plugins?

No, it produces tarballs... Eventually, I will have a look into this part.

leamas, 2019-04-09:
> General comment: On Windows, it is considered somewhat "anti-social" to store app-specific data or binaries in predefined places on the file system. 

I  see the need to be flexible as to where plugins are installed. What I  don't understand is the need to make this to a decision for each and every plugin. Could in be enough that user selects  a "plugin directory" once and for all? And that the plugin manager then uses this location?

The underlying code actually supports multiple directories, so user could select multiple paths, one of them (user-writable) could be used by the plugin manager.

>  I have a full plate right now....

I bet ;) It was just a heads up in case you missed my question -- after all, missing something in these two bugs could easily happen

leamas, 2019-04-10:
@bdbcat  wrote: 
> Uninstalling OCPN, and all its parts, will require thought.

Indeed, and lot's of it. As for a plugin manager, what could be done is to keep track and uninstall what it has installed.  This is basically the plugin dlll, possible helper binaries and plugin data.  Anything outside this scope is by definition outside this effort, which IMHO is big enough anyway.

leamas, 2019-04-13:
My current thinking, some bullets:

   - Plugins are either managed or not. Unmanaged are those installed by current means (NSIS packages etc). They are basically read-only objects for the plugin manager.
   - Managed plugins are installed and uninstalled by the plugin manager. 
   - There is one base directory for plugin dll/so-file, one for binary helpers and one for plugin data. These locations are platform dependent and can be changed by users. All managed plugins uses the same set of locations.
   - The details how users select installation directories are platform-dependent.
   - Plugin packages consists of a simple tarball and a metadata file created by cmake. 
   - The tarballs has three base directories for dll, binaries and data. The manager relocates these three directories to the actual setup when installing. There is really no metadata involved in this.
   - Each platform build is a separate package -- both tarball and metadata file.

@bdbcat: @nohal: I really hate to take your time. However, I seem to have some cycles available upcoming week for this, and would appreciate any feedback before jumping into the water...

nohal, 2019-04-15:
Something like this, with Linux being many platforms, at least linux-amd64, linux-x86, linux-armhf and linux-arm64, but in reality unfortunately more.

I would probably limit the options for the user to select anything, if possible completely (Portable mode = everything in `<binary location>/plugins/[...]`, normal installation `<user_data_location_on the platform>/plugins/[...]`) as we know that while it will benefit few, many will break their systems.

bdbcat, 2019-04-15:
I think this will work, just mapping oesenc_pi into this idea in my head.  oesenc_pi is about as complex as it gets, with helpers, dlls, data, etc.
Essentially shifts the platform dependencies back to the plugin.  For example, the Windows conditional code will need to detect at runtime the correct base path for the various parts, based upon the currently running core.  And may need permissions elevation.  
Pavel:  Will it work to edit the currently running app bundle on Mac?  Never tried this...
Dave

nohal, 2019-04-15:
> Will it work to edit the currently running app bundle on Mac? Never tried this...

Doing anything with the bundle is problematic once it is signed. But not for the portable mode, which a) is probably used very rarely on macOS and b) has to be invoked explicitly by the user so the bundle does not need to be signed at all.
If by editing we mean adding files, that seems to work fine, for now...

leamas, 2019-04-15:
> Doing anything with the bundle is problematic once it is signed.

My idea is to patch the plugins script so they  can create a tarball (i. e., the TGZ alternative) as an option and use this....

leamas, 2019-04-28:
I have found some cycles for this, some things become more clear,  and there are new questions. Some bullets, primary for my own memory:

  - Naming: We already have a pluginmanager wich discovers and loads installed plugins. The 
    new thing installs and uninstalls plugins; let's call it PluginHandler.
  -  Listing installed plugins is trivial, the info is available in the pluginmanager [code available]
  - Listing available plugins from an XML file is also trivial [code available].
  - Installing on Linux is in many ways the most demanding part. The native packaging (deb, rpm,
    flatpak...) cannot be used. A simple tarball can be installed in user directories [code available],
    and works on distros.
  - Let's focus on Linux for now, hard part first.
  - Plugins needs patching to locate data, binaries and libs which are installed in other locations
    than _make install_ once created. Let's try oesenc_pi, a non-trivial example.

leamas, 2019-05-08:
For now, a small step for man: Builds for all platforms from this branch  available at https://sourceforge.net/projects/opencpn-devel/files/5.0.0-post/

rgleason, 2019-05-08:
This is cool!  5 different os builds,  ...opencpn.pdb is 35 mb?

leamas, 2019-05-25:
This discussion is in too many places, my bad. Closing this issue, referring to the PR #1390

closed 2019-05-25 by leamas