OpenCPN/OpenCPN #1344: Render issue on Linux with HiDPI screen

gromain, 2019-04-03:
Hello,

I've got a huge rendering issue in latest v5.0.0.
See the attached screenshot for a visual of the problem, but basically the rendering window is locked to the bottom left of the screen.
I'm running Opencpn in Manjaro Linux (Arch Linux), compiled from sources.

[image]

Also attached is the log at the bottom of this issue.

If you need more information, I'm available for more debugging.

```
________
21:27:11 CEST: 2019-04-03
21:27:11 CEST:  ------- Starting OpenCPN -------
21:27:11 CEST: 5.0.0
21:27:11 CEST: wxWidgets version: wxWidgets 3.0.4 Linux 64 bit wxGTK
21:27:11 CEST: OS Version reports as:  5.0
21:27:11 CEST: MemoryStatus:  mem_total: 31808 mb,  mem_initial: 3 mb
21:27:11 CEST: SData_Locn is /usr/share/opencpn/
21:27:11 CEST: PrivateDataDir is /home/rbazile/.opencpn
21:27:11 CEST: Using existing Config_File: /home/rbazile/.opencpn/opencpn.conf
21:27:11 CEST: Setting Viewpoint Lat/Lon 41.1323, -36.0232
21:27:11 CEST: Setting Ownship Lat/Lon 32.5901, -52.4825
21:27:11 CEST: Switch InlandEcdis mode Off
21:27:11 CEST: Styles loading from /usr/share/opencpn/uidata/styles.xml
21:27:11 CEST: No styles found at: /home/rbazile/
21:27:11 CEST: No styles found at: /home/rbazile/.opencpn/
21:27:11 CEST: Using SVG Icons
21:27:11 CEST: Detected display size (horizontal): 340 mm
21:27:11 CEST: Getting DefaultSystemLocale...
21:27:11 CEST: System default Language:  fr_FR
21:27:11 CEST: Config file language:  en_US
21:27:11 CEST: Adjusted App language:  en_US
21:27:11 CEST: Opencpn language load for:  en_US
21:27:11 CEST: Loading catalog for opencpn core.
21:27:11 CEST: Getting DefaultSystemLocale...
21:27:11 CEST: Opencpn language set to:  en_US
21:27:11 CEST: Loading Configs catalog: /home/rbazile/.opencpn/Configs/configs.xml
21:27:11 CEST: Creating MyFrame...size(1920, 1020)  position(-10, -1)
21:27:12 CEST: Creating glChartCanvas
21:27:12 CEST: Metrics:  m_display_size_mm: 340     wxDisplaySize:  1920:1080   
21:27:12 CEST: Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONIC.IDX
21:27:12 CEST: OpenGL-> Renderer String: Mesa DRI Intel(R) UHD Graphics 630 (Coffeelake 3x8 GT2) 
21:27:12 CEST: OpenGL-> Version reported:  3.0 Mesa 18.3.5
21:27:12 CEST: OpenGL-> Detected Intel renderer, disabling stencil buffer
21:27:12 CEST: OpenGL-> Texture rectangle format: de1
21:27:12 CEST: OpenGL-> Using Vertexbuffer Objects
21:27:12 CEST: OpenGL-> Using Framebuffer Objects
21:27:12 CEST: OpenGL-> Using FBO Stencil buffer
21:27:12 CEST: OpenGL-> Using Depth buffer clipping
21:27:12 CEST: OpenGL-> Using s3tc dxt1 compression
21:27:12 CEST: OpenGL-> Compressed tile size: 128kb (8:1)
21:27:12 CEST: OpenGL-> Minimum cartographic line width:  1.0
21:27:12 CEST: OpenGL-> Minimum symbol line width:  1.1
21:27:12 CEST: ChartDB Cache policy:  Application target is 1024 MBytes
21:27:12 CEST: Loading chart db version: V018
21:27:12 CEST: Chartdb: Chart directory list follows
21:27:12 CEST:   Chart directory #0: /home/rbazile/Charts/GSHHG
21:27:12 CEST:   Chart directory #1: /home/rbazile/Charts/ENC/FR_INLAND
21:27:12 CEST: GPS Watchdog Timeout is: 6 sec.
21:27:12 CEST: OpenCPN Initialized in 621 ms.
21:27:12 CEST: DPMM: 5.64706   ScaleFactorExp: 1
21:27:12 CEST: Looking for UserIcons at /home/rbazile/.opencpn/UserIcons
21:27:12 CEST: Loading navobjects from navobj.xml
21:27:12 CEST: Done loading navobjects, 0 duplicate waypoints ignored
21:27:12 CEST: Loading World Chart Q=0 in 0 ms.
21:27:12 CEST: GSHHS World chart files have wrong version. Found -1, expected 210-220.
21:27:12 CEST: Loading World Chart Q=4 in 0 ms.
21:27:13 CEST: PlugInManager: plugins loading from ~/.local/lib/opencpn:/usr/local/lib/opencpn:/usr/lib/opencpn
21:27:13 CEST: PlugInManager searching for PlugIns in location /home/rbazile/.local/lib/opencpn
21:27:13 CEST:    Directory /home/rbazile/.local/lib/opencpn does not exist.
21:27:13 CEST: PlugInManager searching for PlugIns in location /usr/local/lib/opencpn
21:27:13 CEST:    Directory /usr/local/lib/opencpn does not exist.
21:27:13 CEST: PlugInManager searching for PlugIns in location /usr/lib/opencpn
21:27:13 CEST: Error: Could not open file "opencpn" for reading with errno = 2.
21:27:13 CEST: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libgrib_pi.so
21:27:13 CEST: PlugInManager:  /usr/lib/opencpn/libgrib_pi.so
              API Version detected: 116
              PlugIn Version detected: 401
21:27:13 CEST: 
21:27:13 CEST: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libchartdldr_pi.so
21:27:13 CEST: PlugInManager:  /usr/lib/opencpn/libchartdldr_pi.so
              API Version detected: 113
              PlugIn Version detected: 103
21:27:13 CEST: chartdldr_pi: /home/rbazile/Charts
21:27:13 CEST: chartdldr_pi: /home/rbazile/Charts
21:27:13 CEST: PlugInManager: Loading PlugIn: /usr/lib/opencpn/libwmm_pi.so
21:27:13 CEST: PlugInManager:  /usr/lib/opencpn/libwmm_pi.so
              API Version detected: 108
              PlugIn Version detected: 101
21:27:13 CEST: WMM: WMM model data loaded from file /usr/share/opencpn/plugins/wmm_pi/data/WMM.COF.
21:27:13 CEST: Opencpn language load for:  en_US
21:27:13 CEST: Loading catalog for:  opencpn-grib_pi
21:27:13 CEST: Loading catalog for:  opencpn-chartdldr_pi
21:27:13 CEST: Loading catalog for:  opencpn-wmm_pi
21:27:13 CEST: Loading catalog for opencpn core.
21:27:13 CEST: Getting DefaultSystemLocale...
21:27:13 CEST: Device: 0: HDA Intel PCH: ALC3266 Analog (hw:0,0)
21:27:13 CEST: Device: 1: HDA Intel PCH: HDMI 0 (hw:0,3)
21:27:13 CEST: Device: 2: HDA Intel PCH: HDMI 1 (hw:0,7)
21:27:13 CEST: Device: 3: HDA Intel PCH: HDMI 2 (hw:0,8)
21:27:13 CEST: Device: 4: HDA Intel PCH: HDMI 3 (hw:0,9)
21:27:13 CEST: Device: 5: HDA Intel PCH: HDMI 4 (hw:0,10)
21:27:13 CEST: Device: 6: sysdefault
21:27:13 CEST: Device: 7: front
21:27:13 CEST: Device: 8: surround40
21:27:13 CEST: Device: 9: surround51
21:27:13 CEST: Device: 10: surround71
21:27:13 CEST: Device: 11: hdmi
21:27:13 CEST: Device: 12: samplerate
21:27:13 CEST: Device: 13: speexrate
21:27:13 CEST: Device: 14: pulse
21:27:13 CEST: Device: 15: speex
21:27:13 CEST: Device: 16: upmix
21:27:13 CEST: Device: 17: vdownmix
21:27:13 CEST: Device: 18: dmix
21:27:13 CEST: Device: 19: default
21:27:13 CEST: Device: 0: HDA Intel PCH: ALC3266 Analog (hw:0,0)
21:27:13 CEST: Device: 1: HDA Intel PCH: HDMI 0 (hw:0,3)
21:27:13 CEST: Device: 2: HDA Intel PCH: HDMI 1 (hw:0,7)
21:27:13 CEST: Device: 3: HDA Intel PCH: HDMI 2 (hw:0,8)
21:27:13 CEST: Device: 4: HDA Intel PCH: HDMI 3 (hw:0,9)
21:27:13 CEST: Device: 5: HDA Intel PCH: HDMI 4 (hw:0,10)
21:27:13 CEST: Device: 6: sysdefault
21:27:13 CEST: Device: 7: front
21:27:13 CEST: Device: 8: surround40
21:27:13 CEST: Device: 9: surround51
21:27:13 CEST: Device: 10: surround71
21:27:13 CEST: Device: 11: hdmi
21:27:13 CEST: Device: 12: samplerate
21:27:13 CEST: Device: 13: speexrate
21:27:13 CEST: Device: 14: pulse
21:27:13 CEST: Device: 15: speex
21:27:13 CEST: Device: 16: upmix
21:27:13 CEST: Device: 17: vdownmix
21:27:13 CEST: Device: 18: dmix
21:27:13 CEST: Device: 19: default
21:27:13 CEST: options: got device count: 20
21:27:14 CEST:    ***GPS Watchdog timeout at Lat:32.5901   Lon: -52.4825
21:28:12 CEST: LOGBOOK:  2019-04-03 19:28:12 UTC  DR Lat   32.59010 Lon  -52.48250
21:29:24 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libchartdldr_pi.so
21:29:24 CEST: chartdldr_pi: DeInit
21:29:24 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libwmm_pi.so
21:29:24 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libgrib_pi.so
21:29:24 CEST: opencpn::MyFrame exiting cleanly.
21:29:24 CEST: Chart cache purge
21:29:24 CEST: Chart cache PlugIn purge
21:29:24 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libchartdldr_pi.so
21:29:24 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libwmm_pi.so
21:29:24 CEST: PlugInManager: Deactivating PlugIn: /usr/lib/opencpn/libgrib_pi.so
21:29:24 CEST: Chart cache purge
21:29:24 CEST: opencpn::MyApp starting exit.
21:29:24 CEST: LOGBOOK:  2019-04-03 19:29:24 UTC OFF: Lat   32.59010 Lon  -52.48250
21:29:24 CEST: UnLoading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONIC.IDX
21:29:24 CEST: opencpn::MyApp exiting cleanly...
```

nohal, 2019-04-03:
X/Wayland? The actual resolution of the screen?

gromain, 2019-04-03:
I'm on Xorg with a resolution of 3840x2160.

nohal, 2019-04-03:
The problem obviously is somewhere in the wxWidgets/GTK/X pipeline bellow OpenCPN which reports the resolution as FHD, but does not do the scaling as the OSes normally do. Do you use any UI scaling? If not, what happens if you try to?
I am still reluctant to suggest trying to build wxWidgets 3.1 from source and build against that or brutalities like
```diff
diff --git a/src/chcanv.cpp b/src/chcanv.cpp
index 2537c7459..344b35927 100644
--- a/src/chcanv.cpp
+++ b/src/chcanv.cpp
@@ -6418,7 +6418,8 @@ void ChartCanvas::OnSize( wxSizeEvent& event )
 {

     GetClientSize( &m_canvas_width, &m_canvas_height );
-
+    m_canvas_width *= 2;
+    m_canvas_height *= 2;
 //    Get some canvas metrics

 //          Rescale to current value, in order to rebuild VPoint data structures
```
although they would probably have some effect on your particular system in it's current state.

gromain, 2019-04-03:
Yes, I see what you mean.
There must be scaling somewhere (I think I set up my system with a x2 scale, I'm using a KDE Plasma desktop).
I'll check that patch tomorrow and report back.

nohal, 2019-04-03:
FWIW macs with retina displays behave exactly like this, reporting only the lower logical resolution, but they do the other half of the work as well and scale the coordinates back to the display correctly...

gromain, 2019-07-04:
Hey there! Just a quick word to say I found a workaround. The problem seems to come from gtk scaling. I had to add 'export GDK_SCALE=2' before the command when I launch opencpn. This solves the problem for now, until there is a proper solution in place for hidpi screens.

reneherrero, 2019-09-29:
I had the same issue:
1. looked very carefully at cmake output
2. added all the missing libraries
3. watch problem disappear

see #1451

gromain, 2019-11-25:
@reneherrero do you remember the missing libraries? I don't remember having warnings about missing libs. I'll try to recompile to see if there is something like this and update here.

reneherrero, 2019-11-25:
This worked for me, but you will need to rebuild
'
apt-get install -y --no-install-recommends \
    ca-certificates \
    build-essential \
    cmake \
    gettext \
    git-core \
    gpsd \
    gpsd-clients \
    libgps-dev \
    wx-common \
    libwxgtk3.0-dev \
    libglu1-mesa-dev \
    libgtk2.0-dev \
    wx3.0-headers \
    libbz2-dev \
    libtinyxml-dev \
    libportaudio2 \
    portaudio19-dev \
    libcurl4-openssl-dev \
    libexpat1-dev \
    libcairo2-dev \
    libarchive-dev \
    liblzma-dev \
    libexif-dev \
    libelf-dev \
    libsqlite3-dev \
    libsndfile1-dev \
    libjpeg-dev 
`

gromain, 2020-02-18:
Just to let everyone knows, the problem still exists as of latest commit #037c31a.
I've been digging a bit more in the code, but still no luck. Is there any documentation as how the code is structured? I'm trying to wrap my head around the canvas and viewport and such but I'm still not sure what the display element for the chart is.

Also, to help finding the origin, even if the chart is not shown everywhere, it's still treated as if it's displayed on the full screen (meaning if I create a waypoint by clicking on the top right of my screen, the waypoint is created on the top right of the displayed map).

Also, my cmake output don't show any missing lib, but two things are odd:
- Says it can't find wxWidget yet finds it after
- Gives no value for GL Lib yet it found it

```
~/.../opencpn/build >>> cmake ../                                                                                                                           ±[●][master]
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- cmake version: 3.16.4
-- Found PkgConfig: /bin/pkg-config (found version "1.6.3") 
-- *** Build Architecture is i386
-- Setting C++11 standard via cmake standard mechanism
-- Default compiler options:
-- CMAKE_CXX_FLAGS: 
-- CMAKE_CXX_FLAGS_DEBUG: -g
-- CMAKE_CXX_FLAGS_MINSIZEREL: -Os -DNDEBUG
-- CMAKE_CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_CXX_FLAGS_RELWITHDEBINFO: -O2 -g -DNDEBUG
-- *** Staging to build opencpn 5.0.522 ***
-- *** Package will NOT include documentation ***
-- Found Python: /bin/python3.8 (found version "3.8.1") found components: Interpreter 
-- Using python executable: /bin/python3.8
-- Found OpenGL: /usr/lib/libOpenGL.so   
-- Found OpenGL....
--     GL Lib: /usr/lib/libGL.so/usr/lib/libGLU.so
--     GL Include: /usr/include
--     Revised GL Lib: /usr/lib/libGL.so
-- Main: gl;net;xml;html;adv;aui;core;base;webview
-- Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES) 
-- Could not find wxWidgets webview add-on
-- Found wxWidgets: -pthread;;;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0 (found version "3.0.4") 
-- Found wxWidgets...
--  wxWidgets Include: /usr/lib/wx/include/gtk2-unicode-3.0;/usr/include/wx-3.0
--  wxWidgets Libraries: -pthread;;;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0
--  Revised wxWidgets Libraries: -pthread;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0
-- 
-- *** Staging to build opencpn  ***
-- *** Build type: RelWithDebInfo
-- *** Will install to /usr/local  ***
-- GarminHost is enabled.
-- Found Gettext: /bin/msgmerge (found version "0.20.1") 
-- selected config gtk2-unicode-3.0

-- Found OpenGL....
--     GL Lib: 
--     GL Include: /usr/include
-- Crash reporting enabled
-- Using Linux crash reporting
-- Checking for module 'sqlite3'
--   Found sqlite3, version 3.31.1
-- Found Sqlite: /usr/include  
-- Using systrem SQLite library version 3031001.
-- Found BZip2: /usr/lib/libbz2.so (found version "1.0.8") 
[... 79 lines cut ...]
-- *** Added plugin: /home/rbazile/Documents/codes/opencpn/opencpn/plugins/dashboard_pi
-- *** Staging to build grib_pi ***
-- Found JPEG: /usr/lib/libjpeg.so (found version "80") 
-- Found Jasper: /usr/lib/libjasper.so (found version "2.0.16") 
-- Using system jasper library...
-- Found GTK2_GTK: /usr/lib/libgtk-x11-2.0.so  
-- Building against GTK2...
-- GTK2 headers: /usr/include/freetype2;/usr/include;/usr/include/glib-2.0;/usr/lib64/glib-2.0/include;/usr/include/atk-1.0;/usr/include/gdk-pixbuf-2.0;/usr/include/cairo;/usr/include/pango-1.0;/usr/include/harfbuzz;/usr/include/gtk-2.0;/usr/lib64/gtk-2.0/include
-- GTK2 libs: /usr/lib/libglib-2.0.so;/usr/lib/libgobject-2.0.so;/usr/lib/libatk-1.0.so;/usr/lib/libgio-2.0.so;/usr/lib/libgthread-2.0.so;/usr/lib/libgmodule-2.0.so;/usr/lib/libgdk_pixbuf-2.0.so;/usr/lib/libcairo.so;/usr/lib/libpango-1.0.so;/usr/lib/libpangocairo-1.0.so;/usr/lib/libpangoft2-1.0.so;/usr/lib/libpangoxft-1.0.so;/usr/lib/libgdk-x11-2.0.so;/usr/lib/libgtk-x11-2.0.so
-- *** Added plugin: /home/rbazile/Documents/codes/opencpn/opencpn/plugins/grib_pi
-- Build type: RelWithDebInfo
-- *** Staging to build wmm_pi ***
-- Found OpenGL...
--     Lib: /usr/lib/libGL.so/usr/lib/libGLU.so
--     Include: /usr/include
-- Found OpenGL...
--     Lib: /usr/lib/libGL.so/usr/lib/libGLU.so
--     Include: /usr/include
-- *** Added plugin: /home/rbazile/Documents/codes/opencpn/opencpn/plugins/wmm_pi
-- *** Package will include GSHHS basechart level: NONE ***
-- *** Package will NOT include tide and current data ***
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rbazile/Documents/codes/opencpn/opencpn/build
```

gromain, 2020-02-18:
Here is a screenshot to see what I mean for the chart/mouse location discrepancy.
[image]

gromain, 2020-06-28:
Just an update, this problem still exists in the latest version when not running with "GDK_SCALE=1".

calas, 2021-04-23:
The issue is still present in version 5.2.4 without the GDK_SCALE=1 workaround.

nohal, 2021-04-23:
With the future wxWidgets (3.1.5 and newer) versions the GL canvas behavior for HiDPI screens chages again and similar/same issue is present also on macOS (#2131), without any workaround available.

frfa, 2022-02-08:
Just wanted to note that this problem persists with wxWidgets 3.1.5 on Ubuntu 21.10 "Impish" with OpenCPN 5.6.1 compiled against GTK3, and the GDK_SCALE=1 workaround described above does not seem to work.
Also positioning of the toolbars seems to be somehow arbitrary, and those cannot be moved.

However, touchscreen right-click emulation by long-press on e.g. a waypoint *does* open the context menu, so at least touchscreens seem to be better supported with wxWidgets 3.1.5.

rgleason, 2022-02-08:
This issue is likely more important now.

frfa, 2022-02-09:
Thanks for reviving this issue.

FWIW, disabling OpenGL in the settings restores the full scale chart, but the odd placement of the toolbars remains; it seems that wxWidgets 3.1.5 has problems determining the correct window geometry. Maybe we need to wait for wxWidgets 3.2 anyway.

I can offer help testing commits re. touchscreen functionality on my Ubuntu 21.10 tablet, if desired.

rgleason, 2023-04-28:
@frfa  Can you build the current master and test please. If it is resolved please close this issue, otherwise please provide more current details.

frfa, 2023-04-30:
Issue has been resolved in current master, thanks.

closed 2023-04-30 by nohal