OpenCPN/OpenCPN #1345: x86_64 OpenCPN cant find plugins in /usr/lib64/opencpn?

mdinslage, 2019-04-04:
Hello,

When compiling OpenCPN 4.8.8 on my Slackware machine using the slackbuild I maintain over at www.slackbuilds.org everything builds fine.  However Slackware has both a 32bit and a 64bit edition.  The 32bit installs libraries into /usr/lib while the 64bit version installs to /usr/lib64.  however OpenCPN only sees plugins if they are install into /usr/lib/opencpn.  Is there a cmake variable I can pass or can I patch a file to make opencpn accept plugins installed into /usr/lib64?

Thanks.

mdinslage, 2019-04-04:
I updated my slackbuild to 5.0.0, and after viewing this merge https://github.com/OpenCPN/OpenCPN/pull/1114 I edited pluginmanager.cpp and changed the plugins dir from lib to lib64 and plugins are now working.  Is this an acceptable way to make this work in your guys opinion

leamas, 2019-04-04:
No, it's not. However, the plugin load system supports relocating the plugins using environment  variables -- see the new opencpn.1 manpage

mdinslage, 2019-04-04:
Thank you for the response.

closed 2019-04-04 by mdinslage