OpenCPN/OpenCPN #1347: RFE: Support alternative plugin locations on Windows and OSX

leamas, 2019-04-04:
This is basically some underpinning to #1343.

We need to be able to install and use plugins to/from user-writable locations. This is already supported for Linux, but is required also for other platforms.

First step would be to define suitable locations in table below adding entries, columns and/or rows as required:

               win32                         linux                           OSX
     data      $HOME/?                       ~/.local/share/opencpn/plugins  ?
     dll/so    $HOME/?                       ~/.local/lib/opencpn/           ?
     binaries  $HOME/?                       ~/.local/bin/                   ?

Second part would be to patch pluginmanager.cpp, basically  the functions  PlugInManager::LoadAllPlugIns() and  GetPluginDataDir().

rgleason, 2019-04-04:
win32

read only
data   C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins\iacfleet_pi [plugin] \data
dll   C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins
binaries   C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins\weatherfax_pi (where PVW32con.exe exists for example)

readwrite files and User directory
C:\ProgramData\opencpn\plugins\[plugin-name]
C:\ProgramData\opencpn\plugins\climatology_pi\data
sometimes there are additional folders for specific plugins.

leamas, 2019-04-04:
The ro-locations cannot be used.

Is this to say the the typical climatology_pi plugin could (should?) be installed as

    C:\ProgramData\opencpn\plugins\climatology_pi.dll
    C:\ProgramData\opencpn\plugins\climatology_pi\

where the plugin data is under the last path? 

Does indeed a standard windows installation allow regular users to write in C:\ProgramData\opencpn? 

Some plugins contains helper binaries (e. g., oesenc_pi). Where should they go?

Hakansv, 2019-04-04:
[image]

leamas, 2019-04-04:
Thanks, explains a lot. Am I right supposing the main oesenc_pi.dll lives in  c:\ProgramFiles(x86)\OpenCPN\plugins in thsi case?

Hakansv, 2019-04-04:
Yes as every plugin, see shot. But user data for each plugin, if any, resides in like this: C:\ProgramData\opencpn\oesenc_pi

[image]

leamas, 2019-04-04:
OK, thanks 

Last question: Does indeed a standard windows installation allow regular users to write in C:\ProgramData\opencpn? (sorry, but for a Linux guy this just looks weird)

nohal, 2019-04-04:
If you have the required permissions than yes, if you don't and have permissions to elevate them, which you normally do have on a non-corporate machine, you simply confirm that you do want the installer to write there. That is complicated to achieve from a running application though (= not usable for the plugin manager)

leamas, 2019-04-04:
...which leaves us with $HOME\opencpn\plugins or so?

nohal, 2019-04-04:
Basically, yes. Or any other place where we make sure regular user will have write access during the installation with elevated privileges

Hakansv, 2019-04-04:
Normally (See Pavles notes) yes, but to find them is not straight forward for all average users. You've to check an option in the file explorer to view also "hidden" files. Unchecked by default. But since the log an config (ini) files are there as well many O users learn to look them up.

leamas, 2019-04-04:
Right. And as long as everything works the new plugin loader should handle this transparently without uer intervention. Which basicaly boils down to table, iteration2:

               win32                               linux                                    OSX
     data      $HOME/opencpn/plugins/<plugin>      ~/.local/share/opencpn/plugins/<plugin>  ?
     binaries  $HOME/opencpn/plugins/<plugin>      ~/.local/bin/                            ?
     dll/so    $HOME/opencpn/plugins/<plugin>.dll  ~/.local/lib/opencpn/                    ?

Or?

nohal, 2019-04-04:
On macOS the libraries belong somewhere under `~/Library/Application Support/OpenCPN`, the data may simply stay in `~/Library/Preferences/opencpn` as they are now (To make it even more fun Apple filesystems are by default case-insensitive, but people who switch them to case-sensitive do exist and some of them do use OpenCPN). The less changes to the current state, the better support-wise.

leamas, 2019-04-04:
> On macOS the libraries belong somewhere under` ~/Library/Application Support/OpenCPN`

Does this include the helper binaries?

 > the data may simply stay in `~/Library/Preferences/opencpn`

This is  `~/Library/Preferences/opencpn/<plugin>, right?`

> The less changes to the current state, the better 

Indeed, no reason to change since user already has write permissions.

nohal, 2019-04-04:
> > On macOS the libraries belong somewhere under` ~/Library/Application Support/OpenCPN`
> 
> Does this include the helper binaries?

Yes, it is pretty common to find whole applications there

> 
> > the data may simply stay in `~/Library/Preferences/opencpn`
> 
> This is `~/Library/Preferences/opencpn/<plugin>, right?`

Usually, yes. Should be always.

> 
> > The less changes to the current state, the better
> 
> Indeed, no reason to change since user already has write permissions.

leamas, 2019-04-04:
So, we have table iteration 3:

               win32                               linux                                    OSX
     data      $HOME/opencpn/plugins/<plugin>/     ~/.local/share/opencpn/plugins/<plugin>/ ~/Library/Preferences/opencpn/<plugin>
     binaries  $HOME/opencpn/plugins/<plugin>/     ~/.local/bin/                            ~/Library/Application Support/OpenCPN
     dll/so    $HOME/opencpn/plugins/<plugin>.dll  ~/.local/lib/opencpn/                    ~/Library/Application Support/OpenCPN

This also means that we have usable, user-writable locations to install plugins on both OSX and linux.

However, there is a need to patch pluginmanager.cpp to support the win32 locations on windows as described above https://github.com/OpenCPN/OpenCPN/issues/1347#issue-429214598. Do I dare to say that patches  are welcome?

I might lso make sense to patch the same code to accept environment varables to relocate the plugins also on OSX -- perhaps it could be useful while testing.

rgleason, 2019-04-05:
Leamas, in Windows I can also reach the read write User directory in this manner:

C:\Users\All Users\opencpn\plugins\climatology_pi
C:\Users\All Users\opencpn\plugins\climatology_pi\data
C:\Users\All Users\opencpn\plugins\weatherfax

[image]
[image]
[image]

leamas, 2019-04-05:
Well, IMHO there is nothing here which can be viewed as "correct". Question is if it might work, and everybody seems to agree on that.

While we are on it:  could you make a PR with the required changes to pluginmanager.cpp? (if not, I could make it, but in that case I'd need some help w testing).

Also: one thing to do in preparation for this work is to make plugins use GetPluginDataDir() instead of GetpSharedDataLocation(). One example in https://github.com/leamas/oesenc_pi/commit/b6654d5137b907. (Obviously, it needs to be conditionalized on something else than` #ifdef FLATPAK`.) There is some more under _Packaging Plugins_ in https://github.com/leamas/OpenCPN/blob/flatpak/flatpak/README.md

Any plugin having data needs this fix in order to be handled by a plugin loader.

rgleason, 2019-04-05:
Leamas, are the Windows plugins going to be packaged using NSIS as *.exe?  or willl the dll be used?
The exe normally installs reliably without path issues.

leamas, 2019-04-05:
My idea  is to produce tarballs (zipballs) with installation paths same as today.

The plugin loader then relocates e. g.  C:\program files (x86)\opencpn\plugins\oesenc_pi.dll to $HOME\opencpn\plugins\oesenc_pi.dll and so on. That is,  we keep the plugin installation paths as-is when creating the plugin, but the plugin loader relocates them to  $HOME/opencpn (on win32).

For this to work, we need patches in pluginmanager.cpp. It's really two things: the main program must find the plugin also in the new location, still  finding other plugins in the old location. This already works on Linux and is trivial to implement also on windows.

The other thing is that the plugin must find it's own data. This requires patching GetPluginDataDir() in pluginmanager, basically  implementing what already works in Linux also in windows. But it also means that the plugin must use GetPluginDataDir() to find it's data -- each and every plugin needs such a patch. It's not nice, but I don't see the alternative. 

That is, we move from a single location for plugins to a unix-like model where plugins could be installed at different locations. The manpage opencpn.1 describes the linux stuff. Win would be similar, but with different defaults.

A plugin loader will have to use locations under $HOME to avoid permissions problems on Windows (see above).

bdbcat, 2019-04-05:
And why, short answer, are we considering this at all?
Dave

leamas, 2019-04-06:
Because any plugin loader needs to store the plugins  in an area writable by regular users. As of https://github.com/OpenCPN/OpenCPN/issues/1347#issuecomment-480078359 this already works for OSX and Linux, but not on windows -- see discussion between Pavel and me above starting at https://github.com/OpenCPN/OpenCPN/issues/1347#issuecomment-479888231.

bdbcat, 2019-04-06:
Still a little bit lost here, sorry.
On OCPN Windows, the user has control over the directory in which to install a plugin.  We have no a-priori knowledge (without registry search) to know where the core OCPN image is loaded.  So, the user tells us during the plugin installer script..  There may be multiple copies of OCPN on a machine, and she may choose which one gets the plugin at plugin install time.
Now, we sure could modify pluginmanager.cpp to always load Windows plugins from a known location in the system.  But this is not "Windows-style", and will be problematic in the long term.
linux and Mac use "standard" locations for these things, but Windows does not.
So, pluginmanager will need to understand the native platform differences to be comfortable for users.

We probably could get away with have some fixed directory as a "landing zone" for downloaded plugins, before installation.  But eventually, the plugin installers get executed, and insert the plugin in the selected OCPN runtime tree.  And we should not try stipulate that location structurally, IMHO.

Or maybe I am missing the point entirely.
Dave

leamas, 2019-04-06:
Pavel's point (if I got him right) is that the dialogs required to elevate priviligies is not easily integrated in an automated workflow managed by a plugin manager.

For similar reasons my thinking is that we should produce tarballs/zipballs for automated plugin installations rather than NSIS packages etc.

leamas, 2019-04-06:
That said, I'm a poor reader, at least at this time of the day:
> Now, we sure could modify pluginmanager.cpp to always load Windows plugins from 
> a known location in the system. But this is not "Windows-style", and will be problematic 
> in the long term.

The proposal is really to load them from from multiple locations. one of them under $HOME.  So if user selects to install a NSIS package she can still use that way,  And of course we couldn't break all current installations with installed plugins.

However, a plugin manager is not only about installing plugins -- it's about the complete lifecycle install, update, uninistall. Nothing is impossible, but my idea is that anything manged this way needs to be in known, writable localtions

leamas, 2019-04-06:
> But this is not "Windows-style", and will be problematic in the long term.

Or perhaps it is? The (few) windows applications i have used (browsers, development tools, mutlimedia) *never* uses uses NSIS or similar ways to install plugins. They have plugin managers loading plugins in  more or less hidden formats  storing them known, fixed locations without user options to change location etc.

rgleason, 2019-04-06:
I am not sure I follow all of the discussion above and the structure of the proposed,  but this is from my experience:
1. Bdbcat's point: Windows users executing a plugin with the NSIS packaged exe format are prompted for where to install (picking a version of Opencpn to install into).
2. Is Leamas is thinking that the plugin will be installed from within the Opencpn program, like Linux does? Is this possible? Wouldn't the Opencpn version have to be closed and reopened to have the plugin work? 
3. A note about the c:\programdata\opencpn\plugins\[plugin_pi name]\data.  In the event that the user modifies an xml file for example, Sean has actually programmed to copy "some-file from c:\program files (x86)\opencpn\plugins\[plugin_pi name]\data\some-file were it is read only,  to c:\programdata\opencpn\plugins\[plugin_pi name]\data\some-file, OR C:\user\all users\opencpn\plugins\... etc   (which is read-write), where it can then be modified by the user and saved.  I am not quite sure what the mechanics and sequence is.
4. Some of the plugins that do require the ability to modify files and change settings by the User are:
Watchdog_pi, Weatherfax_pi  (coordinatedatasets.xml, usercoordinatedatasets.xml), Weather_Routing_pi  (weather_routing_configuration.xml and polar files, etc), Celestial Navigation has sights.xml file.  So this is a quite common need for plugins.

Sorry I can't be more help.

leamas, 2019-04-06:
> Is Leamas is thinking that the plugin will be installed from within the Opencpn program, 
> Is this possible?

Yes.  After all, I bet you have been using programs which works this way (the browser you are using to read this comes to mind...).

> is Leamas is thinking that the plugin will be installed from within the Opencpn program,

Yes. Pavel's idea is basically the same, the difference is that he installs from within a plugin rather than the main program.

>  like Linux does?

There is no difference between platforms here, as of today the plugins are installed outside of the main ocpn program in all cases.  

> Wouldn't the Opencpn version have to be closed and reopened to have the plugin work?

I. e., restarted. Perhaps, some apps needs a restart after installing plugins. If so, then we restart... I don't see this as a major problem. 

 > Some of the plugins that do require the ability to modify files and change settings by the User are...

I don't see any particular changes for these plugins. The crucial point is how the plugin locates it's own installed ro data. Whatever locations it then writes to shouldn't be affected, it should work as it always has.

The ro-only data is as of today in` c:\program files (x86)\opencpn\plugins\<plugin_pi name>` The proposed action is that we move it to` $HOME/opencpn\plugins\<plugin_pi name>` in a way that's basically invisible to the plugin as long as it uses GetPluginDataDir() to locate the data.

Hakansv, 2019-04-06:
--I. e., restarted. Perhaps, some apps needs a restart after installing plugins. If so, then we restart... I don't see this as a major problem.--
If a plugin is previously installed and O is running the .dll is of course open and then you can't replace it with an update. So yes O has to be close before any plugin update. Otherwise something like this will occur:
Sharing violation
0 File(s) copied

$HOME/opencpn\plugins\<plugin_pi??  Your use of $HOME is some heritage from the UNIX world and don't exist in the Win world. See chart1.cpp for more info. The %HOMEPATH% refers to user's documents and are not used by O.

leamas, 2019-04-06:
@Hakansv : Yes, there are all sorts of problems when e. g.  doing an update.  However, I don't see them as blocking, they could be handled one way or another. This subthread is really about @rgleason asking whether a a plugin manager is possible. In short, I definitely think it is.

bdbcat, 2019-04-06:
I'll agree that a plugin-plugin can be crafted to manage, download and update the plugins for the particular instance of OCPN that is actually running right now. The update problem is workable, with the possible exception of having the plugin-plugin updating itself.  The install locations can be detected at runtime, and the NSIS installer may be run automatically as a popup window.  Restarts would not be necessary, on most platforms.
This has some advantages:
1.  All the work of figuring out the vagaries of a particular plugin install are contained within the familiar NSIS installer, as is now.
2.  There is no need to pre-define an install location or structure in the plugin metadata.  It is determined at runtime by the installer.
3. Android plugins will have their own platform specific funnys to manage.  Easily done in this model.
4.  We retain decoupling of plugin structure and OCPN core.  This is good.  We cannot predict where plugins will go, functionally, over time.  Best to let the plugin itself be responsible for its setup, within limits.

So, I favor, at least for Windows, a plugin-plugin plan.  And why not for all platforms?.  For instance, in linux, we can just run dpkg quietly in a shell, asking for password as necessary.  On Mac, we run the ".pkg", with GUI, in a popup window. (Mac may require app restart....)

Dave

leamas, 2019-04-06:
> So, I favor, at least for Windows, a plugin-plugin plan.

To implement a plugin manager in a plugin or in the main code is IMHO a separate issue, and perhaps not the most important right now.

>  And why not for all platforms?. For instance, in linux, we can just run dpkg quietly in a 
> shell, asking for password as necessary. On Mac, we run the ".pkg", with GUI, in a 
> popup window. (Mac may require app restart....)

Frankly, this won't work. Linux isn't dpkg --  it's .deb, .rpm, flatpak, snap and what not with their different packaging models. My primary interest in this is to resolve this once and for all with a plugin manager and a universal, simple plugin package. The matrix of maybe 20 plugins x  7 platforms/package formats  is IMHO just not maintainable.  

In the end, using such a universal, plugin format  would transform the Ocpn plugin model to that of other apps (also on Windows) which I have used.

Even the dpkg model for debian/ubuntu is probably not sustainable when we get opencpn accepted into Debian. This will lead to a pressure to create proper plugin packages, something that I doubt we can maintain over time (the ones made by cmake/cpack simply cannot be used in this context).

Of course, we could let windows users continue to use NSIS packages in some automated setup. However, in that case I  think it would be best to decouple this from the other platforms. To be honest, I don't really see the user value in this compared to current situation.

@nohal : thoughts?

nohal, 2019-04-06:
- I am still worried about how is this going to work on Linux with the extreme ABI incompatibilities between distributions and even their versions, the many package formats are just a cherry on this nightmare cake IMHO. The thing somewhat works for other apps only because their platform support is limited or plugins are actually written in an interpreted language like Python or Lua.
- I don't see any point in running the installers by the plugin manager, this thing should simply stay in userspace and be one-click, compared with all the rest, making it happen is trivial. At the other hand, If it has to be so for some reason I still don't see, let it be so...

leamas, 2019-04-06:
> I am still worried about how is this going to work on Linux with the extreme ABI incompatibilities 

Indeed, a plugin must be built on target distro to work reliably.  It's a mess.

The light in the tunnel is the CI builds. I have been able to run a Fedora 28 docker container inside a CI build (last commit in #1335). Basically, this means that we could support quite some platforms without too much effort as long as the plugin format is the same.

nohal, 2019-04-06:
Well, there are ~40 plugins, at least half of them poorly maintained. Someone will have to spend non-trivial amounts of time on just keeping the CI configs up to date for all of them, "forever". If you are the taker, good, I'm sure I've had enough of doing this kind of stuff for them the past years and am certainly not.

leamas, 2019-04-06:
> If you are the taker, good, 

I'm certainly not. But I might be able (banging head to table, why am i doing this...) to create a reasonable workflow for the 2-3 plugins I use. Perhaps it could set an example for others.

The important issue is that while this is hard, producing all sorts of packaging for each of these plugins is impossible. You and I agree seem to agree on this, but not perhaps not Dave...

nohal, 2019-04-06:
We need something we really are able to implement without burning out the available resources. There is no half-way solution as it would simply generate an endless stream of "Why is XY not there?" and "When will XY be there?", which would be mostly ignored by the XY's owners as are any such requests now and the only solution would be to use the same resources we don't have.

Are we able to reasonably deliver Windows-only (~75% of the userbase covered)? Let's do it. Windows and macOS (85+%)? Also good. Windows, macOS and recent Ubuntus (90+%)? Or Windows, macOS and the Flatpak/Snap systems?

I honestly start to think that general Linux support is too much work bringing too little to be worth the effort. And as long as the normal distro-specific packaging stays possible, once there is someone interested in plugin X on distro Y, nothing can stop him, exactly as now.

bdbcat, 2019-04-06:
On Windows:
A user space installer could  work, and I agree may be better for the internet-connected user.  To make this work,  the metadata wants to be sufficiently rich to describe all possible dll, helper binary, and data dependencies, and their final install locations relative to the running OCPN instance.  A hardcoded location specification will not work very well.  We may even need some script triggers defined, (ala dpkg man database update triggers).  I cannot think of any examples offhand for our plugin universe, but easily could happen with e.g. permissions..
At present, on Windows, only the binary plugin installer knows this stuff specifically  Same on Mac, actually.

Dave

rgleason, 2019-04-06:
I agree, this needs to be focused on mac, windows and perhaps the most frequently used linux, but launchpad seems pretty good. 
I like the idea of using CI for all 3. Won't that solve 95%?
I think to or 3 examples for authors to follow is most practical.
It would be nice to know how many users for each pi, then we could cull! Maybe we should start culling them now?
If there is no one maintaining them, just cull those? What does cull mean? Just drop it? Remove documentation too?
For example, ais radar?

nohal, 2019-04-06:
Users seem to love AIS Radar. For some reason I personally don't understand. And it is one of the least broken ones...

rgleason, 2019-04-06:
Ok maybe that is a bad example. Maybe we try to limit the # that are maintaibed somehow.

Hakansv, 2019-04-06:
I've no technical knowledge to add but FWIW some thoughts comes to my mind:
What's all this about? 
For Windows the present installer works great and no users are complaining!? 
For Mac I don't know but I've not seen much user complains.
For Linux there seems to be a lot of different system and the automated update system, PPA, creates a huge job for Pavel if I'm correct. But it's also the system where it's comparable very simple to build from source. So if the job to keep PPA up to date for all Linux system and all plugins couldn't a more automated build from source method be created by you Linux wizards to be used by these Linux users where a PPA is not available?

rgleason, 2019-04-06:
Hakan you could read Leamas comments and Bdbcat's thoughts.
1. Easier maintenance of an xml plugin catalog which describes the parameters necessary to install a plugin. Have a collaborative github plugin repository for selected people to update the catalog.
2. Maybe have a plugin to install the plugins from within Opencpn.
3. Maybe use CI for MacOsx and Window or scripts to make this possible. Maybe use NSIS or not.

rgleason, 2019-04-06:
Maybe keeping the CI builds up to date could be a shared effort? 
(I don't know if this would work.) 
One or two leading programmer's providing the example plugins.
2-3 others to make PR's to the authors after a review by the leading programmer, until the process is understood.
When a change is needed those involved are noticed somehow.
Dev manual has a page with the process outlined, for new programmers to learn.

leamas, 2019-04-07:
@nohal wrote:

> I honestly start to think that general Linux support is too much work bringing too little to 
> be worth the effort.

There is really two things here: The main program and the plugins:

As for the main program we could actually cover all Linux users using Debian packages and flatpak. The Debian packages could be used by the largest debian/ubuntu echosystem, and they are also small enough to be used on a Raspberry. When the downstream package is accepted[1]   the status here would actually be quite good.

The flatpak package (#1115, help needed) works for all Linux users (also debian/ubuntu). It's a a bit heavier, but  not enough to be a practical problem on regular laptops.  The bundling taking place in a flatpak handles the linux distro and distro version ABI mess

Personally I think we should abandon other linux packaging. This includes rpm (#1290) and snap (#1310). They simply does not add much user value.

TL;DR  It's actually not that hard to offer all Linux users a usable packaging of the main program. The plugins is another issue

[1] https://ftp-master.debian.org/new.html

leamas, 2019-04-07:
@bdbcat  wrote:
> On Windows:
A user space installer could work, and I agree may be better for the internet-connected user. 

This is great, some common ground...

> To make this work, the metadata wants to be sufficiently rich to describe all possible dll, helper binary, and data dependencies, and their final install locations relative to the running OCPN instance. A hardcoded location specification will not work very well.

Why? (all other applications with plugins I know have hardcoded paths)

> We may even need some script triggers defined, (ala dpkg man database update triggers). I cannot think of any examples offhand for our plugin universe, but easily could happen with e.g. permissions.

Right, we could implement it as needed when we get into the hairy details of different plugins

leamas, 2019-04-07:
@rgleason wrote
> Maybe keeping the CI builds up to date could be a shared effort?
(I don't know if this would work.)

Perhaps, but -- we need more of common understanding on what to do and how. Looking into a distant future I see a new plugin manager, patching of plugins, new plugin CI build chains and what not. But, we are just not there...

leamas, 2019-04-10:
@Hakansv wrote:

> $HOME/opencpn\plugins<plugin_pi?? Your use of $HOME is some heritage from the UNIX world and don't exist in the Win world. 

Yes... but I'm not the only one. Anyway, there is  a dialog with Dave in #1343 on this. For now, let's  read $HOME as _some user-writable plugin directory selected by user_. 

One open issue is if we can have a common directory (or set of directories) for all plugins or not. 

Sorry for not answering until now, This bug contains too many subthreads for my little brain.

rgleason, 2019-04-24:
Don't know if this will help, but Appveyor can push a built Artifact to a new Github Release.
I've gotten this to work for weatherfax_pi, see
Use of appveyor 'auth_token' to push a new github release #134

leamas, 2019-04-24:
Anything can help. However, still to early. From a plugin perspective, this the todo-list as I see it now.

  - patch the plugin to use the new function GetPluginDataDir() to locate it's data (that is, if it has data).
  - Patch the plugin build system to make it possible to create a tarball instead of e. g., an NSIS 
    package (as an option).
  - Patch the build system to create an XML metadata file.  This will include an url from which the
    plugin can be downloaded, so publishing the built artifacts is indeed a major step.

That should be all for the plugin. The main code is, however, far from even a rough alfa state.

rgleason, 2019-04-24:
It is worth thinking about this IF
> if they have appveyor accounts and set up their auth_token correctly,

Will all the authors have an interest in doing this, with good instructions?
PROS
1. The installation of the plugin gets integrated into Opencpn process, saving time on maintenance.
2. A simple git push origin [tag] like "v1.9.1-ov50" is all that is necessary.
3. I could create both MacOS and WIN the same way, perhaps using different tags?

CONS
1. The authors have to have an appveyor account and set up their auth_token properly.
2. The authors have to follow some detailed instructions.
3. The process is dependent on the authors (just as it is now).
4. Some mistakes will be made issuing Releases at first, and continuing, as people get familiar.

Alternatively - Have OpenCPN Plugin Repository
1. Plugins are cloned into the repository
2. Authentication is handled here
3. Plugins are Setup for pushing to a Git Release upon issuance of a Git Remote [Tag]
4. A PR is made back to the Owner of the plugin.
5. The releases are then in a central repository.
6. A limited number of knowning collaborators maintain the Repos.

Note: Big systems tend to have unanticipated overhead.

rgleason, 2019-04-28:
These are certainly doable, one plugin at a time, with the cooperation of the author.

- patch the plugin to use the new function GetPluginDataDir() to locate it's data (that is, if it has data).
- Patch  the plugin build system to make it possible to create a tarball instead of e. g., an NSIS
package (as an option).

I wouldn't mind changing to a tarball that just unpacks the plugin dll and its directories to the right places.

- Patch the build system to create an XML metadata file. This will include an url from which the
plugin can be downloaded, so publishing the built artifacts is indeed a major step.

I think the above is a good idea to save time and effort.

For an example, I was thinking about how Notepad++ works with plugins and updates on windows, it just seems to know what to do, no paths needed.  On the other hand NotePad++ as I use it does not seem to support installation of parallel versions, which is something I've found to be incredibly useful. 

Currently with Windows, if you just have one version of OpenCPN, the user selects that version and the installer then does all the work (just as for parallel installs). Would this type of functionality change?

The  "build" part of this would rely on the authors use of CI and possibly pushing a new tag to the remote to create new github "Release" or even allowing local builds and publishing to github "Release"?   
With the exception of the patches above, all of the creation/programming of plugins would stay the same, or could be improved in small steps using CI.  Also assumes Plugins can still be built separately as standalone.

Transition to Tarballs  - Would this have to occur all at once or gradually?  I hope gradually, which means OpenCPN would support two both types of plugin installations for awhile anyway?

rgleason, 2019-04-28:
> The ro-locations cannot be used.
>
> Is this to say the the typical climatology_pi plugin could (should?) be installed as
>
> ```
> C:\ProgramData\opencpn\plugins\climatology_pi.dll [...]

> he ro-locations cannot be used.
>
> Is this to say the the typical climatology_pi plugin could (should?) be installed as
>
> ```
> C:\ProgramData\opencpn\plugins\climatology_pi.dll [...]

Normally the DLL and many of the files get installed here:
C:\Program Files (x86)\[OpenCPN 4.8.8]\plugins

leamas, 2019-04-29:
> Transition to Tarballs - Would this have to occur all at once or gradually?

It must be gradually, period.

> Normally the DLL and many of the files get installed here
--users can't usually view these files or access without admin:
C:\Program Files (x86)[OpenCPN 4.8.8]\plugins

Indeed. Which means that we need to change the defaults to some path actually user-writable.

> For an example, I was thinking about how Notepad++ works with plugins and updates on windows, it just seems to know what to do, no paths needed. On the other hand NotePad++ as I use it does not seem to support installation of parallel versions, which is something I've found to be incredibly useful. Does Linux do this? 

Sort of. The main executable could actually be anywhere, and as long as you start it from the commandline different versions is "out of the box". And the plugins loaded could be tweaked using the OPENCPN_PLUGIN_DIRS environment variable.

However, all versions uses the same configuration directory ~/.opencpn. To fix this, we need to resolve #1022.

Anyway, the linux part, sort of the starting point, is nowhere close to an alfa state.

rgleason, 2019-04-29:
Many of these issues have been dealt with by other programs. I say this because the Sigil - ereader github Issues  [cmake: use GNUInstallDirs #414](https://github.com/Sigil-Ebook/Sigil/pull/414#issuecomment-487422154)  long discussion and most recent comment  "standardisation of path declaration"

rgleason, 2019-05-02:
BTW re MacOS
I assume you all know about this.

https://www.computerworld.com/article/3269007/what-apples-32-bit-app-phase-out-on-mojave-means-to-you.html

Apple said in June it will bar all 32-bit applications from running on up-to-date Macs in little more than a year.

leamas, 2019-05-11:
For the record: after poking around it looks that there are basically two reasonable plugin locations on Windows. The first is the current, in the same directory as the main application is installed in. This is not user-writable.

The other is %LOCALAPPDATA% which usually expands to C:\Users\\\<username\>\\AppData\Local.
(Documents and Settings on W7, though). This is the location recommended by MS, and also the one used by chrome and firefox for their plugins (I tend to trust these to make correct decisions).

rgleason, 2019-05-12:
Yes, in Win10 from the command prompt  "dir %LOCALAPPDATA%

C:\Users\Rick>dir %LOCALAPPDATA%
 Volume in drive C is OS
 Volume Serial Number is B845-7168

 Directory of C:\Users\Rick\AppData\Local

05/11/2019  08:24 PM    <DIR>          .
05/11/2019  08:24 PM    <DIR>          ..
---With just some  of the directories---
12/08/2015  04:44 PM    <DIR>          ActiveSync
01/17/2019  01:42 PM    <DIR>          Adobe
04/10/2016  08:50 PM    <DIR>          Apps
04/10/2016  09:18 PM    <DIR>          Atlassian
08/26/2016  04:58 PM    <DIR>          Audacity
07/12/2016  06:10 AM    <DIR>          Chromium
07/07/2017  09:25 AM    <DIR>          Citrix
12/31/2015  02:27 PM    <DIR>          Comms
08/17/2018  03:52 PM    <DIR>          FileZilla
05/02/2019  07:02 PM    <DIR>          GitHub
04/10/2016  09:18 PM    <DIR>          GitHub,_Inc
05/02/2019  06:11 PM    <DIR>          GitHubDesktop
11/09/2018  08:42 PM    <DIR>          Google
05/03/2019  10:52 AM    <DIR>          Microsoft
12/09/2015  11:03 AM    <DIR>          MicrosoftEdge
04/10/2016  08:52 PM    <DIR>          Mozilla
11/13/2018  08:06 PM    <DIR>          OneDrive
08/25/2017  08:51 PM    <DIR>          Package Cache
04/18/2011  01:22 PM    <DIR>          PackageAware
04/30/2019  10:05 PM    <DIR>          Packages
07/17/2018  02:29 PM    <DIR>          PackageStaging
08/25/2017  08:51 PM    <DIR>          Programs
09/10/2018  07:30 PM    <DIR>          Publishers
04/10/2016  08:54 PM    <DIR>          Thunderbird

Hakansv, 2019-05-12:
@leamas 
The plugin binaries, .dll, are now installed in the program files directory %ProgramFiles(x86)% e.g. "C:\Program Files (x86)\OpenCPN\plugins\oeSENC_pi.dll" that's correct and since the normal user don't need access to that it's a good place.

The plugins user data are now where OpenCPN user data lives e.g "C:\ProgramData\opencpn\oesenc_pi\..."  is called by the variable %AllUsersProfile% or %ProgramData%. I can possible agree it's not not the best place since the files ar "hidden" for most users and can be hard to find. You also mostly need admin rights to write. 
To switch to use %LocalAppData% instead may be easier to find for all since the rights are there for all users. But that switch make needs for searching in %ProgramData% for old config file when Opencpn is installed in case of an upgrade from today's O to next version. But that would be doable.

rgleason, 2019-05-12:
I agree with what Hakan wrote.

leamas, 2019-05-12:
> I agree with what Hakan wrote.

And so do I ;)

The linux  side already support multiple locations described by a path like` ~/.local/lib/opencpn:/usr/lib/opencpn:/usr/lib/opencpn`. My idea is to do the same for windows, basically using a path like `%LOCALAPPDATA%;opencpn-install-dir`.

However, ideas are cheap. Working code is a lot of work.

[23 more replies]

closed 2019-05-25 by leamas