OpenCPN/OpenCPN #1349: Compiling under FC28

JONA-GA, 2019-04-05:


JONA-GA, 2019-04-05:
in CMakelists.txt, ligne 435
FIND_PACKAGE(wxWidgets REQUIRED COMPONENTS std,aui,gl)

to compile on FC28.....

nohal, 2019-04-05:
To which branch are you referring?

JONA-GA, 2019-04-06:
I update from Master.
But I realise that it's 4.8.2...
Was working before. i think on an update of wxGTK3 package with a change behavior of wx-config....
Should i wait for a merge of 5 in Master ?....

nohal, 2019-04-06:
Master is newer than O5, supports new cmake versions without problems and the line 435 you refer to is definitely not making sense for your change (https://github.com/OpenCPN/OpenCPN/blob/master/CMakeLists.txt#L435), please check your local tree.

JONA-GA, 2019-04-06:
Ok, i should have a bad local tree....
Thanks for your help

nohal, 2019-04-10:
As there is no more info, I suppose your problem is gone with the correct source tree.

closed 2019-04-10 by nohal

JONA-GA, 2019-04-10:
Thanks for the support, Pavel...
Forgotted to merge.........