OpenCPN/OpenCPN #1351: Undefining USE_S57 is broken

taruti, 2019-04-09:
Not defining USE_S57 seems to have bitrotted.

Should it be fixed or removed in master?

leamas, 2019-04-09:
Good question, again. The best code is the code which is removed, so if there is no usecase for it should certainly be removed. 

If it should be fixed I'd really appreciate if we could postpone it to after merging #1355 to avoid some really bad rebasing mess. 

A  fix should include renaming option to OCPN_USE_S57

taruti, 2019-04-09:
This has absolutely no hurry, waiting for #1335 is fine in any case.

leamas, 2019-04-09:
So, the important question for now: Is there any usecase for disabling USE_S57? 

Unless anyone speaks up in this bug lets say it will be removed. I  will add a final PR after #1335 with still more patches. Adding removal if this shouldn't be that hard.

PR #1371 (2019-04-17): Remove broken option OCPN_USE_S57 a. k. a. USE_S57
This option cannot be set to OFF  without breaking the build since at least 4.8.6. Remove the option completely together with the compilation constant USE_S57.

There should be no hurry with this, things need to settle first.

Closes: #1351
closed 2019-04-17 by nohal